RainerKuemmerle/g2o #58: Some examples do not work when building g2o to static libraries

txq1986, 2015-01-23:
Hi Rainer,

Thank you for the great g2o work and I am trying to use it.

Are you aware of that some examples do not work when building g2o to static libraries (.a files). For example, I got a bunch of errors of tutorial_slam2d
fatal error in creating cache of type TUTORIAL_CACHE_SE2_OFFSET
addEdge: FATAL, cannot resolve caches for edge 0x23416e0
g2o::Cache* g2o::CacheContainer::createCache(const g2o::Cache::CacheKey&)
fatal error in creating cache of type TUTORIAL_CACHE_SE2_OFFSET
addEdge: FATAL, cannot resolve caches for edge 0x23419a0
...

cmake commands:  $cmake -DBUILD_SHARED_LIBS=OFF ../
OS: Ubuntu 14.04

Build to shared libraries are completely perfect and all the examples work well. Do you have an idea what is the possible reasons for it? Thanks

XQ

jsgaobiao, 2017-03-21:
I met the same problem.
Have you solved it ? @txq1986

AlwaysLearningJary, 2017-03-23:
I have the same problem, I was thinking the different version of g2o wasn't compatible. But I remove all files, and reinstall g2o again, it still doesn't work well.
OS:Ubuntu 14.04

jlblancoc, 2017-03-23:
A classic problem with static linking... compiler optimizes out what it thinks is not used, which is a problem with class factories. 

The origin of the problem is this: 

```
    Factory* f = Factory::instance();
    HyperGraph::HyperGraphElement* e = f->construct(key.type());
```

The factory returns `nullptr` for string "TUTORIAL_CACHE_SE2_OFFSET", which means that this line is not being executed:

```
  G2O_REGISTER_TYPE(TUTORIAL_CACHE_SE2_OFFSET, CacheSE2Offset);
```

The trick used in `G2O_REGISTER_TYPE()` to ensure the automatic registration is not working with static linking. 

I know of this problem because I had to switch to a more complex auto-registration macro for a project of mine. 
Solutions:
* Forget about static linking...
* Implement something like [MRPT_INITIALIZER()](https://github.com/MRPT/mrpt/blob/master/libs/base/include/mrpt/utils/initializer.h) in G2O...

Cheers!

taiping-z, 2017-05-25:
@jlblancoc 

Hi, I am sorry. I do not understand how to implement something like MRPT_INITIALIZER() in G2O.
Do you have any demos for solving this problem?

Thank you!

kyosho81, 2019-12-25:
do like this
namespace g2o {
	namespace tutorial {

		G2O_REGISTER_TYPE_GROUP(tutorial_slam2d);

		G2O_REGISTER_TYPE(TUTORIAL_VERTEX_SE2, VertexSE2);
		G2O_REGISTER_TYPE(TUTORIAL_VERTEX_POINT_XY, VertexPointXY);

		G2O_REGISTER_TYPE(TUTORIAL_PARAMS_SE2_OFFSET, ParameterSE2Offset);

		G2O_REGISTER_TYPE(TUTORIAL_CACHE_SE2_OFFSET, CacheSE2Offset);

		G2O_REGISTER_TYPE(TUTORIAL_EDGE_SE2, EdgeSE2);
		G2O_REGISTER_TYPE(TUTORIAL_EDGE_SE2_POINT_XY, EdgeSE2PointXY);
	}
} // end namespace
int main()
{
return 0;
}

closed 2025-02-14 by github-actions[bot]