RainerKuemmerle/g2o #59: Eigen Compatibility Issue

jmtatsch, 2015-02-06:
Hi,

i cannot build g2o with eigen-3.2.3 on OS X  due to:

 no viable conversion from 'Scalar' (aka 'double') to
      'Eigen::Rotation2D<double>'

[ 41%] Building CXX object g2o/types/slam2d/CMakeFiles/types_slam2d.dir/edge_se2.cpp.o
In file included from /Users/tatsch/g2o/g2o/types/slam2d/edge_se2.cpp:27:
In file included from /Users/tatsch/g2o/g2o/types/slam2d/edge_se2.h:30:
In file included from /Users/tatsch/g2o/g2o/types/slam2d/vertex_se2.h:30:
In file included from /Users/tatsch/g2o/build/g2o/config.h:18:
In file included from /Users/tatsch/g2o/g2o/core/eigen_types.h:31:
In file included from /usr/local/include/eigen3/Eigen/Geometry:39:
/usr/local/include/eigen3/Eigen/src/Geometry/Rotation2D.h:74:46: error: no viable conversion from 'Scalar' (aka 'double') to
      'Eigen::Rotation2D<double>'
  inline Rotation2D inverse() const { return -m_angle; }
                                             ^~~~~~~~
/Users/tatsch/g2o/g2o/types/slam2d/se2.h:84:19: note: in instantiation of member function 'Eigen::Rotation2D<double>::inverse' requested here
        ret._R=_R.inverse();
                  ^
/usr/local/include/eigen3/Eigen/src/Geometry/Rotation2D.h:41:7: note: candidate constructor (the implicit copy constructor) not viable: no known
      conversion from 'Scalar' (aka 'double') to 'const Eigen::Rotation2D<double> &' for 1st argument
class Rotation2D : public RotationBase<Rotation2D<_Scalar>,2>
      ^
/usr/local/include/eigen3/Eigen/src/Geometry/Rotation2D.h:41:7: note: candidate constructor (the implicit move constructor) not viable: no known
      conversion from 'Scalar' (aka 'double') to 'Eigen::Rotation2D<double> &&' for 1st argument
class Rotation2D : public RotationBase<Rotation2D<_Scalar>,2>
      ^
1 error generated.
make[2]: *** [g2o/types/slam2d/CMakeFiles/types_slam2d.dir/edge_se2.cpp.o] Error 1
make[1]: *** [g2o/types/slam2d/CMakeFiles/types_slam2d.dir/all] Error 2
make: *** [all] Error 2

Can you check if you can still compile with eigen-3.2.3?

RainerKuemmerle, 2015-02-06:
I can see the same error on Travis for the OSX build.
https://travis-ci.org/RainerKuemmerle/g2o/jobs/45540873

Looking at the Eigen changelog for 3.2.4, I guess it is fixed with 3.2.4.
http://eigen.tuxfamily.org/index.php?title=ChangeLog#Eigen_3.2.4

I might test with clang under Linux and Eigen 3.2.4, gcc with 3.2.3 is fine (my current setup).

jmtatsch, 2015-02-09:
Thank you, I can confirm this fixed with Eigen 3.2.4. I will make a pull request for eigen 3.2.4 with home-brew.

closed 2015-02-09 by jmtatsch