RainerKuemmerle/g2o #61: How to check for convergence

skasperski, 2015-03-13:
Hello,

I have constructed a very simple slam3d-graph problem with just 3 nodes and 3 edges. The optimization works correctly, however it always does as many iterations as I give it in SparseOptimizer::optimize(int iterations). Shouldn't such a simple problem converge much faster? Or do I have to call it with iterations=1 and check any criteria myself?

Best regards,
Sebastian

agpratik, 2015-03-13:
Hi, Sebastian

You can add a terminate action for your optimizer based on the error gain threshold.

See this for an example: https://github.com/RainerKuemmerle/g2o/blob/master/g2o/apps/g2o_cli/g2o.cpp#L211

closed 2015-03-13 by agpratik

skasperski, 2015-03-13:
I added a SparseOptimizerTerminateAction, but now it always stops at the number of iterations set with terminateAction->setMaxIterations(), no matter whet I use as gain threshold.