RainerKuemmerle/g2o #66: types_slam3d_addons will not compile using /Ob0 flag

Sahloul, 2015-03-26:
I am not sure if this applies to all platforms as I am currently using **Visual Studio 2010**.

When disabling the inline function expansion (i.e setting */Ob0* on one of the *CMAKE_C??_FLAGS_???*), the project will fail to compile due to *g2o::BaseEdge<,>::information()* being already defined. Error codes **LNK2005** and **LNK1169** are thrown.

Please note that the default behaviors in the Debug mode is /Ob0.

Suggested workaround: force **/Ob1** or **/Ob2** in the compiler flags using the *CMake file*.

SkyRiderMike, 2015-05-15:
I meet the same problem when I try to build with visual studio. But how to solve this problem?

Sahloul, 2015-05-15:
I think I already wrote my suggestion!
http://www.cmake.org/Wiki/CMake_FAQ#Make_Override_Files

SkyRiderMike, 2015-05-15:
Thanks！I don't know how to change a CMake file but I'll follow your suggestion and learn it...thank you sincerely!

commit 17463c22 (2015-09-07): Merge pull request #80 from jlblancoc/fix_msvc_dup_link_error

MSVC linker: fix duplicated symbols in debug (Fixes #66)
commit 541cef66 (2015-09-07): MSVC linker: fix duplicated symbols in debug (Fixes #66)
commit 92744f62 (2015-09-08): Merge pull request #2 from jlblancoc/fix_msvc_dup_link_error

MSVC linker: fix duplicated symbols in debug (Fixes #66)
closed 2015-09-07 by RainerKuemmerle