RainerKuemmerle/g2o #71: C++03 compatibility

skasperski, 2015-05-21:
Hello,

after the last pull from the repo it seems that g2o requires C++11 compiler. (unconditional use of <unordered_map>) Was it intended to drop backward compatibility? (e.g. ROS requires C++03 afaik)

Best regards,
Sebastian

RainerKuemmerle, 2015-05-27:
Hi Sebastian,

in some aspects it was intentional. Before that we used tr1 unordered_map and had several pre-processor statements to adjust the includes and typedefs for other compilers. For example, Visual C has no tr1 but provides unordered_map.
Furthermore, C++11 is available for quite some time and I preferred activating C++11 over using boost.

Cheers,
Rainer

skasperski, 2015-05-29:
Hello Rainer,

I think I can handle this by activating C++11 locally for now. However, how is the g2o-ros package (http://wiki.ros.org/libg2o) affected by this? Is this still being updated and/or recommended to use? I remember that I tried using it once but failed for some reason. (Would have to check again to see why it failed exactly.)

Thanks,
Sebastian

croesmann, 2015-05-29:
Hi Sebastian, 

I've noticed this conversation since my ros package also depends on libg2o. I think I can explain how libg2o is affected by this issue.
g2o itself is not a "wet" (catkin based) ros package, rather it is included as a third party package (libg2o). The libg2o package is therefore maintained externally, currently by Vincent.
The current *Jade* release has been compiled from the most recent version of the g2o master branch and therefore is actually affected by this issue. In order to be compliant with REP003 (http://www.ros.org/reps/rep-0003.html) we could "step back" and compile the last version before C++11 as libg2o has been enabled (I guess it was in Aug '14). But on the other hand, we wouldn't be able to update libg2o with new fixes and features from this repo. What do you think is the best way to go? As libg2o is a non-core stack and rep003 is more like a guideline in that case, it is not really crucial to require C++11 activation in my opinion.

Best regards,
Christoph

skasperski, 2015-05-29:
Hello Christoph,

I don't think that just staying at an old version and never updating g2o again is a solution. Sooner or later people will need new features or even fixes, making the ROS package useless. A maintained ROS package is really convenient to use, but so far I had not too much trouble using g2o as an external dependency. So to answer your question: I would prefer to be able to use updates from g2o, even if it means to require C++11. But as far as I understand it, it will not work to publish a ROS package (libg2o) that needs C++11 enabled. So there should be some recommended way how to use g2o in other ROS packages and what restrictions this implicates.

Best regards,
Sebastian

croesmann, 2015-05-29:
Hi Sebastian,

actually it is possible to publish C++11 only packages, since the compilers utilized for building the official debs support it. The ROS Jade version of libg2o (which is indeed C++11 only) is already available on the public repos. In contrast the ROS Indigo release of libg2o wasn't updated for approx. a year, therefore it is still C++03 compliant (and does not contain recent updates). 

Best regards,
Christoph

skasperski, 2015-06-01:
Just to give a short update: I now built g2o from current master-branch and tried to use it in Indigo by activating C++11 support in the CMakeFile. It did compile, but runs into SIGSEGV immediately. I was told that this is a common thing to experience when mixing code generated by different compilers. 
Now I am very unsure how to continue with using g2o. I run into a strange problem with the optimization in my (1year+ old) g2o installation and cannot tell whether this still exists on current master.

Maybe it would be an option to use either C++11 or Boost depending on the availability? This should make g2o portable to a wide range of applications.

mikepurvis, 2016-01-25:
Any further discussion on this? In order of preference, I would favour:

- Automatically detecting available of std::unordered_map at configure time, and switching an #ifdef based on that.
- Adding a manual var for USE_CXX11, defaulted to ON, and switching based on that.

But even there, there's a question mark what the correct fallback behaviour is. Use the tr1 namespace? The container from boost? Something else?

mikepurvis, 2016-01-26:
Possibly the best solution would be if someone could prepare a patch which turns on boost as a dependency and uses the boost containers, then @vrabaud could integrate that as part of the bloom build for ROS.

vrabaud, 2016-02-01:
Hi all, jumping in late.
I do not use g2o anymore so it would make more sense if a real user (that follows the current bug reports like this one) takes over the maintainership.
@skasperski , I am surprised by your SIGSEGV, mixing compilers and options is common (I use newer gcc/clang with Ubuntu's packages which are compiled with a set gcc). I would not be surprised if this is a real fixable code.

As far as using C++11 in ROS, it is only an issue for core packages which have to be portable (like on embedded devices). As long as you don't use features that were not implemented in Trusty's gcc version, you are fine. Try out, worst case, you can rollback :)

Please fix the SIGSEGV and I can then make a release, or please step in as a maintainer. Thx.

gunshi, 2016-04-24:
Hi, the ORB_SLAM repo provides a slightly modified version of g2o, the we're supposed to build before building ORB_SLAM.
I have the following issue that I have reported on the ORB_SLAM issues page, while building g2o:
https://github.com/raulmur/ORB_SLAM/issues/105
But I think this is a more relevant thread to post, since its not related to the modifications done by ORB_SLAM authors to g2o, and related to the x11 headers required by g2o, from what I understand from the discussion.
What changes do I make to the cmakelists.txt file, to get it to build successfully? (I have added set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -stdlib=libc++") to the file already)
Or should I just remove the "tr1/" part and use #include<unordered_map> since that should probably work on OSX. 
Thanks!

closed 2020-10-24 by RainerKuemmerle