RainerKuemmerle/g2o #83: g2o and Data Association

francocurotto, 2015-09-09:
Hi,

I'm trying to do data association  with g2o, and I'm having issues getting the necessary information form the framework. In particular, I'm trying to implement the data association method presented in [this paper](http://robots.stanford.edu/papers/thrun.graphslam.pdf) (GraphSLAM), for which I need the marginalized information matrix between two landmarks, to compute the likelihood of these landmarks being the same. Is there a way to get such matrix directly from g2o? I found the computeMarginals() function from SparseOptimizer class, but I'm not sure what it does.

Also, the GraphSLAM paper provides an equation for computing the marginalized information matrix, but it involves knowing some elements of the covariance matrix of the system, which I'm not sure if you can get those from g2o (or if g2o even compute them). 

Thanks
Franco

PD: I think this question is related to [this one](https://github.com/RainerKuemmerle/g2o/issues/63).

RainerKuemmerle, 2015-09-12:
computeMarginals() allows you to compute the required elements of the covariance matrix.
An example of this can be seen in the main of g2o itself or in the EdgeLabeler class.

closed 2015-09-12 by RainerKuemmerle

francocurotto, 2015-09-14:
Thanks for the answer!