RainerKuemmerle/g2o #87: incremental optimization with "online = true"

merfelsc, 2015-11-15:
Hi,

Thanks for the great work on g2o!

I am successfully using g2o to optimize a 2D graph in an online fashion. That is, I am constantly adding and removing nodes but the biggest part of the graph stays the same between two time steps. This works nicely, but I can't stop wondering about the "online" option for SparseOptimizer::optimize(). Should I use this? What I am currently doing is adding/removing nodes, calling SparseOptimizer::initializeOptimization() and then call SparseOptimizer::optimize() (without the online option).

I tried using the online-parameter and calling initializeOptimization before every optimization or only once at the very beginning, but g2o crashed in every combination (trying to free some memory). I can attach the stack trace and error message if that is helpful. (I am currently using LinearSolverEigen, BlockSolver and OptimizationAlgorithmGaussNewton as provided by g2o.)

Am I using g2o correctly for my problem? Would you advise to use the online optimization option?

- Christian

RainerKuemmerle, 2015-12-08:
The online option is a hack to avoid some initialization if you are constantly adding nodes. As you are also removing nodes, the hack fails.
The hack probably should be removed at some point.
You are using it correctly without the online option.

closed 2015-12-08 by RainerKuemmerle

jumurray, 2017-11-20:
Hi all,

Just wanted to make sure the online option was still considered a hack, to inform the decision about whether or not to take a dependency on it.  If the thinking on this has evolved at all since this issue was closed, please let me know.

Thanks!

hanzhi713, 2021-07-04:
After some search through the codebase, I found this g2o_cli application https://github.com/RainerKuemmerle/g2o/blob/bcb4cb1713f8262b81f1986bd9d6d223a076d01a/g2o/apps/g2o_cli/g2o.cpp#L498 that shows an example of how to use online optimization. (The line number reference isn't exact. Please check out the file)

To summarize in pseudocode:

```cpp
// if this is the first time we call optimize or we need to optimize after the removal of some nodes, 
// we need to make sure that
we correctly reinitialize everything
if (first_time || nodes_removed) {
    optimizer.initializeOptimization();
    optimizer.initializeOptimization(num_iters, false);
} else {
// otherwise we perform incremental optimization for newly added nodes or edges
    optimizer.updateInitialization(nodes_added, edges_added);
    optimizer.initializeOptimization(num_iters, true)
}
```

wlzkstlz, 2021-12-16:
> After some search through the codebase, I found this g2o_cli application
> 
> https://github.com/RainerKuemmerle/g2o/blob/bcb4cb1713f8262b81f1986bd9d6d223a076d01a/g2o/apps/g2o_cli/g2o.cpp#L498
> 
> that shows an example of how to use online optimization. (The line number reference isn't exact. Please check out the file)
> I think I've also figured out how to deal with node/edge removal. To summarize in pseudocode:
> 
> ```c++
> // if this is the first time we call optimize or we need to optimize after the removal of some nodes, 
> // we need to make sure that we correctly reinitialize everything
> if (first_time || nodes_removed) {
>     optimizer.initializeOptimization();
>     optimizer.initializeOptimization(num_iters, false);
> } else {
> // otherwise we perform incremental optimization for newly added nodes or edges
>     optimizer.updateInitialization(nodes_added, edges_added);
>     optimizer.initializeOptimization(num_iters, true)
> }
> ```

here , "optimizer.initializeOptimization(num_iters, true)" actually means "optimizer.optimize(num_iters,true)", right?