RainerKuemmerle/g2o #88: 'MatrixXD' and VectorXD was not declared in this scope

fdp0525, 2015-11-28:
code:
    typedef  g2o::BlockSolver_6_3  SlamBlockSolver;
    typedef  g2o::LinearSolverCSparse<SlamBlockSolver::PoseMatrixType>  SlamLinearSolver;

    SlamLinearSolver* linearSolver = new SlamLinearSolver();
    linearSolver->setBlockOrdering(false);
    SlamBlockSolver* blockSolver = new SlamBlockSolver(linearSolver);
    g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg(blockSolver);

    g2o::SparseOptimizer  globalOptimizer;
    globalOptimizer.setAlgorithm(solver);
    globalOptimizer.setVerbose(false);

//    g2o::Isometry3d
    g2o::VertexSE3  *v = new g2o::VertexSE3();
    v->setId(currIndex);
    v->setEstimate(Eigen::Isometry3d::Identity());
    v->setFixed(true);
    globalOptimizer.addVertex(v);

error:

/usr/local/include/g2o/types/slam3d/se3_ops.h:37: error: 'Matrix3D' does not name a type
/usr/local/include/g2o/types/slam3d/se3_ops.h:38: error: 'Vector3D' does not name a type
/usr/local/include/g2o/types/slam3d/se3_ops.h:39: error: 'Vector2D' does not name a type
/usr/local/include/g2o/types/slam3d/se3_ops.h:40: error: 'Vector3D' does not name a type
/usr/local/include/g2o/types/slam3d/se3_ops.h:41: error: 'Vector3D' does not name a type
/usr/local/include/g2o/types/slam3d/se3_ops.h:42: error: 'Vector4D' does not name a type
In file included from /usr/local/include/g2o/types/slam3d/se3_ops.h:44,
                 from /usr/local/include/g2o/types/slam3d/se3quat.h:30,
                 from /usr/local/include/g2o/types/slam3d/isometry3d_mappings.h:31,
                 from /usr/local/include/g2o/types/slam3d/vertex_se3.h:33,
                 from /usr/local/include/g2o/types/slam3d/types_slam3d.h:27,
                 from main.cpp:4:
/usr/local/include/g2o/types/slam3d/se3_ops.hpp:27: error: 'Matrix3D' does not name a type
/usr/local/include/g2o/types/slam3d/se3_ops.hpp:40: error: 'Vector3D' does not name a type
/usr/local/include/g2o/types/slam3d/se3_ops.hpp:49: error: 'Vector2D' does not name a type
/usr/local/include/g2o/types/slam3d/se3_ops.hpp:57: error: 'Vector3D' does not name a type
/usr/local/include/g2o/types/slam3d/se3_ops.hpp:66: error: 'Vector3D' does not name a type
/usr/local/include/g2o/types/slam3d/se3_ops.hpp:75: error: 'Vector4D' does not name a type
In file included from /usr/local/include/g2o/types/slam3d/isometry3d_mappings.h:31,
                 from /usr/local/include/g2o/types/slam3d/vertex_se3.h:33,
                 from /usr/local/include/g2o/types/slam3d/types_slam3d.h:27,
                 from main.cpp:4:
/usr/local/include/g2o/types/slam3d/se3quat.h:47: error: 'Vector3D' does not name a type
/usr/local/include/g2o/types/slam3d/se3quat.h:55: error: expected ',' or '...' before '&' token
/usr/local/include/g2o/types/slam3d/se3quat.h:59: error: expected ',' or '...' before '&' token
/usr/local/include/g2o/types/slam3d/se3quat.h:93: error: 'Vector3D' declared as an 'inline' field
/usr/local/include/g2o/types/slam3d/se3quat.h:93: error: expected ';' before '&' token
/usr/local/include/g2o/types/slam3d/se3quat.h:95: error: expected ';' before 'inline'
/usr/local/include/g2o/types/slam3d/se3quat.h:95: error: expected ',' or '...' before '&' token
/usr/local/include/g2o/types/slam3d/se3quat.h:116: error: 'Vector3D' declared as an 'inline' field
/usr/local/include/g2o/types/slam3d/se3quat.h:116: error: expected ';' before 'operator'
/usr/local/include/g2o/types/slam3d/se3quat.h:120: error: expected ';' before 'inline'
/usr/local/include/g2o/types/slam3d/se3quat.h:214: error: 'Vector3D' does not name a type
/usr/local/include/g2o/types/slam3d/se3quat.h:287: error: expected type-specifier before 'Isometry3D'

closed 2015-11-30 by fdp0525

jokereactive, 2016-06-22:
How did you resolve this?

RainerKuemmerle, 2016-06-23:
The types are defined in g2o/core/eigen_types.h

Hope this helps.

jokereactive, 2016-06-24:
Worked! Thanks. :)