RainerKuemmerle/g2o #89: Minimal examples

nlw0, 2015-12-04:
I know this is probably a very new project, and very complicated, but it is quite frustrating to start working with it right now. How do I even run anything? What is the format of the data files? Where is the tutorial from the "tutorial_slam2d"? And why did my viewer did not get built? I would appreciate very much a single example at the README, just a single command line I could run to even check my build was OK and that something works at all.

RainerKuemmerle, 2015-12-08:
Some description of the format of the data files is specified in the wiki.
Unfortunately, the wiki does for some reason not render the LaTeX code. It
was working once. The svn on OpenSLAM contains example files.
g2o provides some examples but documentation is bad. The project was
created while being a PhD student. Right now it's a spare time project and
there are so many other things to do...
Probably the viewer did not build because libqglviewer is missing. As
tutorial_slam2d also relies on the libqglviewer lib, it's likely that it's
missing for the same reason.

nlw0, 2015-12-09:
Great, thanks for the reply. I can definitely relate to the problem of maintaining your PhD code! But I already managed to start understanding the given examples, and I hope I can contribute something in the future.

closed 2015-12-10 by RainerKuemmerle