RainerKuemmerle/g2o #91: matrix_operations.h causes internal compiler error with msvc2013x64

dtmoodie, 2015-12-10:
I have a windows 7x64 computer trying to compile from master branch with msvc 2013 x 64 update 5 professional.
I get the following error:

g2o\core\matrix_operations.h(51): fatal error C1001: An internal error has occurred in the compiler.
 (compiler file 'msc1.cpp', line 1325)

When trying to compile with Eigen 3.2.6.

I believe this is the same issue as here:  http://stackoverflow.com/questions/24840094/visual-studio-express-shows-error-internal-error-occured-in-compiler  
Has anyone found a solution for this?

learnsor, 2016-04-09:
I encount exactly the same error

ScottSWu, 2016-04-21:
I was working with a modified version of g2o, and I got a similar error: 'C1060 compiler is out of heap space'. Commenting out `template<>` on lines 51 and 65 worked for me.

This was with Eigen 3.2.8 and VS2015 64-bit.

rushikesh988, 2017-02-09:
```
// g2o - General Graph Optimization
// Copyright (C) 2011 R. Kuemmerle, G. Grisetti, W. Burgard
// All rights reserved.
//
// Redistribution and use in source and binary forms, with or without
// modification, are permitted provided that the following conditions are
// met:
//
// * Redistributions of source code must retain the above copyright notice,
//   this list of conditions and the following disclaimer.
// * Redistributions in binary form must reproduce the above copyright
//   notice, this list of conditions and the following disclaimer in the
//   documentation and/or other materials provided with the distribution.
//
// THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS
// IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED
// TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A
// PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT
// HOLDER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
// SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED
// TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR
// PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF
// LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING
// NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
// SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

#ifndef G2O_CORE_MATRIX_OPERATIONS_H
#define G2O_CORE_MATRIX_OPERATIONS_H

#include <Eigen/Core>

namespace g2o {
  namespace internal {

    template<typename MatrixType>
    inline void axpy(const MatrixType& A, const Eigen::Map<const Eigen::VectorXd>& x, int xoff, Eigen::Map<Eigen::VectorXd>& y, int yoff)
    {
      y.segment<MatrixType::RowsAtCompileTime>(yoff) += A * x.segment<MatrixType::ColsAtCompileTime>(xoff);
    }

    template<int t>
    inline void axpy(const Eigen::Matrix<double, Eigen::Dynamic, t>& A, const Eigen::Map<const Eigen::VectorXd>& x, int xoff, Eigen::Map<Eigen::VectorXd>& y, int yoff)
    {
      y.segment(yoff, A.rows()) += A * x.segment<Eigen::Matrix<double, Eigen::Dynamic, t>::ColsAtCompileTime>(xoff);
    }

	template<>
	//inline void axpy(const Eigen::MatrixXd& A, const Eigen::Map<const Eigen::VectorXd>& x, int xoff, Eigen::Map<Eigen::VectorXd>& y, int yoff)
	inline void axpy<Eigen::MatrixXd>(const Eigen::MatrixXd& A, const Eigen::Map<const Eigen::VectorXd>& x, int xoff, Eigen::Map<Eigen::VectorXd>& y, int yoff)
	{
		y.segment(yoff, A.rows()) += A * x.segment(xoff, A.cols());
	}

	template<typename MatrixType>
	inline void atxpy(const MatrixType& A, const Eigen::Map<const Eigen::VectorXd>& x, int xoff, Eigen::Map<Eigen::VectorXd>& y, int yoff)
	{
		y.segment<MatrixType::ColsAtCompileTime>(yoff) += A.transpose() * x.segment<MatrixType::RowsAtCompileTime>(xoff);
	}

	template<int t>
	inline void atxpy(const Eigen::Matrix<double, Eigen::Dynamic, t>& A, const Eigen::Map<const Eigen::VectorXd>& x, int xoff, Eigen::Map<Eigen::VectorXd>& y, int yoff)
	{
		y.segment<Eigen::Matrix<double, Eigen::Dynamic, t>::ColsAtCompileTime>(yoff) += A.transpose() * x.segment(xoff, A.rows());
	}

	template<>
	//inline void atxpy(const Eigen::MatrixXd& A, const Eigen::Map<const Eigen::VectorXd>& x, int xoff, Eigen::Map<Eigen::VectorXd>& y, int yoff)
	inline void atxpy<Eigen::MatrixXd>(const Eigen::MatrixXd& A, const Eigen::Map<const Eigen::VectorXd>& x, int xoff, Eigen::Map<Eigen::VectorXd>& y, int yoff)
	{
		y.segment(yoff, A.cols()) += A.transpose() * x.segment(xoff, A.rows());
	}

  } // end namespace internal
} // end namespace g2o

#endif
```

It Worked when I replaced contents of my matrix_operations.h file to above. 

File source [https://github.com/Beewe/ORB_SLAM2_Windows/blob/master/Thirdparty/g2o/g2o/core/matrix_operations.h](url)

RainerKuemmerle, 2021-01-05:
Are fixes still needed on MSVC? I run CI on AppVeyor but apart of that do not use Windows.

RainerKuemmerle, 2021-02-06:
Had closer look at the proposed diff now. Already changed accordingly on master.

closed 2021-02-06 by RainerKuemmerle