RainerKuemmerle/g2o #95: Suite-Sparse link is broken in README.md

ningt, 2016-01-26:
Hi I try to use g2o in xcode, but I got errors like the following:
```
Undefined symbols for architecture x86_64:
  "g2o::csparse_extension::cs_cholsolsymb(cs_di_sparse const*, double*, cs_di_symbolic const*, double*, int*)", referenced from:
      g2o::LinearSolverCSparse<Eigen::Matrix<double, 7, 7, 0, 7, 7> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7, 0, 7, 7> > const&, double*, double*) in KeyFrameGraph.o
```

It seems that something wrong with suite-sparse build. I installed suite-sparse from homebrew (it downloads suite-sparse from http://faculty.cse.tamu.edu/davis/suitesparse.html) and I found the link for suite-sparse in README.md is broken.

Any suggestions?

felixendres, 2016-01-26:
Do you specify to link against libg2o_csparse_extension.so? If you use cmake, add `${G2O_SOLVER_CSPARSE_EXTENSION}` to the call of `target_link_libraries` for your target.

ningt, 2016-01-26:
I am using Xcode, and I think already linked against `libg2o_csparse_extension`.

[image]

felixendres, 2016-01-27:
The symbol (function) actually comes from g2o. It is in csparse_helpers.h/cpp. According to the makefile it should be in libg2o_csparse_extensions.so, but I've checked its symbol table (`readelf -s libg2o...`) and it doesn't show up. I don't know why. But I guess that it has nothing to do with SuiteSparse.

ningt, 2016-02-01:
Since it's not related to SuiteSparse, I updated the title of this issue. Do you have any suggestions about this? Thanks.

closed 2016-02-24 by ningt

hongyan90, 2017-03-01:
Hi, I met this question,too. If you have solved this question, could you give me some suggestions?

alibenD, 2018-09-12:
Does anyone has a solution for it now? I met this question, too.

songyongkang, 2019-10-26:
> Since it's not related to SuiteSparse, I updated the title of this issue. Do you have any suggestions about this? Thanks.

Does anyone has a solution for it now? I met this question, too.