RainerKuemmerle/g2o #98: reserved identifier violation

elfring, 2016-01-28:
I would like to point out that an identifier like "[`_DInvSchur`](https://github.com/RainerKuemmerle/g2o/blob/4b525d8437f027a3418617dd672b3d3cd5ff78fd/g2o/core/block_solver.h#L149)" [does eventually not fit](https://www.securecoding.cert.org/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier "Do not use identifiers that are reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

elfring, 2025-01-22:
:thought_balloon: Would you become interested to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/bugprone/reserved-identifier.html "Checking the application of reserved identifiers")?

RainerKuemmerle, 2025-01-24:
Thanks for the hint. I started actually a while back using it on my experimental branch: https://github.com/RainerKuemmerle/g2o/blob/pymem/.clang-tidy 
On that branch among a lot of other API breaking changes I changed the style of class member variables as given in the above linked clang-tidy configuration.

closed 2025-04-29 by github-actions[bot]