RainerKuemmerle/g2o #100: Where is the CHOLMOD solver?

Link5697, 2016-03-21:
I want to build G2O by cmake and I need G2O_SOLVER_CHOLMOD.lib .
Please tell me how to build G2O with CHOLMOD,thanks.

felixendres, 2016-03-21:
It should be built automatically and placed in the `lib` directory. If there is a problem please provide some more information, e.g., what OS are you using, what happens when you build g2o (terminal output) and what you would expect.

ursln, 2016-06-09:
I am using windows 7. I build g2o project with cmake, but i could not find cholmod project under build/g2o/solvers. Thus g2o cannot generate g20_solver_cholmod.lib 
Then I download SuiteSparse to refer cholmod files explicitly from g2o project, but i could not achieve. 
I guess there is a problem in cmake files. Is there a solution?
or if you have build folder for cholmod , could you please share it.

Planchee, 2016-09-15:
Trying to build LSD-SLAM (and new to programming) on Win8x64; Cmake 3.6.2 and a pre-built g2o. With Visual Studio 2015 (vc14). 
Same problem when configuring with Cmake: 

> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
G2O_SOLVER_CHOLMOD
    linked by target "sample_app" in directory C:/Users/xxx/LSD_SLAM/lsd_slam_noros-master/apps/slam
G2O_SOLVER_CHOLMOD_DEBUG
    linked by target "sample_app" in directory C:/Users/xxx/LSD_SLAM/lsd_slam_noros-master/apps/slam

> 

Not a lot to find.. tried rebuilding g2o from fresh (with EIGEN3) - but get even more errors then. Any idea?? 

There is no "issues" page for that partical package (LSD-SLAM for Windows so it's hard to ask there)

Much appreciated, thanks!

RainerKuemmerle, 2016-10-25:
You would need Cholmod compiled on Windows.
There is a project which provides cmake support for compiling it. See https://github.com/jlblancoc/suitesparse-metis-for-windows

closed 2016-10-25 by RainerKuemmerle

cusoh, 2019-11-25:
@RainerKuemmerle It seems CHOLMOD has many modules :   
```
    # CHOLMOD Library Modules, which appear in -lcholmod
    # Core       requires: none
    # Check      requires: Core
    # Cholesky   requires: Core, AMD, COLAMD. optional: Partition, Supernodal
    # MatrixOps  requires: Core
    # Modify     requires: Core
    # Partition  requires: Core, CCOLAMD, METIS.  optional: Cholesky
    # Supernodal requires: Core, BLAS, LAPACK
    #

```

Which modules should I build for use with g2o. Also, it would be helpful to know if there are version dependencies. 
Thankyou for your greatly helpful work.

Best Regards,
cusoh