RainerKuemmerle/g2o #103: Different results running the same optimizer

IvanMalin, 2016-04-01:
Hello!
I'm having a problem with a simple example below. What I have is a g2o optimizer saved with .save method (the saved file is attached as well). 
I load it in a loop, optimize and get different results. To be exactly, sometimes the optimizer works for 10 iterations and returns the correct results, sometimes it stops after two iterations with the incorrect one.
It looks like a memory/uninitialized variables problem, to reproduce it steadily, I had to add a call of a dummy function before optimization. 
I run this code compiled with Clang-3.6 under Ubuntu 14.04.

```
#include <g2o/core/sparse_optimizer.h>
#include <g2o/core/block_solver.h>
#include <g2o/solvers/csparse/linear_solver_csparse.h>
#include <g2o/core/optimization_algorithm_dogleg.h>
#include <g2o/types/slam3d/vertex_se3.h>

static void init(std::string str)
{

}

int main(int argc, char** argv)
{
    init("***************************************");

    for (int k = 0; k < 10; ++k)
    {
        g2o::SparseOptimizer optimizer;
        optimizer.load("g2oOptimizer_good.txt");
        g2o::LinearSolverCSparse<g2o::BlockSolverX::PoseMatrixType>* linearSolver = new g2o::LinearSolverCSparse<g2o::BlockSolverX::PoseMatrixType>();
        g2o::BlockSolverX* solver = new g2o::BlockSolverX(linearSolver);
        g2o::OptimizationAlgorithmDogleg* algo = new g2o::OptimizationAlgorithmDogleg(solver);
        optimizer.setVerbose(true);
        optimizer.setAlgorithm(algo);
        optimizer.initializeOptimization();
        optimizer.optimize(100);
    }

    return 0;
}
```
[g2oOptimizer_good.txt](https://github.com/RainerKuemmerle/g2o/files/200230/g2oOptimizer_good.txt)

IvanMalin, 2016-04-02:
Further investigation showed that difference seems to start in optimization_algorithm_dogleg.cpp.
After the code fragment below I have different `cache->w2n()` for the same edge. The edge has the same `_measurement` and `vertices()[1]->estimate()` both times, but values in cache differ.

```
      // apply the update and see what happens
      _optimizer->push();
      _optimizer->update(_hdl.data());
```

Any ideas what I'm missing?

IvanMalin, 2016-04-04:
Further investigation showed that in case everything works fine, we have the same cache for the vertex and for the edge connected to that vertex. Thus, when we do `_optimizer->update(_hdl.data())`, cache update for vertices is called, so the edge's cache is updated as well (they have the same cache).
When everything goes wrong, they have different caches, and I have no idea yet, why it happens this way.

RainerKuemmerle, 2016-04-04:
I will try to investigate this issue as soon as possible.

IvanMalin, 2016-04-04:
Thanks for the attention to the problem!
If it helps, I made some more work on this problem and managed to reproduce on even smaller example, without optimization etc. 

```
int main(int argc, char** argv)
{
    for (int i = 0; i < 10; ++i)
    {
        g2o::SparseOptimizer optimizer;
        std::cerr << "---- PREPARING ----" << std::endl;
        optimizer.load("/home/imalin/g2oOptimizer_good.txt");
    }
    return 0;
}
```
As I understand, when resolveCache for edges is called during optimizer loading, it should always find the cache of the connected vertex and assign. But there seems to be a memory corruption, so during two runs of the loop for the same edge the order of keys of `CacheContainer* container= v->cacheContainer();` in `OptimizableGraph::Edge::resolveCache` is different! 
Although valgrind showed no problems except for leaks.

kasra, 2016-04-04:
This might be irrelevant, but go to the g2o's build directory and see if you're using OpenMP.

`cat CMakeCache.txt | grep USE_OPENMP`

IvanMalin, 2016-04-05:
> This might be irrelevant, but go to g2o's build directory and see if you're using OpenMP.

Not really, `G2O_USE_OPENMP:BOOL=OFF`

IvanMalin, 2016-04-05:
I think I found the cause.
I guess in **cache.cpp**
```
if (_type < c._type)
      return true;
```
should be changed to
```
if (_type < c._type)
    return true;
else if (c._type < _type)
    return false;
```

closed 2016-04-12 by RainerKuemmerle