RainerKuemmerle/g2o #105: homebrew support

adrelino, 2016-04-23:
Hi,

I am writing a homebrew formula https://github.com/adrelino/homebrew-science/blob/master/g2o.rb  so one will be able to easily install g2o and all its dependencies on OSX in one line using:

```sh
brew install homebrew/science/g2o
```

Before I make my pull-request to homebrew-science, I need your help, because Homebrew guidelines require tagged versions of the software, so could you please create and push a tag to github? E.g:

```sh
git tag -a 1.0.0 -m 'first tagged version'
git push origin 1.0.0
```

This allows to to have stable zip's which don't change after each commit.
(This is why I can't just take https://github.com/RainerKuemmerle/g2o/archive/master.zip,
but need a tagged zip like in my fork of your repo: https://github.com/adrelino/g2o/archive/1.0.0.zip)

I would have made a pull request, but they don't support to attach tags.

Adrian

sjulier, 2016-04-23:
Don't know if it is of use to you, but this is one I had set up for just doing a version from the head branch. I knocked it together in about five minutes a couple of years ago so it might / might not be of any help:

require 'formula'

class G2o < Formula
  homepage 'http://openslam.org/g2o.html'
  url 'https://github.com/RainerKuemmerle/g2o.git', :branch => 'master'
  version '13.'

  depends_on 'cmake'
  depends_on 'eigen'
  depends_on 'suite-sparse'
  depends_on 'qt'
  depends_on 'libqglviewer'

  def options
    [
      ['--no_apps',"Don't build applications"],
      ['--deprecated', "Build deprecated types"],
      ['--examples', "Build examples"],
      ['--linked', "Build linked applications"],
      ['--csparse', "Build csparse"],
      ['--no_opengl',"Disable use opengl"],
      ['--use_openmp', "Use openmp"],
    ]
  end

  def install
    args = std_cmake_parameters.split

    if ARGV.include? '--no_apps'
      args << "-DG2O_BUILD_APPS:BOOL=OFF"
    end
    if ARGV.include? '--deprecated'
      args << "-DG2O_BUILD_DEPRECATED_TYPES:BOOL=ON"
    end
    if ARGV.include? '--examples'
      args << "-DG2O_BUILD_EXAMPLES:BOOL=OFF"
    end
    if ARGV.include? '--linked'
      args << "-DG2O_BUILD_LINKED_APPS:BOOL=ON"
    end
    if ARGV.include? '--csparse'
      args << "-DBUILD_CSPARSE:BOOL=ON"
    end
    if ARGV.include? '--no_opengl'
      args << "DG2O_USE_OPENGL:BOOL=OFF"
    end
    if ARGV.include? '--use_opengl'
      args << "DG2O_USE_OPENMP:BOOL=ON"
    end

    system "mkdir build"
    args << ".."
    Dir.chdir 'build' do
      system "cmake", *args
      system "make install"
    end
  end
end

adrelino, 2016-04-23:
Yes, I found that one
https://github.com/Homebrew/legacy-homebrew/issues/21869

but to get accepted homebrew/science, it has to be a stable version. This would allow to easily install g2o without tapping other unpopular repositories...

RainerKuemmerle, 2016-04-24:
I tagged the current master branch.
There is a build on travis also compiling for OSX. Let me know if I can modify it to better support OSX.

adrelino, 2016-04-25:
Thanks @RainerKuemmerle ,
I just made my pull request to homebrew-science with your tagged version 2016-04-24
#https://github.com/Homebrew/homebrew-science/pull/3584

adrelino, 2016-06-14:
Good news, you can finally

`brew install g2o`

Since there were some flaws in my original pull request Homebrew/homebrew-science#3584 regarding install and test targets (and also the description starting with an 'A'), my pull request did not get merged and I had not enough experience+time to fix the remaining issues. 

But Luckily, @dpo took the time to remove the remaining flaws so his pull request   Homebrew/homebrew-science#3736 was accepted into homebrew-science 3 days ago.

So here is how the valid homebrew formula for g2o looks like: https://github.com/Homebrew/homebrew-science/commit/6790179955b2d78cc435c861d28823bc72c251eb

closed 2016-06-14 by adrelino