RainerKuemmerle/g2o #106: How to put in constraints like GPS (or any other global positioning) for slam

tinokl, 2016-04-27:
I am currently working on an outdoor slam solution with pose graphs and the results are good in general but unfortunately the graph drifts too much for building a complete consistent map.

The graph is build up by connecting the poses either with odometry or scan matching measurements (if available). The robot has an gps but now i am not sure how to put constraints with this new measurement source.

I thought about connecting each pose with the fixed one (the first) and add the computed distance of the gps readings? Therefore i would not need to transform the gps fix to the map frame.

Is there any example or advise how someone can feed global positions (with uncertainty) into such an pose graph?

thanks

mojovski, 2016-04-27:
would be great to have a discussion and of course a solution for the GPS use case! :+1:

felixendres, 2016-05-02:
You could write a Unary Edge class of which the measurement is the GPS coordinates and  that connects to the pose vertex type of your choice. I'd think it would be similar to [EdgeSE2Prior](http://docs.ros.org/fuerte/api/re_vision/html/classg2o_1_1EdgeSE2Prior.html), but without the rotation.

tinokl, 2016-05-09:
thanks for those hints, i tried several implementations by now. 
The first i tried was the unary edge but here i got stuck because of the difficult (at least for me) transformation from gps fix to the map frame. (The poses and the mapping is within map frame but i do not have a correct transformation from gps to the map)
Then i looked at the gps example and i understand how to use the graph with multiple measurements for one single pose estimate but again no glue how to add different "frame measurements" 

In my current solution i am using a custom gps edge, which basically connects two SE2 poses and computes the error term by computing the gps-distance minus the "map-frame" distance of those poses. It connects each 50th pose with the fixed one and inserts some additional edges with previous poses. It worked quite well on my main dataset and still performs in realtime.

now i need to test it a little further and what it does on gps dropouts..

felixendres, 2016-05-09:
Since @mojovski wanted a discussion, I'll share my thoughts concerning the unary edges: As long as you have only relative measurements except for the GPS, you could just drop the common assumption that the first graph node is fixed at the origin, and let that pose be found by the optimizer.

tinokl, 2016-06-06:
This sounds like an interesting approach but would that also deal with unmodified GPS data (latitute, longitude)? When i have time i'll try that out as well. Concerning my solution i got a lot of trouble with really noisy measurements, now i am trying to remove some unlogical edges but is there any other validation procedure in g2o?

RainerKuemmerle, 2016-06-06:
You may have a look at robust_kernel.h and the set of specific implementations provided inside g2o.
It allows you to reduce the influence of outliers.

closed 2016-06-22 by RainerKuemmerle

ethcamera, 2016-06-24:
@tinokl. Would you be possible to share your current code implementation of inserting GPS as constraints. Do you minimize the distance error between two GPS poses and two map poses. Thank you.

tinokl, 2016-07-14:
yes of course its possible, just dont know how to upload it right.. but i'll post the main idea just here for everyone. Basically i compute the distance between two poses (from gps measurements and of the poses) and then i minimize the error which is the distance difference.

This for example is the code for gps edges cpp:

`
#include <pose_graph_mapping/custom_types/gps_edge.h>
#include <pose_graph_mapping/custom_types/robot_gps.h>

#include <iostream>

using namespace std;
using namespace Eigen;

double haversine(double lat1, double long1, double lat2, double long2)
{
    double dlong = (long2 - long1) * d2r;
    double dlat = (lat2 - lat1) * d2r;
    double a = pow(sin(dlat/2.0), 2) + cos(lat1) * cos(lat2) * pow(sin(dlong/2.0), 2);
    double c = 2 * atan2(sqrt(a), sqrt(1-a));
    double d = EARTH_RADIUS * c;

    return d;
}

namespace g2o {

    GPSEdge::GPSEdge()
    {
    }

    void GPSEdge::computeError()
    {
      const VertexSE2* v1 = static_cast<const VertexSE2*>(_vertices[0]);
      const VertexSE2* v2 = static_cast<const VertexSE2*>(_vertices[1]);
      Vector2d t = v1->estimate().translation()-v2->estimate().translation();

      double current_dist = t.norm();

      _error[0] = current_dist - _measurement;
    }

    void GPSEdge::setInformation(const double& i)
    {
      _information[0] = i;
    }

    void GPSEdge::setMeasurement(Eigen::Vector2d gps1, Eigen::Vector2d gps2)
    {
      // all in radians
      double lat1 = gps1[0] * d2r;
      double lat2 = gps2[0] * d2r;
      double lon1 = gps1[1] * d2r;
      double lon2 = gps2[1] * d2r;

      _measurement = haversine(lat1, lon1, lat2, lon2) * r2d; // get meter
    }
} // end namespace
`

the corresponding h file:
`
#ifndef G2O_GPS_EDGE_H
#define G2O_GPS_EDGE_H

#include "g2o_custom_types_data_api.h"
#include <g2o/core/base_unary_edge.h>
#include <g2o/types/slam2d/vertex_se2.h>

#include "g2o/core/base_binary_edge.h"

#include <Eigen/Core>

#define d2r (M_PI / 180.0)
#define r2d (180.0 / M_PI)
#define EARTH_RADIUS 6371000 // R is earth’s radius (mean radius = 6,371km)

namespace g2o {

class G2O_CUSTOM_TYPES_DATA_API GPSEdge : public BaseBinaryEdge<1, double, VertexSE2, VertexSE2>
{
  public:
    EIGEN_MAKE_ALIGNED_OPERATOR_NEW
    GPSEdge();

    void computeError();

    virtual bool read(std::istream& /*is*/)
    {
      return false;
    }

    virtual bool write(std::ostream& /*os*/) const
    {
      return false;
    }

    virtual void setInformation(const double& i);
    virtual void setMeasurement(Eigen::Vector2d gps1, Eigen::Vector2d gps2);
  };
} // end namespace

#endif
`

These edges can just be added to the graph with as normal edges, hope this helps ;)

sjulier, 2016-07-14:
I'm not sure why you are turning your GPS measurement, which provides full 3D position information, into just a 1D Euclidean distance. Isn't this effectively throwing quite a lot of information away?

We have used the GPS => UTM conversion from GeographicLib:

http://geographiclib.sourceforge.net/

This converts the GPS data into UTM (local east-north-up coordinate frame). Also, if you check the references, you can see that quite a lot of the maths can get pretty nasty pretty quickly.

Note this is only correct if the distance between the vehicle and the GPS is "small" (because of the spherical shape of the Earth). Small is, from what I recall, the order of 10s of kms.

ethcamera, 2016-07-15:
Thank you tinokl,  could you share the some pseudo which add the gps as edges, and init the nodes, subsequently call the solver. Thank you alot.

tinokl, 2016-07-19:
@sjulier you are right, we are throwing a lot of information away but the reasons why we did that are pretty simple. we could not get the gps bearing in a correct relation to the pose graph. What we tried was: "navsat_transform_node" from [http://wiki.ros.org/robot_localization](http://wiki.ros.org/robot_localization). The bearing of the start or initial pose and if this relation does not fit very very good the graph will be unpleasant deformed.

Then we thought about just skipping the part of any transformation and how to get data right in and here we are^^

tinokl, 2016-07-19:
for completeness here is a simple code snippet how we added gps edges:
everything else (optimizing, creating nodes, etc. ) is the same as normal graph slam ;)

```
Eigen::Vector2d current_pose_gps(current_pose->getLat(), current_pose->getLon());
Eigen::Vector2d first_pose_gps(first_pose->getLat(), first_pose->getLon());

GPSEdge* goe = new GPSEdge();
goe->vertices()[0] = current_pose;
goe->vertices()[1] = first_pose;
goe->setInformation(gpsinf);

goe->setMeasurement(current_pose_gps, first_pose_gps);
graph->addEdge(goe);
```

dannyway03, 2016-11-14:
Hi @tinokl ,
i recently got into g2o, and curious about integrating gps data. 
sorry, but maybe i'm missing something. 
You say: "In my current solution i am using a custom gps edge, which basically connects two SE2 poses and computes the error term by computing the gps-distance minus the "map-frame" distance of those poses." , but In the snipped you posted i can only see the gps-distance term..
Maybe there are other relevant parts of code? would you mind to release a minimal working example?
that would be great!

thanks,
-danny

AlexandrGraschenkov, 2018-10-24:
Hi everyone
We came to the same idea with scale fix. But we have more degree of freedom. We use SIM3 for optimization, and try to integrate GPS into SIM3. To improve quality we manually calculate scale factor for every point, and after this run optimization with `EdgeSim3`, `EdgeSim3ProjectXYZ` and custom GPS edge. Unfortunately it works not good.
We test GPS edges in SE3 and it works great.
Does anybody have success with GPS edges in SIM3?