RainerKuemmerle/g2o #108: Conversion of unnormalized quaternion to rotation matrix

ibtaylor, 2016-04-30:
In `EdgeSE3::read`, it looks like it sets the measurement with the result of `fromVectorQT`.
`fromVectorQT` creates a quaternion and converts it to a rotation matrix without first normalizing it.
It does not seem like the input should be relied upon to be in normalized form due to the output precision of floating point values (EdgeSE3::write) or the input comes from another source other than g2o.

If this is true, then, less importantly `VertexSE3` would also have the same issue.

Eigen documentation (https://eigen.tuxfamily.org/dox/classEigen_1_1Quaternion.html) has a warning about usage of Quaternion for rotation when not in normalized form.

Is this a legitimate concern?

ibtaylor, 2016-05-03:
Sorry, looks like this is done in `EdgeSE3::read` right before `setMeasurement`

closed 2016-05-03 by ibtaylor

ibtaylor, 2016-05-03:
What about `setMeasurementData` in `slam3d/edge_se3.h`? Though, it doesn't seem like anything ever calls those functions.

JulioPlaced, 2023-11-29:
Has this been updated/fixed at any point since this topic was opened?