RainerKuemmerle/g2o #109: How to use setMarginalized() ?

jingpang, 2016-05-08:
Hello, I'm a bit confused by the function setMarginalized().

How and when should i call this function?

Thanks for any help !

zhangxaochen, 2017-01-04:
Hi @jingpang have u figured it out?
I'm also confused though the doc string of `setMarginalized` says "**//! true => this node should be marginalized out during the optimization**".
I tried to comment that statement, while getting the following error:
```
Assertion failed: (!(RowsAtCompileTime!=Dynamic) || (nbRows==RowsAtCompileTime)) && (!(ColsAtCompileTime!=Dynamic) || (nbCols==ColsAtCompileTime)) && (!(RowsAtCompileTime==Dynamic && MaxRowsAtCompileTime!=Dynamic) || (nbRows<=MaxRowsAtCompileTime)) && (!(ColsAtCompileTime==Dynamic && MaxColsAtCompileTime!=Dynamic) || (nbCols<=MaxColsAtCompileTime)) && nbRows>=0 && nbCols>=0 && "Invalid sizes when resizing a matrix or array.", file d:\sdk\eigen3210\eigen\src/Core/PlainObjectBase.h, line 241
```
It seems that vertices have to be **setMarginalized**, why? 
Are they of no use if being marginalized out? or what are the specific usage of the marginalized vertices?

jingpang, 2017-01-04:
As i know in bundle adjustment, the points are setMarginalized, but poses are not. Implementation in ORBSLAM may be a good example of the usage.

closed 2017-01-04 by jingpang

felixendres, 2017-01-11:
@zhangxaochen, I think your problem might be the template parameters of the BlockSolver used by your LinearSolver. They may need to be different depending on the used vertices. I use
`LinearSolverCSparse<BlockSolver_6_3::PoseMatrixType>();` for a graph containing features and

    typedef BlockSolver< BlockSolverTraits<6, 1> > BlockSolver_6_1;
    auto* lin_slvr = new LinearSolverCSparse<BlockSolver_6_1::PoseMatrixType>();

for a pose graph. If you use `BlockSolverX` instead, you should be fine in any case, at the cost of a minor (probably negligible)performance penalty.

MinjieLin, 2017-11-24:
@jingpang   do you figure out why  the features needed to be "setmarginalized"  , Thanks

xushangnjlh, 2018-01-10:
I think this may be used for Shur Complement for the sparse matrix decomposition, to accelerating solve Hx = b problem. In visual SLAM, the mappoints always need setMarginalize() to tremendously decrease the dimension of H.

mojovski, 2018-06-03:
check the Thruns book "probabilistic robotics" (chapter1). There, it is clearly explained, why and how to "marginalize" some variables out of the optimization problems.

ns15417, 2020-12-17:
I found a very clear note about how to use setMarginalized() in g2o, it takes ORBSLAM as an example. I post it here so you can have reference while it is in Chinese: https://zhuanlan.zhihu.com/p/37843131