RainerKuemmerle/g2o #111: Compilation on Fedora 24

MalcolmMielle, 2016-06-06:
Hello,

Eigen didn't link on Fedora 24 and I had to change the line 251 in the CMakeList.txt to : 

```
SET(EIGEN3_INCLUDE_DIR ${G2O_EIGEN3_INCLUDE})
FIND_PACKAGE(Eigen3)
IF(EIGEN3_FOUND)
  #Here I'm commenting out the CACHE PATH part of this line
  SET(G2O_EIGEN3_INCLUDE ${EIGEN3_INCLUDE_DIR} )#CACHE PATH "Directory of Eigen3")
  message (STATUS "EIgen FOUND 1")
ELSE(EIGEN3_FOUND)
  message (STATUS "EIgen FOUND 2")
  SET(G2O_EIGEN3_INCLUDE "" CACHE PATH "Directory of Eigen3")
ENDIF(EIGEN3_FOUND)
```

for the complation to work.

I was just wondering why :) ?

elmuz, 2016-06-06:
I think it is a problem of gcc > 6.x. It's reported on Eigen bugzilla.
http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1221
If the problem is the same you can overcome it by taking the latest sources
from bitbucket https://bitbucket.org/eigen/eigen/
Hope it helps

Alessio

elmuz, 2016-06-06:
Oops I didn't read it carefully... it's not the same issue.

MalcolmMielle, 2016-06-06:
The compilation was working with Fedora 23 that _I think_ had gcc > 6 :)

MalcolmMielle, 2016-06-06:
Correction, the problem also exist on Fedora 23

commit c986d956 (2016-06-22): removing G2O_EIGEN3_INCLUDE

- fix #111
closed 2016-06-22 by RainerKuemmerle