RainerKuemmerle/g2o #120: Documentation

nlw0, 2016-08-24:
It would be great if the actual documentation from g2o was available somewhere. It is only possible to easily find the ICRA article here and on openslam.org...

felixendres, 2016-08-29:
There is a [pdf document](https://github.com/RainerKuemmerle/g2o/blob/master/doc/g2o.pdf) with a detailed description of how the library is structured and how to use/extend it in the folder "doc". The API documentation you can generate yourself as described in the readme in [doc/doxygen](https://github.com/RainerKuemmerle/g2o/tree/master/doc/doxygen) 

Is that what you mean by "actual documentation"?

nlw0, 2016-08-29:
That's the one! Just wanted to suggest leaving the link explicitly at the README and maybe at the openslam.org page.

NadimDeeb, 2023-08-31:
The pdf document is not available anymore, I am looking for a api documentation since a function from the g2o library is causing me trouble and I would like to see how it behaves. It's the make_function.