RainerKuemmerle/g2o #123: Vertex Color in vizualization

MalcolmMielle, 2016-08-29:
Hello all,

I searched in the code but couldn't really find how the vertex colours are handled. Let's say I'm creating a new vertex that inheriting from an existing vertex type in the code. The only thing I would like to change is the colour of said vertex type in g2o_viewer. How can I do this ?

Thanks,

felixendres, 2016-08-29:
You need to implement a class derived from DrawAction. Check out the class VertexPointXYZDrawAction as implemented for VertexPointXYZ: [header](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/types/slam3d/vertex_pointxyz.h) and [cpp](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/types/slam3d/vertex_pointxyz.cpp)

As far as I see, the point is drawn in the cpp at line 90 according to the definition of the global vertex color in [opengl_primitives.h](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/stuff/opengl_primitives.h)

MalcolmMielle, 2016-08-30:
Ok thanks :). I'm going to look it up

MalcolmMielle, 2016-08-30:
So I created a class `EdgeSE2Prior` as such 

```
/**
   * \brief Prior for a two D pose
   */
  class G2O_TYPES_SLAM2D_API EdgeSE2Prior : public EdgeSE2
  {
    public:
//       EIGEN_MAKE_ALIGNED_OPERATOR_NEW;
      EdgeSE2Prior();

  };

  
  #ifdef G2O_HAVE_OPENGL
  class G2O_TYPES_SLAM2D_API EdgeSE2PriorDrawAction: public DrawAction{
  public:
    EdgeSE2PriorDrawAction();
    virtual HyperGraphElementAction* operator()(HyperGraph::HyperGraphElement* element, 
            HyperGraphElementAction::Parameters* params_);
  protected:
    virtual bool refreshPropertyPtrs(HyperGraphElementAction::Parameters* params_);
    FloatProperty* _triangleX, *_triangleY;

  };
	#endif
```

I obtain then a g2o file like this when I run tutorial_slam2d :

```
PARAMS_SE2OFFSET 0 0.2 0.1 -0.1 
VERTEX_SE2 0 0 0 0
FIX 0
VERTEX_SE2_PRIOR 16 1.01324 0.00252476 -0.057989
VERTEX_SE2 25 1.96914 -0.0563184 -0.058688
VERTEX_SE2 31 3.01631 -0.123993 -0.0270203
VERTEX_SE2 37 4.00952 -0.153144 -0.0817478
VERTEX_SE2 43 5.0036 -0.222595 1.53324
VERTEX_SE2 47 5.04214 0.702837 1.51069
VERTEX_SE2 54 5.08896 1.68908 1.47979
VERTEX_SE2 56 5.1745 2.66687 1.50676
VERTEX_SE2 64 5.24444 3.6512 1.51244
EDGE_PRIOR_SE2 0 16 1.01324 0.00252476 -0.057989 400 0 0 10000 0 820.702
EDGE_PRIOR_SE2 16 25 0.957703 -0.00334367 -0.00069902 400 0 0 10000 0 820.702
EDGE_PRIOR_SE2 25 31 1.04933 -0.00613693 0.0316677 400 0 0 10000 0 820.702
EDGE_PRIOR_SE2 31 37 0.993634 -0.00230763 -0.0547275 400 0 0 10000 0 820.702
EDGE_PRIOR_SE2 37 43 0.99643 0.0119546 1.61499 400 0 0 10000 0 820.702
EDGE_PRIOR_SE2 43 47 0.926227 -0.00376753 -0.0225514 400 0 0 10000 0 820.702
EDGE_PRIOR_SE2 47 54 0.987277 0.0125182 -0.0308977 400 0 0 10000 0 820.702
EDGE_PRIOR_SE2 54 56 0.981513 0.0036778 0.0269667 400 0 0 10000 0 820.702
EDGE_PRIOR_SE2 56 64 0.986786 -0.00680919 0.00568871 400 0 0 10000 0 820.702
```

But when I run it on `./g2o_viewer` on this file I have this : 

[image]

When I used the normal `EdgeSE2` I have this : 

[image]

This is my opengl primitive : 

```
#define POSE_VERTEX_COLOR 0.8f,0.8f,0.8f
#define POSE_PARAMETER_COLOR 0.5f,0.5f,0.8f
#define POSE_EDGE_COLOR 0.4f,0.4f,0.7f
#define POSE_EDGE_GHOST_COLOR 0.4f,0.4f,0.7f

#define LANDMARK_VERTEX_COLOR 0.8f,0.5f,0.3f
#define LANDMARK_EDGE_COLOR   0.7f,0.4f,0.2f
#define LANDMARK_EDGE_GHOST_COLOR   0.7f,0.4f,0.2f

#define POSE_VERTEX_PRIOR_COLOR 0.8f,0.8f,0.8f
#define POSE_EDGE_PRIOR_COLOR 0.7f,0.7f,0.7f
#define POSE_EDGE_PRIOR_GHOST_COLOR 0.7f,0.7f,0.7f
```

I can't fiure out why `g2o_viewer` don't run through `EdgeSE2PriorDrawAction`. Any help on this will be greatly appreciated :)

felixendres, 2016-08-31:
I forgot to mention that youneed to register the new Edge and the new DrawAction in g2o/types/slam2d/types_slam2d.cpp, sorry.

MalcolmMielle, 2016-08-31:
THank you so much for you help. It works now :). Weirdly enough, my new Vertex class was registered but not the draw action.

closed 2016-08-31 by MalcolmMielle