RainerKuemmerle/g2o #125: Cholesky failure

elmuz, 2016-09-05:
Hi guys what does the following error message refers to: "Cholesky failure, writing debug.txt (Hessian loadable by Octave)"?
I mean, I know what a Cholesky decomposition is, but what's wrong with my hypergraph so that it throws this exception?
At first it was working ok, then I tuned some edges information matrices and suddenly it doesn't like the it anymore. Any ideas on that?
Thank you

izhengfan, 2016-09-05:
You may try reviewing whether something wrong with the float point computation and some Nan number existing in your information matrix, Jacobian or somewhere else.

elmuz, 2016-09-06:
Some info:
- Keeping standard information matrix it never throws any exception (so that must be the reason).
- I checked that all edges have SPD information matrix, both using integrated g2o function and Eigen's.
- In the debug.txt, CTRL+F for "NaN" doesn't return any occurrance. Is there any better way to check numerical problems?
Thank you

felixendres, 2016-09-13:
In my experience, this also happens if the statistical errors get large, e.g. when the optimizer diverges. The statistical error also gets large if you use large values in the information matrix.

elmuz, 2016-09-13:
Thank you @felixendres that's a nice idea.

RainerKuemmerle, 2016-10-25:
Closing this for now. Please provide a specific example if the error occurs again.

closed 2016-10-25 by RainerKuemmerle

dingshuoshuo, 2022-03-08:
My problem is the float data error in the info matrix