RainerKuemmerle/g2o #131: Another Cholesky failure case

ShaowuYang, 2016-10-24:
Dear Rainer, Dear colleagues,
I met another Cholesky failure case, and haven't solved this issue yet:
> int g2o::csparse_extension::cs_cholsolsymb(const cs*, double*, const css*, double*, int*): No valid input!
> Cholesky failure, writing debug.txt (Hessian loadable by Octave)
I have checked that there was at least one fixed vertex, and all information matrices of edges are PSD. Below is an example graph in my implementation. Could you find any hint for me?

VERTEX_SE3:QUAT 0 0 0 0 0 0 0 1 
FIX 0
VERTEX_SE3:QUAT 1 0.0347799 0.144352 1.25976 0.554364 -0.474954 0.484748 -0.481787 
FIX 1
VERTEX_SE3:QUAT 2 0.0195101 0.159557 1.47892 0.542343 -0.479582 0.511138 -0.463252 
FIX 2
VERTEX_SE3:QUAT 3 -0.12918 0.22604 1.91578 0.527072 -0.493718 0.52321 -0.452425 
VERTEX_SE3:QUAT 4 -0.108082 -0.032968 1.56173 0.556677 -0.467679 0.508721 -0.461075 
VERTEX_SE3:QUAT 5 -0.307843 -0.00510025 1.54579 0.5315 -0.504853 0.514999 -0.444305 
VERTEX_SE3:QUAT 6 -0.238333 0.258375 1.51427 0.56705 -0.452506 0.511054 -0.460995 
VERTEX_SE3:QUAT 7 -0.242008 0.394232 1.67181 0.554387 -0.468675 0.482178 -0.490411 
VERTEX_SE3:QUAT 8 -0.204792 0.617972 1.67376 0.553827 -0.486654 0.494627 -0.460203 
VERTEX_SE3:QUAT 9 -0.129992 0.791813 1.54349 0.551915 -0.464187 0.512288 -0.466349 
VERTEX_SE3:QUAT 10 -0.157345 1.008 1.48391 0.537896 -0.47558 0.501338 -0.482857 
VERTEX_SE3:QUAT 11 -0.224181 1.23026 1.47017 0.541251 -0.481511 0.497527 -0.477138 
VERTEX_SE3:QUAT 12 -0.294008 1.44916 1.54254 0.545361 -0.476836 0.503579 -0.470762 
VERTEX_SE3:QUAT 13 -0.335424 1.65394 1.52685 0.517904 -0.49688 0.500796 -0.483828 
VERTEX_SE3:QUAT 14 -0.576661 1.70307 1.53628 0.5394 -0.476916 0.502966 -0.478147 
VERTEX_SE3:QUAT 15 -0.814478 1.74373 1.55963 -0.54005 0.466101 -0.489034 0.501938 
VERTEX_SE3:QUAT 16 -1.04534 1.78358 1.61239 -0.558504 0.454012 -0.478212 0.503248 
VERTEX_SE3:QUAT 17 -1.24642 1.8268 1.6658 0.53732 -0.478585 0.501846 -0.479994 
VERTEX_SE3:QUAT 18 -1.44895 1.86633 1.63462 0.505472 -0.511309 0.527133 -0.452982 
VERTEX_SE3:QUAT 19 -1.66587 1.90404 1.5293 0.498698 -0.513705 0.53252 -0.451475 
VERTEX_SE3:QUAT 20 -1.91249 1.9338 1.4926 0.499512 -0.515544 0.528958 -0.452664 
VERTEX_SE3:QUAT 21 -2.16403 1.93463 1.51049 0.500337 -0.509228 0.53485 -0.451979 
VERTEX_SE3:QUAT 22 -2.42084 1.92845 1.56372 0.500242 -0.510297 0.534355 -0.451463 
VERTEX_SE3:QUAT 23 -2.64178 1.92336 1.63852 0.503557 -0.51406 0.528606 -0.450276 
VERTEX_SE3:QUAT 24 -2.86306 1.90977 1.74049 0.502968 -0.5117 0.527998 -0.454317 
VERTEX_SE3:QUAT 25 -3.26374 1.89593 1.95771 -0.511197 0.51812 -0.518053 0.449278 
VERTEX_SE3:QUAT 26 -3.48925 1.87917 1.8381 0.511329 -0.503799 0.526956 -0.455023 
VERTEX_SE3:QUAT 27 -3.62919 1.86593 1.63272 0.512508 -0.506262 0.51554 -0.463953 
VERTEX_SE3:QUAT 28 -3.84472 1.82014 1.56805 0.520884 -0.495433 0.534875 -0.443998 
VERTEX_SE3:QUAT 29 -4.06324 1.79421 1.59439 0.503355 -0.516695 0.523657 -0.453258 
VERTEX_SE3:QUAT 30 -4.25077 1.75003 1.73974 0.506039 -0.511503 0.526875 -0.452429 
VERTEX_SE3:QUAT 31 -4.30721 1.80362 1.96046 0.495803 -0.505751 0.541157 -0.45337 
VERTEX_SE3:QUAT 32 -4.46526 1.8544 1.78689 0.495581 -0.508805 0.536145 -0.456141 
VERTEX_SE3:QUAT 33 -4.70213 1.80976 1.75309 0.503718 -0.502431 0.530962 -0.460337 
VERTEX_SE3:QUAT 34 -4.86644 1.73769 1.89998 0.498106 -0.512576 0.531598 -0.454488 
VERTEX_SE3:QUAT 35 -5.01594 1.59881 1.95032 0.505926 -0.503475 0.54926 -0.434587 
VERTEX_SE3:QUAT 36 -5.1484 1.42192 1.8507 0.496539 -0.513573 0.537641 -0.447922 
VERTEX_SE3:QUAT 37 -5.20019 1.19447 1.83019 0.501635 -0.504598 0.534584 -0.45603 
VERTEX_SE3:QUAT 38 -5.15524 0.97316 1.91656 0.502192 -0.506661 0.513512 -0.476869 
VERTEX_SE3:QUAT 39 -5.06465 0.810731 2.04452 0.503668 -0.493679 0.535729 -0.464321 
VERTEX_SE3:QUAT 40 -5.06984 0.583918 2.07857 0.499284 -0.490134 0.553598 -0.451678 
VERTEX_SE3:QUAT 41 -5.15006 0.439445 1.91313 0.493886 -0.497891 0.542299 -0.462703 
VERTEX_SE3:QUAT 42 -5.26557 0.315775 1.76622 0.508086 -0.511248 0.535778 -0.439791 
VERTEX_SE3:QUAT 43 -5.29987 0.100111 1.70949 0.506215 -0.491156 0.545421 -0.4528 
VERTEX_SE3:QUAT 44 -5.26919 -0.105595 1.81428 0.512074 -0.496299 0.519431 -0.470807 
VERTEX_SE3:QUAT 45 -5.21736 -0.319301 1.86286 0.52732 -0.494375 0.525241 -0.449054 
VERTEX_SE3:QUAT 46 -5.00306 -0.431339 1.84339 0.529246 -0.48789 0.526566 -0.452316 
VERTEX_SE3:QUAT 47 -4.76494 -0.468403 1.81797 0.520591 -0.497185 0.521957 -0.457551 
VERTEX_SE3:QUAT 48 -4.52505 -0.482985 1.80646 0.524058 -0.49387 0.527635 -0.450618 
VERTEX_SE3:QUAT 49 -4.29547 -0.481209 1.82279 0.512389 -0.508118 0.509683 -0.468506 
VERTEX_SE3:QUAT 50 -4.0718 -0.497766 1.86283 0.53172 -0.488658 0.522586 -0.453201 
VERTEX_SE3:QUAT 51 -3.84007 -0.489725 1.92585 0.524012 -0.497254 0.516571 -0.459678 
VERTEX_SE3:QUAT 52 -3.61005 -0.481148 1.99571 0.51547 -0.510252 0.508949 -0.463578 
VERTEX_SE3:QUAT 53 -3.38359 -0.50359 2.06181 0.51737 -0.505394 0.517913 -0.456805 
VERTEX_SE3:QUAT 54 -2.94041 -0.578375 2.00229 0.526261 -0.487618 0.541231 -0.438573 
VERTEX_SE3:QUAT 55 -2.73867 -0.567817 1.87461 0.514306 -0.50255 0.524602 -0.45577 
VERTEX_SE3:QUAT 56 -2.5291 -0.563812 1.74104 0.515991 -0.504079 0.523477 -0.453464 
VERTEX_SE3:QUAT 57 -2.28713 -0.552796 1.6898 0.510089 -0.506408 0.524229 -0.456666 
VERTEX_SE3:QUAT 58 -2.05132 -0.525342 1.74738 0.520424 -0.497772 0.520019 -0.459307 
VERTEX_SE3:QUAT 59 -1.83915 -0.55352 1.82009 0.513504 -0.501603 0.507028 -0.477105 
VERTEX_SE3:QUAT 60 -1.60985 -0.62686 1.81429 0.523807 -0.500205 0.513271 -0.460407 
[... 51 lines cut ...]
EDGE_SE3:QUAT 43 44 0.212421 -0.081681 0.0468558 -0.00914426 -0.0226682 0.0213259 0.999474 0.18522 0 0 0 0 0 0.18522 0 0 0 0 0.18522 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 44 45 0.214147 -0.0357105 0.0583484 -0.0145556 0.00510922 -0.0226721 0.999624 0.197868 0 0 0 0 0 0.197868 0 0 0 0 0.197868 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 45 46 0.0996412 0.0213908 0.22037 0.00472171 -0.00477517 -0.00334489 0.999972 0.169638 0 0 0 0 0 0.169638 0 0 0 0 0.169638 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 46 47 0.0239186 0.0197545 0.23977 -0.000474279 0.00385383 0.0134194 0.999902 0.17108 0 0 0 0 0 0.17108 0 0 0 0 0.17108 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 47 48 0.00424809 0.00323517 0.240817 -0.000765875 0.00306966 -0.00959413 0.999949 0.17235 0 0 0 0 0 0.17235 0 0 0 0 0.17235 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 48 49 -0.0104024 -0.0262138 0.228382 -0.00177276 -0.00573267 0.0307838 0.999508 0.188844 0 0 0 0 0 0.188844 0 0 0 0 0.188844 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 49 50 0.0105005 -0.0487877 0.222218 -0.000434671 -0.0044837 -0.0335802 0.999426 0.192784 0 0 0 0 0 0.192784 0 0 0 0 0.192784 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 50 51 -0.00595003 -0.0740368 0.229045 -0.000647807 0.00161738 0.0145622 0.999892 0.172477 0 0 0 0 0 0.172477 0 0 0 0 0.172477 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 51 52 -0.0089162 -0.0809574 0.225642 -0.00431987 0.00459059 0.0166004 0.999842 0.173768 0 0 0 0 0 0.173768 0 0 0 0 0.173768 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 52 53 0.0174224 -0.0776064 0.223654 0.00263251 0.00492182 -0.0109725 0.999924 0.17747 0 0 0 0 0 0.17747 0 0 0 0 0.17747 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 53 54 0.0500213 0.0424502 0.450113 0.00736529 0.00909409 -0.033767 0.999361 0.0483312 0 0 0 0 0 0.0483312 0 0 0 0 0.0483312 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 52 53 0.0181057 -0.0782822 0.223309 0.00266738 0.00484376 -0.0107964 0.999926 0.177548 0 0 0 0 0 0.177548 0 0 0 0 0.177548 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 54 55 -0.040566 0.11718 0.205274 -0.00207453 -0.00395792 0.0303817 0.999528 0.17387 0 0 0 0 0 0.17387 0 0 0 0 0.17387 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 55 56 -0.0265136 0.124474 0.213329 -0.0032844 0.000387163 -0.000737949 0.999994 0.162059 0 0 0 0 0 0.162059 0 0 0 0 0.162059 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 56 57 -0.028564 0.0401298 0.24331 0.00332154 0.00277262 0.00563234 0.999975 0.162269 0 0 0 0 0 0.162269 0 0 0 0 0.162269 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 57 58 -0.0380318 -0.0678884 0.231503 -0.00102047 -0.0129146 -0.0062955 0.999896 0.167648 0 0 0 0 0 0.167648 0 0 0 0 0.167648 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 58 59 0.026366 -0.0784356 0.210338 0.00395878 -0.0102486 0.0206691 0.999726 0.195736 0 0 0 0 0 0.195736 0 0 0 0 0.195736 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 59 60 0.0691901 0.0015148 0.230533 -0.00960383 0.00563884 -0.0172377 0.999789 0.172608 0 0 0 0 0 0.172608 0 0 0 0 0.172608 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 60 61 0.0595093 0.0318242 0.234299 -0.0035611 0.00923235 -0.000915471 0.999951 0.0420521 0 0 0 0 0 0.0420521 0 0 0 0 0.0420521 0 0 0 1 0 0 1 0 1 
[... cut ...]
EDGE_SE3:QUAT 63 64 -0.0331386 0.0403719 0.23864 -0.00834322 -0.00165614 0.00890147 0.999924 0.167569 0 0 0 0 0 0.167569 0 0 0 0 0.167569 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 64 65 -0.131201 -0.00530564 0.197346 0.00859558 -0.00589136 0.0012805 0.999945 0.177976 0 0 0 0 0 0.177976 0 0 0 0 0.177976 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 2 66 0.0454226 -0.475824 -0.238753 -0.00656401 0.09282 -0.0130228 0.995576 0.024658 0 0 0 0 0 0.024658 0 0 0 0 0.024658 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 65 66 -0.181687 -0.120611 0.0942272 0.00899224 0.0106895 -0.00626574 0.999883 0.177192 0 0 0 0 0 0.177192 0 0 0 0 0.177192 0 0 0 1 0 0 1 0 1

ShaowuYang, 2016-10-24:
Also, I have confirmed that the graph is connected.

ShaowuYang, 2016-10-25:
I even got this error with such simple graph (.g2o file):

> VERTEX_SE3:QUAT 0 0 0 0 0 0 0 1 
FIX 0
VERTEX_SE3:QUAT 1 0.0347799 0.144352 1.25976 0.554364 -0.474954 0.484748 -0.481787 
VERTEX_SE3:QUAT 2 0.0216476 0.160348 1.4788 0.542794 -0.479836 0.510896 -0.462728 
VERTEX_SE3:QUAT 3 -0.0241951 0.166449 1.70919 0.527232 -0.491554 0.525779 -0.451617 
VERTEX_SE3:QUAT 4 -0.12053 0.210705 1.92158 0.527529 -0.496163 0.518954 -0.454118 
EDGE_SE3:QUAT 0 1 0.0347799 0.144352 1.25976 0.554364 -0.474954 0.484748 -0.481787 1 0 0 0 0 0 1 0 0 0 0 1 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 1 4 -0.0253705 -0.654643 -0.193208 0.00355554 0.0553313 -0.00538922 0.998447 1 0 0 0 0 0 1 0 0 0 0 1 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 1 3 0.0182141 -0.44677 -0.0898523 0.00835458 0.056671 -0.00976805 0.99831 1 0 0 0 0 0 1 0 0 0 0 1 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 3 4 -0.0277274 -0.210306 -0.103904 -0.00502714 -0.00135549 0.004112 0.999978 1 0 0 0 0 0 1 0 0 0 0 1 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 2 3 0.0236593 -0.22753 -0.0584976 0.00255161 0.0255556 0.00265438 0.999667 1 0 0 0 0 0 1 0 0 0 0 1 0 0 0 1 0 0 1 0 1 
EDGE_SE3:QUAT 1 2 0.00400111 -0.219419 -0.0277131 0.00540444 0.0311868 -0.0124836 0.999421 1 0 0 0 0 0 1 0 0 0 0 1 0 0 0 1 0 0 1 0 1

RainerKuemmerle, 2016-10-25:
I tried both files with the g2o command line tool. No failure is reported.

ShaowuYang, 2016-10-26:
Hi, Rainer,

Thanks for the reply. I just tried the g2o_viewer. You are right that there's no failure reported. But I think that was because Cholmod solver is used there. Now I've solved my problem by changing CSparse solver to Cholmod solver. Now everything works fine. I'll try to figure out what's wrong when I was using the CSparse solver later when I have time. 

Thanks!

riematrix, 2019-10-29:
@ShaowuYang Hi, 
Did you figure out why CSparse solver causes diverge graph? Have you ever encountered this issue after switch to Cholmod solver?

songyongkang, 2019-11-08:
I have the same problem

closed 2021-02-06 by RainerKuemmerle