RainerKuemmerle/g2o #133: Ubuntu 16.04.1 error

anweshangit, 2016-11-11:
When I try to run g2o_viewer it shows an error:

*** Error in `./g2o_viewer': realloc(): invalid pointer: 0x00007f73ad5e3820 *** 
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f73ade497e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x851d0)[0x7f73ade571d0]
/lib/x86_64-linux-gnu/libc.so.6(realloc+0x22f)[0x7f73ade55d3f]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QListData7reallocEi+0x1f)[0x7f73a950e9cf]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QListData6appendEi+0x81)[0x7f73a950eaa1]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x1d6d78)[0x7f73a95dad78]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_Z21qRegisterResourceDataiPKhS0_S0_+0x2e6)[0x7f73a95d6b16]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x7bcc3)[0x7f73a947fcc3]
/lib64/ld-linux-x86-64.so.2(+0x104ea)[0x7f73afce14ea]
/lib64/ld-linux-x86-64.so.2(+0x105fb)[0x7f73afce15fb]
/lib64/ld-linux-x86-64.so.2(+0xcfa)[0x7f73afcd1cfa]
======= Memory map: ========
00400000-00402000 r-xp 00000000 08:12 919717                             /home/anweshan/g2o/bin/g2o_viewer
00601000-00602000 r--p 00001000 08:12 919717                             /home/anweshan/g2o/bin/g2o_viewer
00602000-00603000 rw-p 00002000 08:12 919717                             /home/anweshan/g2o/bin/g2o_viewer
01af5000-01b27000 rw-p 00000000 00:00 0                                  [heap]
7f73a0000000-7f73a0021000 rw-p 00000000 00:00 0 
7f73a0021000-7f73a4000000 ---p 00000000 00:00 0 
7f73a40b7000-7f73a596d000 r-xp 00000000 08:12 22683986                   /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7f73a596d000-7f73a5b6c000 ---p 018b6000 08:12 22683986                   /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7f73a5b6c000-7f73a5b6d000 r--p 018b5000 08:12 22683986                   /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7f73a5b6d000-7f73a5b6e000 rw-p 018b6000 08:12 22683986                   /usr/lib/x86_64-linux-gnu/libicudata.so.55.1
7f73a5b6e000-7f73a5b91000 r-xp 00000000 08:12 22683866                   /usr/lib/x86_64-linux-gnu/libgraphite2.so.3.0.1
7f73a5b91000-7f73a5d90000 ---p 00023000 08:12 22683866                   /usr/lib/x86_64-linux-gnu/libgraphite2.so.3.0.1
7f73a5d90000-7f73a5d92000 r--p 00022000 08:12 22683866                   /usr/lib/x86_64-linux-gnu/libgraphite2.so.3.0.1
7f73a5d92000-7f73a5d93000 rw-p 00024000 08:12 22683866                   /usr/lib/x86_64-linux-gnu/libgraphite2.so.3.0.1
7f73a5d93000-7f73a5d98000 r-xp 00000000 08:12 22683319                   /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7f73a5d98000-7f73a5f97000 ---p 00005000 08:12 22683319                   /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7f73a5f97000-7f73a5f98000 r--p 00004000 08:12 22683319                   /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7f73a5f98000-7f73a5f99000 rw-p 00005000 08:12 22683319                   /usr/lib/x86_64-linux-gnu/libXdmcp.so.6.0.0
7f73a5f99000-7f73a5ffd000 r-xp 00000000 08:12 22684257                   /usr/lib/x86_64-linux-gnu/libpcre16.so.3.13.2
7f73a5ffd000-7f73a61fd000 ---p 00064000 08:12 22684257                   /usr/lib/x86_64-linux-gnu/libpcre16.so.3.13.2
7f73a61fd000-7f73a61fe000 r--p 00064000 08:12 22684257                   /usr/lib/x86_64-linux-gnu/libpcre16.so.3.13.2
7f73a61fe000-7f73a61ff000 rw-p 00065000 08:12 22684257                   /usr/lib/x86_64-linux-gnu/libpcre16.so.3.13.2
7f73a61ff000-7f73a637e000 r-xp 00000000 08:12 22684000                   /usr/lib/x86_64-linux-gnu/libicuuc.so.55.1
7f73a637e000-7f73a657e000 ---p 0017f000 08:12 22684000                   /usr/lib/x86_64-linux-gnu/libicuuc.so.55.1
7f73a657e000-7f73a658e000 r--p 0017f000 08:12 22684000                   /usr/lib/x86_64-linux-gnu/libicuuc.so.55.1
7f73a658e000-7f73a658f000 rw-p 0018f000 08:12 22684000                   /usr/lib/x86_64-linux-gnu/libicuuc.so.55.1
7f73a658f000-7f73a6593000 rw-p 00000000 00:00 0 
7f73a6593000-7f73a67e5000 r-xp 00000000 08:12 22683988                   /usr/lib/x86_64-linux-gnu/libicui18n.so.55.1
7f73a67e5000-7f73a69e5000 ---p 00252000 08:12 22683988                   /usr/lib/x86_64-linux-gnu/libicui18n.so.55.1
7f73a69e5000-7f73a69f4000 r--p 00252000 08:12 22683988                   /usr/lib/x86_64-linux-gnu/libicui18n.so.55.1
7f73a69f4000-7f73a69f5000 rw-p 00261000 08:12 22683988                   /usr/lib/x86_64-linux-gnu/libicui18n.so.55.1
7f73a69f5000-7f73a6a51000 r-xp 00000000 08:12 22676963                   /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0.10000.1
7f73a6a51000-7f73a6c51000 ---p 0005c000 08:12 22676963                   /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0.10000.1
7f73a6c51000-7f73a6c52000 r--p 0005c000 08:12 22676963                   /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0.10000.1
7f73a6c52000-7f73a6c53000 rw-p 0005d000 08:12 22676963                   /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0.10000.1
7f73a6c53000-7f73a6c57000 r-xp 00000000 08:12 5509656                    /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7f73a6c57000-7f73a6e56000 ---p 00004000 08:12 5509656                    /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7f73a6e56000-7f73a6e57000 r--p 00003000 08:12 5509656                    /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7f73a6e57000-7f73a6e58000 rw-p 00004000 08:12 5509656                    /lib/x86_64-linux-gnu/libuuid.so.1.3.0
7f73a6e58000-7f73a6e5f000 r-xp 00000000 08:12 22683691                   /usr/lib/x86_64-linux-gnu/libffi.so.6.0.4
7f73a6e5f000-7f73a705e000 ---p 00007000 08:12 22683691                   /usr/lib/x86_64-linux-gnu/libffi.so.6.0.4
7f73a705e000-7f73a705f000 r--p 00006000 08:12 22683691                   /usr/lib/x86_64-linux-gnu/libffi.so.6.0.4
7f73a705f000-7f73a7060000 rw-p 00007000 08:12 22683691                   /usr/lib/x86_64-linux-gnu/libffi.so.6.0.4
7f73a7060000-7f73a70ce000 r-xp 00000000 08:12 5509601                    /lib/x86_64-linux-gnu/libpcre.so.3.13.2
7f73a70ce000-7f73a72ce000 ---p 0006e000 08:12 5509601                    /lib/x86_64-linux-gnu/libpcre.so.3.13.2
7f73a72ce000-7f73a72cf000 r--p 0006e000 08:12 5509601                    /lib/x86_64-linux-gnu/libpcre.so.3.13.2
7f73a72cf000-7f73a72d0000 rw-p 0006f000 08:12 5509601                    /lib/x86_64-linux-gnu/libpcre.so.3.13.2
7f73a72d0000-7f73a72d2000 r-xp 00000000 08:12 22683308                   /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7f73a72d2000-7f73a74d2000 ---p 00002000 08:12 22683308                   /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7f73a74d2000-7f73a74d3000 r--p 00002000 08:12 22683308                   /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7f73a74d3000-7f73a74d4000 rw-p 00003000 08:12 22683308                   /usr/lib/x86_64-linux-gnu/libXau.so.6.0.0
7f73a74d4000-7f73a7536000 r-xp 00000000 08:12 22683345                   /usr/lib/x86_64-linux-gnu/libXt.so.6.0.0
7f73a7536000-7f73a7736000 ---p 00062000 08:12 22683345                   /usr/lib/x86_64-linux-gnu/libXt.so.6.0.0
7f73a7736000-7f73a7737000 r--p 00062000 08:12 22683345                   /usr/lib/x86_64-linux-gnu/libXt.so.6.0.0
[... 94 lines cut ...]
7f73ab299000-7f73ab29a000 r--p 00015000 08:12 22683174                   /usr/lib/x86_64-linux-gnu/libICE.so.6.3.0
7f73ab29a000-7f73ab29b000 rw-p 00016000 08:12 22683174                   /usr/lib/x86_64-linux-gnu/libICE.so.6.3.0
7f73ab29b000-7f73ab29e000 rw-p 00000000 00:00 0 
7f73ab29e000-7f73ab2a5000 r-xp 00000000 08:12 22683294                   /usr/lib/x86_64-linux-gnu/libSM.so.6.0.1
7f73ab2a5000-7f73ab4a4000 ---p 00007000 08:12 22683294                   /usr/lib/x86_64-linux-gnu/libSM.so.6.0.1
7f73ab4a4000-7f73ab4a5000 r--p 00006000 08:12 22683294                   /usr/lib/x86_64-linux-gnu/libSM.so.6.0.1
7f73ab4a5000-7f73ab4a6000 rw-p 00007000 08:12 22683294                   /usr/lib/x86_64-linux-gnu/libSM.so.6.0.1
7f73ab4a6000-7f73ab4f8000 r-xp 00000000 08:12 22683846                   /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.1
7f73ab4f8000-7f73ab6f7000 ---p 00052000 08:12 22683846                   /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.1
7f73ab6f7000-7f73ab6f8000 r--p 00051000 08:12 22683846                   /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.1
7f73ab6f8000-7f73ab6f9000 rw-p 00052000 08:12 22683846                   /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.1
7f73ab6f9000-7f73ab79d000 r-xp 00000000 08:12 22683739                   /usr/lib/x86_64-linux-gnu/libfreetype.so.6.12.1
7f73ab79d000-7f73ab99c000 ---p 000a4000 08:12 22683739                   /usr/lib/x86_64-linux-gnu/libfreetype.so.6.12.1
[... cut ...]
7f73abbe1000-7f73abde0000 ---p 00024000 08:12 5509613                    /lib/x86_64-linux-gnu/libpng12.so.0.54.0
7f73abde0000-7f73abde1000 r--p 00023000 08:12 5509613                    /lib/x86_64-linux-gnu/libpng12.so.0.54.0
7f73abde1000-7f73abde2000 rw-p 00024000 08:12 5509613                    /lib/x86_64-linux-gnu/libpng12.so.0.54.0
7f73abde2000-7f73abef1000 r-xp 00000000 08:12 5509514                    /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.1
7f73abef1000-7f73ac0f0000 ---p 0010f000 08:12 5509514                    /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.1Aborted (core dumped)

creuther, 2016-11-11:
Interestingly enough I recently switched to Ubuntu 16.04 as well and am getting the exact same error. What I noticed is that cmake wouldn't find QGLViewer until I installed the libqglviewer-dev package, which is the Qt5 version. Installing libqglviewer-dev-qt4 alone was not enough, and obviously the error message is referencing Qt5 libraries. I will investigate further why it is possible to compile but not run it, should be due to the mixup of Qt4 and Qt5 somehow.

creuther, 2016-11-11:
Right, so I was able to get it to work by fixing the FindQGLViewer cmake module, installing libqglviewer-dev-qt4 (will uninstall libqglviewer-dev) and then recompiling. You can see my changes [here](https://github.com/creuther/g2o/commit/96d22ed1beab7c7e0c215366b692036f1369e197). Unfortunately this is as much as I have time for right now, there still seems to be that problem when compiling against Qt5.

Anyway, @RainerKuemmerle should I submit a PR for the Qt4 fix? I'm not sure whether it's a generic enough fix, that's why I am asking.

anweshangit, 2016-11-11:
We have also got it to work but it seems your method is much better.
What we did:
1. Force downgraded the cmake 3.5 to 2.8 (so that the current cmake file finds the X11 libraries).
2. Removed Qt4 from the system as g2o_viewer was trying to link both Qt4 and Qt5 libraries.

nh2, 2016-12-09:
@creuther I could make it work on Ubuntu 16.04 after `sudo apt-get install libqglviewer-dev-qt4` and applying this patch to your commit 96d22ed1beab7c7e0c215366b692036f1369e197

```patch
diff --git a/cmake_modules/FindQGLViewer.cmake b/cmake_modules/FindQGLViewer.cmake
index 5cf7db3..cf5516e 100644
--- a/cmake_modules/FindQGLViewer.cmake
+++ b/cmake_modules/FindQGLViewer.cmake
@@ -19,7 +19,7 @@ FIND_PATH(QGLVIEWER_INCLUDE_DIR qglviewer.h
   )
 
 find_library(QGLVIEWER_LIBRARY_RELEASE
-  NAMES qglviewer-qt4 qglviewer QGLViewer QGLViewer2
+  NAMES qglviewer-qt4 QGLViewer-qt4 qglviewer QGLViewer QGLViewer2
   PATHS /usr/lib
         /usr/local/lib
         /usr/lib/x86_64-linux-gnu # Ubuntu 16.04 and derivatives
```

In other words, adding case-sensitive `QGLViewer-qt4` to the already existing `qglviewer-qt4`.

Lucky8Young, 2017-01-08:
@creuther I am very sorry to bother you,but I have the same error when I run g2o_viewer in Ubuntu16.04.
And I use your solution(fixing the FindQGLViewer cmake module, installing libqglviewer-dev-qt4 (will uninstall libqglviewer-dev) and then recompiling.) but not working, because there is an error:No rule to make target '/usr/lib/x86_64-linux-gnu/libQGLViewer.so' when I "make" .I have try some methods to fix it but sitll cannot run g2o_viewer in Ubuntu16.04. Could you please give me some advice?

creuther, 2017-01-12:
@Lucky8Young Sorry, I didn't notice the notification for a while. The error you are getting is very strange, as there should be no make target for QGLViewer, it should just be linked to the g2o_viewer binary. I suggest deleting the temporary files created by cmake in g2o and building it again - cmake does (disclaimer here: I don't have that much knowledge in this area) cache a lot of configuration data the first time you build g2o. If you then change things on the system (like installing/uninstalling devel-libraries etc) and run "make" again, it might not consider these changes and fail to build.

RainerKuemmerle, 2017-01-12:
4bcabdfb48d4a0a39663fff3623f005fe253f74b changes g2o to rely on Qt5 (no more support of Qt4). This should also fix this issue.

closed 2017-01-12 by RainerKuemmerle

cgbcgb, 2017-11-08:
@nh2  thanks to you, I meet the same problem in Ubuntu16.04 when running g2o_viewer, like this:

find_library(QGLVIEWER_LIBRARY_RELEASE
#  NAMES qglviewer-qt4 qglviewer QGLViewer QGLViewer2
  NAMES qglviewer-qt4 QGLViewer-qt4 qglviewer QGLViewer QGLViewer2

Young532042725, 2017-12-14:
@creuther   dear friend, i am sorry to interrupt you. i have confronted with the same problem as @anweshangit . could you give me some advice.
[image]

cgbcgb, 2017-12-14:
@Young532042725 , please see 5楼 ‘nh2 commented on 9 Dec 2016’，it help me solved this problem

Young532042725, 2017-12-14:
@cgbcgb  i am ok ,thank you

Young532042725, 2017-12-14:
@cgbcgb dear friend, are you Chinese? make friends? hahaha

cgbcgb, 2017-12-14:
@Young532042725 原来是Tj的童鞋，其实5楼漏了一点，要先把已经安装的g2o卸载再操作。。。。。g2o有毒，听说Google的ceres库维护的比较好

Young532042725, 2017-12-14:
@cgbcgb 加你QQ好友吧   嘿嘿。我的用户名就是QQ