RainerKuemmerle/g2o #138: (Inverse) depth optimization

hit2sjtu, 2016-12-06:
Any idea how can I optimize the depth/inverse depth of 3D points instead of XYZ like in visual slam.
I did not see any in the documents.
Thanks in advance.

Best Regards!

felixendres, 2016-12-06:
Check out the slam3d types in [pointxyz_depth.h/cpp](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/types/slam3d/edge_se3_pointxyz_depth.h)  and [edge_se3_pointxyz_depth.h/cpp](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/types/slam3d/edge_se3_pointxyz_depth.h).

For inverse depth, I guess you'd need to adapt the computeError and linearizeOplus methods in the edge.

hit2sjtu, 2016-12-15:
Thanks. Rewriting computeError linearizeOplus  works for me.

closed 2016-12-15 by hit2sjtu