RainerKuemmerle/g2o #146: Question about SE3 Jacobian matrix computation in g2o

hlc111, 2017-01-06:
Hi Author,

Sorry to bother you!
Those days i'm study G2O, it's a great optimizer project.
For more deep understand G2O, i'm debugging the code now.
But for how to get Jacobian i can't understand in 
[Jacobian computation of EdgeSE3 linearizeOplus computeEdgeSE3Gradient](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/types/slam3d/isometry3d_gradients.h#L194)

Could you kindly explain or provide some math formula reference help me understand it?

Currently i can understand use rotation matrix get Quaternion w, but don't know why use qw get dq_dR and use dq_dR and use those data get Ji and Jj.

Thanks!

lianbin, 2020-12-31:
> Hi Author,
>
> Sorry to bother you!
> Those days i'm study G2O, it's a great optimizer project.
> For more deep understand G2O, i'm debugging the code now.
> But for how to get Jacobian i can't understand in [...]

Hi,Do you know how to derive Jacobi now ? Can you tell me?

RainerKuemmerle, 2021-01-03:
You may have a look at chain rule and friends.
All together this is a detail of implementation and you may obtain the values of the Jacobian by other means, i.e., the default numerical implementation or by AD (as we are currently working on in a branch, see https://github.com/RainerKuemmerle/g2o/blob/rainer/auto-diff/g2o/core/auto_differentiation.h )

closed 2025-02-12 by github-actions[bot]