RainerKuemmerle/g2o #147: CMake Error: The following variables are used in this project, but they are set to   NOTFOUND. Please set them or make sure they are set and tested correctly in the CMake files: G2O_SOLVER_CHOLMOD     linked by target "svo" in directory /root/ros_catkin_ws/src/rpg_svo/svo  -- Configuring incomplete, errors occurred! See also "/root/ros_catkin_ws/build/CMakeFiles/CMakeOutput.log". See also "/david/ros_catkin_ws/build/CMakeFiles/CMakeError.log". Invoking "cmake" failed

barnett88, 2017-01-09:
Run catkin_make command error
CMake Error: The following variables are used in this project, but they are set to

NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
G2O_SOLVER_CHOLMOD
linked by target "svo" in directory /root/ros_catkin_ws/src/rpg_svo/svo

-- Configuring incomplete, errors occurred!
See also "/david/ros_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/david/ros_catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

How can I fix it?

RainerKuemmerle, 2017-01-12:
Did you install libsuitesparse-dev? Otherwise the Cholmod solver is not built.

barnett88, 2017-01-13:
Has been resolved, thank you.

closed 2017-01-13 by barnett88