RainerKuemmerle/g2o #148: Memory leak of Factory

MalcolmMielle, 2017-01-10:
Hello all,

I have a memory leak problem where, when run with Valgrind I get:

```
==17905== 16 bytes in 1 blocks are definitely lost in loss record 136 of 356
==17905==    at 0x4C29216: operator new(unsigned long) (vg_replace_malloc.c:334)
==17905==    by 0x4E7159E: g2o::Factory::registerType(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, g2o::AbstractHyperGraphElementCreator*) (in /usr/local/lib/libg2o_core.so)
==17905==    by 0x5E95D52: g2o::RegisterTypeProxy<g2o::EdgeSE2TwoPointsXY>::RegisterTypeProxy(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (in /usr/local/lib/libg2o_types_slam2d.so)
==17905==    by 0x5E80F7B: __static_initialization_and_destruction_0(int, int) [clone .constprop.24] (in /usr/local/lib/libg2o_types_slam2d.so)
==17905==    by 0x400F739: call_init.part.0 (in /usr/lib64/ld-2.22.so)
==17905==    by 0x400F84A: _dl_init (in /usr/lib64/ld-2.22.so)
==17905==    by 0x4000CB9: ??? (in /usr/lib64/ld-2.22.so)
```

It seems like the Factory is not freeing the memory and I run g2o::Factory::destroy(); at the end of my program. However I did put some prints in this function and none of them appears so I'm wondering if it even access it and why it wouldn't.

This is my main file:

```
#include "g2o/types/slam2d/parameter_se2_offset.h"
#include "g2o/core/sparse_optimizer.h"
#include "g2o/core/block_solver.h"
#include "g2o/core/factory.h"
#include "g2o/core/optimization_algorithm_factory.h"
#include "g2o/core/optimization_algorithm_gauss_newton.h"
#include "g2o/solvers/csparse/linear_solver_csparse.h"
#include "g2o/core/robust_kernel_impl.h"

int maint(){
	std::cout << "FACFAC" << std::endl;
	g2o::Factory::destroy(); 
	std::cout << "OAFactory destroy" << std::endl;
	g2o::OptimizationAlgorithmFactory::destroy();
	std::cout << "HGFactory destroy" << std::endl;
}
```

ANy hints as to why I have a memory leak ?

sergiud, 2017-01-10:
Have you tried calling `g2o::Factory::destroy` just before your program exits?

xzwang12345, 2020-12-29:
Hi @MalcolmMielle,I have the same problem with you,did you solve it ? Thanks

RainerKuemmerle, 2020-12-29:
This is buggy. Two things could happen here. First, the some registration proxy again calls instance() during its d'tor leading to a new allocation of the factory after destroy. Furthermore, the above example may segfault due to double free, which is for sure a bug.
I did so far not address this, as I thought it's just a minor glitch at the end of the runtime of main. But I guess it might interfere when searching for memory leaks in general.

RainerKuemmerle, 2020-12-29:
Using the following code
```#include "g2o/core/factory.h"
#include "g2o/core/optimization_algorithm_factory.h"

G2O_USE_TYPE_GROUP(slam2d);
G2O_USE_TYPE_GROUP(slam3d);

G2O_USE_OPTIMIZATION_LIBRARY(eigen)

int main(){
	g2o::Factory::destroy();
	g2o::OptimizationAlgorithmFactory::destroy();
}
```

with the PR https://github.com/RainerKuemmerle/g2o/pull/472 the example now yields
```
==20791== Memcheck, a memory error detector
==20791== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==20791== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==20791== Command: ../bin/simple_optimize
==20791== 
==20791== 
==20791== HEAP SUMMARY:
==20791==     in use at exit: 0 bytes in 0 blocks
==20791==   total heap usage: 645 allocs, 645 frees, 117,386 bytes allocated
==20791== 
==20791== All heap blocks were freed -- no leaks are possible
==20791== 
==20791== For lists of detected and suppressed errors, rerun with: -s
==20791== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

PR #472 (2020-12-29): Use smart pointer in the factories
fixes #148
commit 38d7ab55 (2020-12-29): Use smart pointer in the factories (#472)

* Use smart pointer in the factories
* Avoid unregister in factory proxies

fixes #148
closed 2020-12-29 by RainerKuemmerle