RainerKuemmerle/g2o #151: g2o_viewer

Exception4U, 2017-01-23:
g2o_viewer unable to open.  gives segmentation fault on ubuntu 14,04

RainerKuemmerle, 2017-01-23:
Can you please provide the output of ldd.
Best guess no Qt5 QGLViewer available un Ubuntu 14.04 leading to linkage of both Qt4 and Qt5.
Qt4 is no longer maintained (end of life Dec. 2015), therefore I removed support for it.

niosus, 2017-01-23:
I guess the problem is that Ubuntu 14.04 links to qt4 by default in many places (e.g. in ROS). So if g2o is used as part of ROS-based pipeline (just an example) there is no way to avoid "bleeding" qt4 libraries into the build.

AhmedElsafy, 2017-01-27:
Segmentation fault on Ubuntu 14.04

AhmedElsafy, 2017-01-30:
I think it might be the error is hear 

safy@safy-Satellite-L840D:~/Downloads/g2o (2)/bin$ ldd g2o_viewer | grep Qt
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f82080c0000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f8206e98000)
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f8206848000)
	libQt5Xml.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Xml.so.5 (0x00007f8206608000)
	libQt5OpenGL.so.5 => /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5 (0x00007f82063a0000)
	libQtXml.so.4 => /usr/lib/x86_64-linux-gnu/libQtXml.so.4 (0x00007f8204b18000)
	libQtOpenGL.so.4 => /usr/lib/x86_64-linux-gnu/libQtOpenGL.so.4 (0x00007f8204818000)
	libQtGui.so.4 => /usr/lib/x86_64-linux-gnu/libQtGui.so.4 (0x00007f8203b60000)
	libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4 (0x00007f8203678000)

Exception4U, 2017-02-09:
ldd g2o_viewer a.g2o 
g2o_viewer:
ldd: ./g2o_viewer: No such file or directory
a.g2o:
	not a dynamic executable

zainmehdi, 2017-02-13:
I have the same issue. Can someone Kindly let me know what’s the solution ?

zainmehdi, 2017-02-18:
I got it figured out. As mentioned above its dependency issue. I manually installed qt5 and libqglviewer library and modified the cmake file to include libqglviewer-qt5 path and then compiled it.

lastsongforu, 2017-02-22:
@zainmehdi but which cmake file should be modified and how to? can you make an instruction in detail?

zainmehdi, 2017-02-22:
@lastsongforu I did as follows.
1) install qt5 (use `qmake -v` to check the version, it should be qt5)
2) dont install libqglviewer -dev from apt instead download it from here [http://libqglviewer.com/installUnix.html] and follow the instructions
3) After you have installed both libqglviewer and qt5 access the g2o folder and navigate to cmake_modules folder. Open FindQGLViewer.cmake
4)Modify as shown in pic 
[image]
5) now compile and run g2o viewer should work.
Let me know in case it doesn’t  work. I will be more than happy to help.

buqing2009, 2017-02-22:
In fact, just insure that the Qt5 library depends on libqglviewer and g2o are the same.
In my ubuntu 14.04, i installed ROS indigo , libqt5* , libqt4* and even QT5.7 packages from [QT official website](https://www.qt.io/qt5-7/), and build [libqglviewer 2.6.4](http://libqglviewer.com/installUnix.html#linux) from source. I just use cmake-curses-gui to configure libqglviewer and g2o, let them have the same QT dependency, and g2o_viewer works.

lastsongforu, 2017-02-23:
@zainmehdi @buqing2009 I install qt5 and libqglviewer as zainmehdi said. Then i modify FindQGLViewer.cmake. When i compile such errors occur
[image]
despairing  T.T

zainmehdi, 2017-02-24:
@lastsongforu did you compile libqglviewer using qt5 ?

TurtleZhong, 2017-02-24:
@zainmehdi I followed yours and installed it suscessfully!
 but  i got this,error while loading shared libraries: libQGLViewer-qt5.so.2: cannot open shared object file: No such file or directory
then I ran sudo ldconfig -v
and i got this
 m@m-pc  ~  
╰─$ g2o_viewer result_after.g2o
[1]    10633 segmentation fault (core dumped)  g2o_viewer result_after.g2o
Do you have some ideal about this ?

lastsongforu, 2017-02-26:
finally i got it solved... with ubuntu 16.04. i tried my best in 12.04 & 14.04 but there are too many strange errors about libQGLViewer , QT version and dependence conflicts. in 16.04, everything compiles successfully just once.

TurtleZhong, 2017-02-26:
May be cuz Ubuntu 16.04 based on QT5 and 14.04 is QT4? I decided to give up the g2o_viewer module~

TurtleZhong, 2017-02-26:
May be cuz Ubuntu 16.04 based on QT5 and 14.04 is QT4? I decided to give up the g2o_viewer module~@lastsongforu

lastsongforu, 2017-02-26:
@TurtleZhong yes it's a problem. but when i use QT4 only in 12.04, g2o_viewer still can't compile. i'm sure the settings are right. i guess current g2o version has poor support for old ubuntu.

zainmehdi, 2017-02-27:
@TurtleZhong its because your system is mixing qt4 and qt5. you need to completely remove qt4 and then install qt5 if qtchooser command doesnt work in your case. If its important for you I can come live on video and help you out if interested.

TurtleZhong, 2017-02-28:
@zainmehdi Really thanks！I think may be it is useful to others. I install ros indigo on ubuntu 14.04 and it relies qt4. so may be I could not remove qt4~ If possible I will try 16.04~Thanks anyway!

xuao0728, 2017-03-05:
@zainmehdi Hi, i was solving the same problem, i want to know the solution you proposed could work well only under the ubuntu 16.04 or is also useful for ubuntu 14.04? Thank you very much.

zainmehdi, 2017-03-05:
@xuao0728 Hi, I found my solution in Ubuntu 14.04. For 16.04 you dont need the solution as its already qt5 based. :)

xuao0728, 2017-03-05:
@zainmehdi Thank you for your reply! i have followed your steps ,and g2o could be compiled correctly ,but g2o viewer still couldn't work, and when i run roslaunch rgbdslam rgbdslam.launch in the[(https://github.com/felixendres/rgbdslam_v2)] , error occurred like  :

- error while loading shared libraries: libpcl_ros_tf.so: cannot open shared object file: No such file or directory

Ｂefore doing that, i could run roslaunch rgbdslam rgbdslam.launch correctly,
so could you search the file libpcl_ros_tf.so in you folders , i want to know if you have this file or not～

zainmehdi, 2017-03-05:
@xuao0728 its unable to locate your shared object file. I didnt use rgbdslam but I used LSD Slam and it works. moreover you need to make sure you have all the dependencies installed properly. I have installed both qt4 and qt5 and I switch versions as needed. But for g2o viewer to work you need to have your system free of qt4 and then install qt5 and compile. Later on you can install qt4 as well and switch as needed. and you can try to launch in sudo mode it solves the problems some times. Try it and let me know if you are still unable to run rgbdslam.

blairlpp, 2017-04-10:
@buqing2009  Could you tell me how to let them have the same QT dependency in detail? Thanks a lot!

blairlpp, 2017-04-10:
@zainmehdi  I followed your instruction on 22 Feb, but it didn't work. On ubuntu14.04 ros indigo. Segment fault all the time when run g2o_viewer.

zainmehdi, 2017-04-10:
check your qt version ... it should be qt5 when you compile g2o

blairlpp, 2017-04-10:
@zainmehdi  Thanks for your quick reply. I use `qmake -v`. It tells me that ```
QMake version 3.0
 Using Qt version 5.2.1 in /usr/lib/x86_64-linux-gnu
```
I think it's qt5, is it?

zainmehdi, 2017-04-10:
yes its qt5. did you modify cmake file as I did ? When I compiled g2o I only had qt5 installed and no qt4 later on I installed it and they dont make any discrepency

blairlpp, 2017-04-10:
@zainmehdi 

`~/git_src/g2o/bin$ ldd g2o_viewer | grep Qt`

[image]

zainmehdi, 2017-04-10:
I am not sure whats wrong. I will provide you a detailed procedure at night. A little busy now. Sorry

blairlpp, 2017-04-10:
@zainmehdi  So nice of you! I'll wait for your reply!

blairlpp, 2017-04-11:
@lastsongforu  @zainmehdi  I met the same problem as urs when I make in G2O's build directory.
[image]

blairlpp, 2017-04-12:
I solved it. g2o_viewer can run on ubuntu14.04. But you have to install the right g2o and qglviewer. If someone needs them, I'd love to send a copy to you.

hitlocking, 2017-04-12:
@blairlpp I come to the same problem in ubuntu 14.04. and i don't change to 16.04. could you send me a copy of right g2o and qglviewer? my email is zacharywang@sjtu.edu.cn， or hitwangzan@gmail.com. thank you very much.

blairlpp, 2017-04-13:
https://github.com/gaoxiang12/rgbd-slam-tutorial-gx.git
 Right g2o and qglviewer for ubuntu14.04 are available at the above website.

ChristopherLu, 2017-07-08:
@zainmehdi I try to reproduce what you did, but got the error when I make `libQGLViewer-2.7.0`:

**config.h:79:25: fatal error: QOpenGLWidget: No such file or directory
 #include <QOpenGLWidget>**

I am using ubuntu 14.04 and qmake version is qt5, thx

zainmehdi, 2017-07-08:
@ChristopherLu  did you tried @blairlpp's solution ?

ChristopherLu, 2017-07-10:
@zainmehdi I tried @blairlpp 's solution, but got the following error when I try to `make`:
[ 48%] Linking CXX executable ../../../../bin/g2o_viewer
/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libQGLViewer.so: undefined reference to `gluNewQuadric'
/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libQGLViewer.so: undefined reference to `gluUnProject'
/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libQGLViewer.so: undefined reference to `gluCylinder'
/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libQGLViewer.so: undefined reference to `gluPickMatrix'
/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libQGLViewer.so: undefined reference to `gluSphere'
/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libQGLViewer.so: undefined reference to `gluProject'
collect2: error: ld returned 1 exit status
make[2]: *** [../bin/g2o_viewer] Error 1

WeiXin19940107, 2017-08-27:
@ChristopherLu  Hi, i currently meet the same problem when i make libQGLViewer:
QOpenGLWidget: No such file or directory 
Besides, i am using ubuntu 14.04 and qmake version is also qt5.
So have u fixed this problem? Thx！

zainmehdi, 2017-08-28:
@ChristopherLu  What steps are you following. Its about dependency which if not fulfilled properly creates issues.

WeiXin19940107, 2017-08-28:
@zainmehdi  Hi, the thing is that my QT5 is not the latest version and the libQGLViewer needs at  least version QT5.4.After i have upgraded my qt to qt5.9.1, my problem is fixed.
However, still my g2o_viewer doesn't work. And could you give me some advice? 
My e-mail address is weixinde@mail.ustc.edu.cn.  Thank u so much!

Young532042725, 2017-12-14:
@zainmehdi  hello! I am a postgraduate，when i use the g2o_viewer to visualize a file xx.g2o Suffix name，some errors occur.here is the screenshot. could you help me? thank you very much.(my system is Ubuntu 16.04)
[image]

zainmehdi, 2017-12-14:
@Young532042725 check if you can start g2o viewer without the file i.e just run the ./g2o_viewer

Young532042725, 2017-12-14:
@zainmehdi in which directory? in the dir which include .g2o? in that case, here is the result.
[image]

Young532042725, 2017-12-14:
@zainmehdi Thanks for your reply, i execute the g2o_viewer in the dir.there is the result...
[image]

zainmehdi, 2017-12-14:
@Young532042725 try the solution [here](https://github.com/RainerKuemmerle/g2o/issues/133) it worked for them

Young532042725, 2017-12-14:
@zainmehdi   i have solved the problem.  thank  you very much

ZhangPein, 2017-12-15:
@blairlpp hello,I have download the qglviewer for ubuntu14.04 you provided,but I can't install it.could you please help me to solve this problem?Thank you very much!
/home/zhangpei/图片/2017-12-15 12:08:17屏幕截图.png

ZhangPein, 2017-12-15:
Hello,I have downloaded the qglviewer for ubuntu14.04 you provided,but I can't install it.Could you please help me to solve this problem?Thank you very much!
[image]

csufangyu, 2018-02-02:
＠ChristopherLu  I meet the same problem,could you help me?

HuangXiaoquan127, 2018-04-24:
In addition, if you install qt4 and qt5 at the same time, you also need to annotate this line:
#FIND_PACKAGE(Qt4 COMPONENTS QtCore QtXml QtOpenGL QtGui)
IF(NOT Qt4_FOUND)
	FIND_PACKAGE(Qt5 QUIET COMPONENTS Core Xml OpenGL Gui Widgets)

foreverlms, 2019-04-30:
If there are still someone has a problem with `g2o_viewer` under 14.04, I wrote a [blog](https://blog.bobliao.fun/2019/04/30/g2o-in-ubuntu14-04/). You can have it a look.If you have any questions, contact with me.

zainmehdi, 2021-02-07:
@IcneDy sorry for the late response. Its been a while since I used the viewer but as I remember it compiles fine in ubuntu 18.04. You wont have to go through this process. Which qt version and which distro of ubuntu are you using ?

IcneDy, 2021-02-07:
@zainmehdi Thanks for your response. I use Qt 5.12 on fedora 33 because ubuntu 18.04 can't be installed on my device. I uninstalled and reinstalled g2o. g2o_viewer can be open but its can't show correctly (not a black screen, a bit like garbled). octovis has the similar problem.

zainmehdi, 2021-02-07:
for fedora I havnt tried it. Can you let me know of any error / screen response. and did u try it with sudo  ?

IcneDy, 2021-02-07:
Thanks for your help. I try it with sudo and su- and g2o_viewer can show correctly(but 'su' still has error). T here are the interface of g2o_viewer and shell on the 'su'. Thanks for your help again. 
[image]
[image]

zainmehdi, 2021-02-07:
@IcneDy so its working ? and the warning message is most probably some folder permission issue. Can be ignored I guess.

IcneDy, 2021-02-07:
@zainmehdi The above is the wrong interface(su). It can work correctly now. 
[image]

heroacool, 2023-07-31:
install libqglviewer-dev-qt5