RainerKuemmerle/g2o #152: Linking Issue on OSx using CMake

mehoggan, 2017-01-26:
Building my project came in the following manner:

`/Applications/CLion.app/Contents/bin/cmake/bin/cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_VERBOSE_MAKEFILE=On -DCMAKE_BUILD_TYPE=Release -G "CodeBlocks - Unix Makefiles" /Users/mhoggan/Development/bluenote/image_pose_adjustment/image_pose_adjustment/native/graph`

I get the following build error when my library goes to link against g2o.

```
cd /Users/mhoggan/Development/bluenote/image_pose_adjustment/image_pose_adjustment/native/graph/cmake-build-debug/src && /Applications/CLion.app/Contents/bin/cmake/bin/cmake -E cmake_link_script CMakeFiles/graph_optimization.dir/link.txt --verbose=1
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++    -std=c++11 -Wall -Werror -O3 -DNDEBUG -Wl,-search_paths_first -Wl,-headerpad_max_install_names  CMakeFiles/graph_optimization.dir/main.cpp.o CMakeFiles/graph_optimization.dir/BundleRigidGraphOptimizer.cpp.o CMakeFiles/graph_optimization.dir/FlexibleGraphOptimizer.cpp.o CMakeFiles/graph_optimization.dir/LoopClosure.cpp.o CMakeFiles/graph_optimization.dir/PoseJsonReader.cpp.o  -o graph_optimization  -L/Users/mhoggan/Development/bluenote/image_pose_adjustment/image_pose_adjustment/native/graph/cmake-build-debug/g2o-src/lib  /usr/local/lib/libboost_filesystem-mt.a /usr/local/lib/libboost_system-mt.a -lg2o_cli -lg2o_core -lg2o_interface -lg2o_parser -lg2o_solver_csparse -lg2o_solver_dense -lg2o_solver_pcg -lg2o_stuff -lg2o_types_icp -lg2o_types_sba -lg2o_types_sim3 -lg2o_types_slam2d -lg2o_types_slam3d -lg2o_csparse_extension -lcsparse -ljsoncpp -Wl,-rpath,/Users/mhoggan/Development/bluenote/image_pose_adjustment/image_pose_adjustment/native/graph/cmake-build-debug/g2o-src/lib 
Undefined symbols for architecture x86_64:
  "g2o::csparse_extension::cs_cholsolsymb(cs_sparse const*, double*, cs_symbolic const*, double*, int*)", referenced from:
      g2o::LinearSolverCSparse<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> > const&, double*, double*) in LoopClosure.cpp.o
  "g2o::csparse_extension::writeCs2Octave(char const*, cs_sparse const*, bool)", referenced from:
      g2o::LinearSolverCSparse<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> > const&, double*, double*) in LoopClosure.cpp.o
  "g2o::csparse_extension::cs_chol_workspace(cs_sparse const*, cs_symbolic const*, int*, double*)", referenced from:
      g2o::LinearSolverCSparse<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::solveBlocks(double**&, g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> > const&) in LoopClosure.cpp.o
      g2o::LinearSolverCSparse<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::solvePattern(g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> >&, std::__1::vector<std::__1::pair<int, int>, std::__1::allocator<std::__1::pair<int, int> > > const&, g2o::SparseBlockMatrix<Eigen::Matrix<double, -1, -1, 0, -1, -1> > const&) in LoopClosure.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

However, the nm utility shows the symbols are present:

```
mhoggan-C02S81PRG8WM:graph mhoggan$ nm /Users/mhoggan/Development/bluenote/image_pose_adjustment/image_pose_adjustment/native/graph/cmake-build-debug/g2o-src/lib/libg2o_csparse_extension.dylib | grep cs_cholsolsymb
0000000000002120 T __ZN3g2o17csparse_extension14cs_cholsolsymbEPK12cs_di_sparsePdPK14cs_di_symbolicS4_Pi
mhoggan-C02S81PRG8WM:graph mhoggan$ nm /Users/mhoggan/Development/bluenote/image_pose_adjustment/image_pose_adjustment/native/graph/cmake-build-debug/g2o-src/lib/libg2o_csparse_extension.dylib | grep writeCs2Octave
00000000000027d0 T __ZN3g2o17csparse_extension14writeCs2OctaveEPKcPK12cs_di_sparseb
mhoggan-C02S81PRG8WM:graph mhoggan$ nm /Users/mhoggan/Development/bluenote/image_pose_adjustment/image_pose_adjustment/native/graph/cmake-build-debug/g2o-src/lib/libg2o_csparse_extension.dylib | grep cs_chol_workspace
0000000000002210 T __ZN3g2o17csparse_extension17cs_chol_workspaceEPK12cs_di_sparsePK14cs_di_symbolicPiPd
```

sergiud, 2017-01-26:
I guess you compiled only a 32-bit version of g2o, but trying to link a 64-bit target to it. This is from your log:
```
ld: symbol(s) not found for architecture x86_64
```

mehoggan, 2017-01-26:
From the following:

mhoggan-C02S81PRG8WM:cmake-build-debug mhoggan$ file ./g2o-src/lib/libg2o_csparse_extension.dylib
./g2o-src/lib/libg2o_csparse_extension.dylib: Mach-O 64-bit dynamically linked shared library x86_64

I would assume that last comment to be incorrect.

sergiud, 2017-01-26:
Is the C++ standard being used by the compiler (i.e., the `-std=c++xx` compile flag) consistent across the targets?

mehoggan, 2017-01-26:
Outside of g2o there is only one other target, a target added using add_executable and it has -std=c++11 set in CXX_FLAGS. As seen in the following compiler line:

`cd /Users/mhoggan/Development/bluenote/image_pose_adjustment/image_pose_adjustment/native/graph/cmake-build-debug/src && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++    -I/usr/local/include -I/usr/local/include/eigen3 -I/Users/mhoggan/Development/bluenote/image_pose_adjustment/image_pose_adjustment/native/graph/cmake-build-debug/g2o-build -I/Users/mhoggan/Development/bluenote/image_pose_adjustment/image_pose_adjustment/native/graph/cmake-build-debug/g2o-src -I/Users/mhoggan/Development/bluenote/image_pose_adjustment/image_pose_adjustment/native/graph/cmake-build-debug/g2o-src/EXTERNAL/csparse -I/Users/mhoggan/Development/bluenote/image_pose_adjustment/image_pose_adjustment/native/graph/include  -std=c++11 -Wall -Werror -O3 -DNDEBUG   -o CMakeFiles/graph_optimization.dir/PoseJsonReader.cpp.o -c /Users/mhoggan/Development/bluenote/image_pose_adjustment/image_pose_adjustment/native/graph/src/PoseJsonReader.cpp`

This is the link right before the executable goes to link against g2o.

mehoggan, 2017-01-27:
This only happens on a El Capitan OSx. But is a result of including linear_solver_csparse.h from a executable linking against g2o.

rancheng, 2017-03-24:
Hi, I just figured out this problem, by adding this line in my` CMakeLists.txt` file:

`set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -stdlib=libstdc++ -lstdc++ -std=c++11")`

Please note: This is a Fu*king typical problem on `Clang`, and you can search the funning history with google: 
The **libstdC++ vs libC++** issue!

hope you guys work well, good luck.

rancheng, 2017-03-24:
Another issue you guys may face is:

fatal error: 'type_traits' file not found

and this is caused by using libstdc++, and libstdc++ does not support c++11 traits, in another word:
this is a dead end on Mac..

One more thing: **Just don't  waste your time on this problem, don't try your hard time frustrating on this platform, install a VM and run a linux, then find other problems to hack on.**

mehoggan, 2017-04-27:
Unfortunately, moving off mac for developers is not an option. All developers in our office use macs for 95% of their local development. We deploy to production using docker containers created from local debian VM's.

Also I am now starting to see this error on debian Jessie. I will post my compilation errors in a bit.

mehoggan, 2017-04-27:
`uber@ford-ct-5:~/graph-optimization/debian$ cat /etc/*release*
PRETTY_NAME="Debian GNU/Linux 8 (jessie)"
NAME="Debian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=debian
HOME_URL="http://www.debian.org/"
SUPPORT_URL="http://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
uber@ford-ct-5:~/graph-optimization/debian$ http://www.target.com/p/promark-photo-tripod/-/A-50626812^C
uber@ford-ct-5:~/graph-optimization/debian$ /usr/bin/c++ --version
c++ (Debian 4.9.2-10) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

Here are the linker errors on Linux:

`../lib/libgraph_optimization.so: undefined reference to `g2o::csparse_extension::cs_chol_workspace(cs_sparse const*, cs_symbolic const*, int*, double*)'
../lib/libgraph_optimization.so: undefined reference to `g2o::csparse_extension::writeCs2Octave(char const*, cs_sparse const*, bool)'
../lib/libgraph_optimization.so: undefined reference to `g2o::csparse_extension::cs_cholsolsymb(cs_sparse const*, double*, cs_symbolic const*, double*, int*)'`

I am going to work on a patch since this is now blocking us from using your library.

mehoggan, 2017-04-28:
For what its worth. When using/linking against the library in question from g2o and using the LinearSolverCSparse templated type. Explicit Template Instantiation (ETI) from a cpp file for linear_solver_csparse.h solves the problem. Below is the patch to apply to g2o if you are having the same problem. Because it uses ETI it is not a solution for everyone unless we new the set of all MatrixType(s) that would be used with the LinearSolverCSparse. This does appear to be a problem with certain tool chains. The patch below works on g++-4.9 in debian but was not tested in MacOSX. I would assume that it would work there since forcing the type to be compiled into the library insure the symbols are placed in the shared library.

```
From cc703029f3ca366906f05748171235303ace4336 Mon Sep 17 00:00:00 2001
From: Matthew Hoggan <mhoggan@uber.com>
Date: Fri, 28 Apr 2017 21:22:18 +0000
Subject: [PATCH] Explicit template instantiation.

---
 g2o/solvers/csparse/CMakeLists.txt            |   3 +-
 g2o/solvers/csparse/linear_solver_csparse.cpp | 335 ++++++++++++++++++++++++++
 g2o/solvers/csparse/linear_solver_csparse.h   | 288 ++--------------------
 3 files changed, 357 insertions(+), 269 deletions(-)
 create mode 100644 g2o/solvers/csparse/linear_solver_csparse.cpp

diff --git a/g2o/solvers/csparse/CMakeLists.txt b/g2o/solvers/csparse/CMakeLists.txt
index 20728d1..9861be5 100644
--- a/g2o/solvers/csparse/CMakeLists.txt
+++ b/g2o/solvers/csparse/CMakeLists.txt
@@ -15,14 +15,13 @@ ENDIF()
 
 ADD_LIBRARY(solver_csparse ${G2O_LIB_TYPE}
   solver_csparse.cpp
-  linear_solver_csparse.h
+  linear_solver_csparse.cpp
   g2o_csparse_api.h
 )
 SET_TARGET_PROPERTIES(solver_csparse PROPERTIES OUTPUT_NAME ${LIB_PREFIX}solver_csparse)
 
 TARGET_LINK_LIBRARIES(solver_csparse csparse_extension core)
 
-
 INSTALL(TARGETS solver_csparse csparse_extension
   RUNTIME DESTINATION ${CMAKE_INSTALL_PREFIX}/bin
   LIBRARY DESTINATION ${CMAKE_INSTALL_PREFIX}/lib
diff --git a/g2o/solvers/csparse/linear_solver_csparse.cpp b/g2o/solvers/csparse/linear_solver_csparse.cpp
new file mode 100644
index 0000000..7266a27
--- /dev/null
+++ b/g2o/solvers/csparse/linear_solver_csparse.cpp
@@ -0,0 +1,335 @@
+// g2o - General Graph Optimization
+// Copyright (C) 2011 R. Kuemmerle, G. Grisetti, W. Burgard
+// All rights reserved.
+//
+// Redistribution and use in source and binary forms, with or without
+// modification, are permitted provided that the following conditions are
+// met:
+//
+// * Redistributions of source code must retain the above copyright notice,
+//   this list of conditions and the following disclaimer.
+// * Redistributions in binary form must reproduce the above copyright
+//   notice, this list of conditions and the following disclaimer in the
+//   documentation and/or other materials provided with the distribution.
+//
+// THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS
+// IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED
+// TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A
+// PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT
+// HOLDER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
+// SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED
+// TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR
+// PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF
+// LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING
+// NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS
+// SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
+
+#include "g2o/solvers/csparse/linear_solver_csparse.h"
+
+#include "g2o/core/block_solver.h"
[... 609 lines cut ...]
-          delete[] _ccsA->x;
-          delete[] _ccsA->i;
-          _ccsA->i = new int[_ccsA->nzmax];
-          _ccsA->x = new double[_ccsA->nzmax];
-        }
-      }
-      _ccsA->m = m;
-      _ccsA->n = n;
+    void computeSymbolicDecomposition(const SparseBlockMatrix<MatrixType>& A);
 
-      if (onlyValues) {
-        this->_ccsMatrix->fillCCS(_ccsA->x, true);
-      } else {
-        int nz = this->_ccsMatrix->fillCCS(_ccsA->p, _ccsA->i, _ccsA->x, true); (void) nz;
-        assert(nz <= _ccsA->nzmax);
-      }
-      _ccsA->nz=-1; // tag as CCS formatted matrix
-    }
+    void fillCSparse(const SparseBlockMatrix<MatrixType>& A, bool onlyValues);
 };
 
 } // end namespace
-- 
2.1.4
```

closed 2025-02-12 by github-actions[bot]