RainerKuemmerle/g2o #153: Segmentation fault when optimize is called

arihantb2, 2017-02-06:
Program received signal SIGSEGV, Segmentation fault in g2o::SparseOptimizer::optimize(int, bool) ().

I have verified that the vertices and edges are added correctly by placing checks in my code. I have also checked the number of vertices and edges before calling the optimize function.

arihantb2, 2017-02-06:
[image]

arihantb2, 2017-02-06:
The program seg faults when I call g2o::SparseOptimizer::clear() also. For some reason, it is not able to reference the vertices and edges

closed 2017-02-07 by arihantb2

corenel, 2017-06-16:
@arihantb2 
Could you tell me how you solve this problem?
I've met this situation too.