RainerKuemmerle/g2o #156: ROS package libg2o has no Config.cmake

skasperski, 2017-02-08:
If I understand it correctly, every ROS package should have a *Config.cmake (and some other files that get generated automatically) in a place like `/opt/ros/kinetic/share/libg2o/cmake/`. It is currently not found if I write this to my package's CMakeLists:

`find_package(catkin REQUIRED COMPONENTS libg2o)`

Am I missing anything?

mikepurvis, 2017-03-07:
G2O isn't a catkin package— it's a regular CMake package, so the regular CMake rules for how to use it apply, see:

https://cmake.org/Wiki/CMake:How_To_Find_Libraries#Using_external_libraries

skasperski, 2017-03-07:
So I have to add `libg2o` to the manifest, then add the FindG2O.cmake from this repository to my own package and add

`find_package(G2O REQUIRED)`

to the CMakeLists.txt?

felixendres, 2017-03-08:
It worked for me for my rgbdslam project. The relevant parts should be
 * [set module path](https://github.com/felixendres/rgbdslam_v2/blob/indigo/CMakeLists.txt#L53)
 * [find_package g2o](https://github.com/felixendres/rgbdslam_v2/blob/indigo/CMakeLists.txt#L90)
 * an adapted [FindG2O.cmake](https://github.com/felixendres/rgbdslam_v2/blob/indigo/cmake-modules/FindG2O.cmake)

skasperski, 2017-03-08:
This  works in parts. Could you explain what are the relevant changes to the FindG2O.cmake? It finds the installation, but sets the include-path incorrectly, e.g.:

`Found g2o headers in: /opt/ros/kinetic/include/g2o`

which doesn't work, because g2o includes look like `#include g2o/core/sparse_optimizer.h`.

felixendres, 2017-03-08:
Sorry, it is too long ago that I adapted that script and I can't find the original source.
You might try to remove the first PATH_SUFFIX in line 23 to get correct paths.

skasperski, 2017-03-09:
It seems that I had some cached cmake-values which made it look like it worked. If I completely rerun cmake it still does not find g2o.

I tried again with the original [FindG2O.cmake](https://github.com/RainerKuemmerle/g2o/blob/master/cmake_modules/FindG2O.cmake) and it now works if I remove all the NO_DEFAULT_PATH entries. Looks like catkin adds the correct root directory to the search path automatically.

zainmehdi, 2017-03-20:
@skasperski I am also having the same issue. I want to use g2o under ros so that I can include publisher and subscriber using ros header files but I am unable to do so. Can you please provide some hints.

skasperski, 2017-03-24:
@zainmehdi First you add libg2o as run_dependency and build_dependency to your project's manifest so that g2o is installed. 

Add this modified cmake/FindG2O.cmake to your project:
```
# Find the header files
FIND_PATH(G2O_INCLUDE_DIR g2o/core/base_vertex.h)

# Macro to unify finding both the debug and release versions of the
# libraries; this is adapted from the OpenSceneGraph FIND_LIBRARY
# macro.
MACRO(FIND_G2O_LIBRARY MYLIBRARY MYLIBRARYNAME)

  FIND_LIBRARY("${MYLIBRARY}_DEBUG" "g2o_${MYLIBRARYNAME}_d")
  FIND_LIBRARY("${MYLIBRARY}_DEBUG" "g2o_${MYLIBRARYNAME}_d")
  FIND_LIBRARY(${MYLIBRARY} "g2o_${MYLIBRARYNAME}")
  FIND_LIBRARY(${MYLIBRARY} "g2o_${MYLIBRARYNAME}")
  
  IF(NOT ${MYLIBRARY}_DEBUG)
    IF(MYLIBRARY)
      SET(${MYLIBRARY}_DEBUG ${MYLIBRARY})
    ENDIF(MYLIBRARY)
  ENDIF( NOT ${MYLIBRARY}_DEBUG)
  
ENDMACRO(FIND_G2O_LIBRARY LIBRARY LIBRARYNAME)

# Find the core elements
FIND_G2O_LIBRARY(G2O_STUFF_LIBRARY stuff)
FIND_G2O_LIBRARY(G2O_CORE_LIBRARY core)

# Find the CLI library
FIND_G2O_LIBRARY(G2O_CLI_LIBRARY cli)

# Find the pluggable solvers
FIND_G2O_LIBRARY(G2O_SOLVER_CHOLMOD solver_cholmod)
FIND_G2O_LIBRARY(G2O_SOLVER_CSPARSE solver_csparse)
FIND_G2O_LIBRARY(G2O_SOLVER_CSPARSE_EXTENSION csparse_extension)
FIND_G2O_LIBRARY(G2O_SOLVER_DENSE solver_dense)
FIND_G2O_LIBRARY(G2O_SOLVER_PCG solver_pcg)
FIND_G2O_LIBRARY(G2O_SOLVER_SLAM2D_LINEAR solver_slam2d_linear)
FIND_G2O_LIBRARY(G2O_SOLVER_STRUCTURE_ONLY solver_structure_only)
FIND_G2O_LIBRARY(G2O_SOLVER_EIGEN solver_eigen)

# Find the predefined types
FIND_G2O_LIBRARY(G2O_TYPES_DATA types_data)
FIND_G2O_LIBRARY(G2O_TYPES_ICP types_icp)
FIND_G2O_LIBRARY(G2O_TYPES_SBA types_sba)
FIND_G2O_LIBRARY(G2O_TYPES_SCLAM2D types_sclam2d)
FIND_G2O_LIBRARY(G2O_TYPES_SIM3 types_sim3)
FIND_G2O_LIBRARY(G2O_TYPES_SLAM2D types_slam2d)
FIND_G2O_LIBRARY(G2O_TYPES_SLAM3D types_slam3d)

# G2O solvers declared found if we found at least one solver
SET(G2O_SOLVERS_FOUND "NO")
IF(G2O_SOLVER_CHOLMOD OR G2O_SOLVER_CSPARSE OR G2O_SOLVER_DENSE OR G2O_SOLVER_PCG OR G2O_SOLVER_SLAM2D_LINEAR OR G2O_SOLVER_STRUCTURE_ONLY OR G2O_SOLVER_EIGEN)
  SET(G2O_SOLVERS_FOUND "YES")
ENDIF(G2O_SOLVER_CHOLMOD OR G2O_SOLVER_CSPARSE OR G2O_SOLVER_DENSE OR G2O_SOLVER_PCG OR G2O_SOLVER_SLAM2D_LINEAR OR G2O_SOLVER_STRUCTURE_ONLY OR G2O_SOLVER_EIGEN)

# G2O itself declared found if we found the core libraries and at least one solver
SET(G2O_FOUND "NO")
IF(G2O_STUFF_LIBRARY AND G2O_CORE_LIBRARY AND G2O_INCLUDE_DIR AND G2O_SOLVERS_FOUND)
  SET(G2O_FOUND "YES")
ENDIF(G2O_STUFF_LIBRARY AND G2O_CORE_LIBRARY AND G2O_INCLUDE_DIR AND G2O_SOLVERS_FOUND)
```

Then you have to use g2o like a regular CMake package by adding the following lines to your CMakeLists.txt:
```
list(APPEND CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake)
find_package(G2O REQUIRED)

include_directories(
    ...
    ${G2O_INCLUDE_DIR}
)
...
target_link_libraries(my_lib
    ...
    ${G2O_CORE_LIBRARY}
    ${G2O_OTHER_LIB_YOU NEED}
)
```

zainmehdi, 2017-03-26:
@skasperski  thanks alot :)

Bradysm, 2018-03-29:
@skasperski I've added your FindG2O.cmake to cmake_modules in my rgbdslam package but it still does not work. My package.xml has <build_depend>libg2o</build_depend> and  <run_depend>libg2o</run_depend> in it. My relevent code in CMakeLists.txt looks like 
```
# g2o #######################

#set(libg2o_DIR "/home/astrobotics/g2ofork/lib/")
list(APPEND CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake-modules)
find_package(G2O REQUIRED)

include_directories(${G2O_INCLUDE_DIRS})
link_directories(${G2O_LIBRARY_DIRS})
link_libraries(${G2O_LIBRARIES})
message("G2o-libraries " ${G2O_LIBRARIES})
target_link_libraries(rgbdslam ${G2O_LIBRARIES})

```
Would you have any suggestions?

zainmehdi, 2018-03-29:
@Bradysm Actually I didnt try the approach mentioned above. Whats the error message you are getting ?

HugoNip, 2020-05-25:
> So I have to add `libg2o` to the manifest, then add the FindG2O.cmake from this repository to my own package and add
> 
> `find_package(G2O REQUIRED)`
> 
> to the CMakeLists.txt?
> 
> Update:
> I just tried this out and it doesn't seem to work. G2O is not found although it is installed.

try "find_package(g2o REQUIRED)", it's OK.

closed 2025-02-12 by github-actions[bot]