RainerKuemmerle/g2o #158: how to build against libg2o with non-aligned eigen code?

v01d, 2017-02-10:
I have a code which uses G2O and also instantiates eigen types for its own use. When I enabled -march=native on my build, I started getting the not aligned assertion from eigen. Since I do not care about vectorization of my code, I added EIGEN_DONT_VECTORIZE. The problem with this is that I get "memory corruption" error which comes from the "new" operator of a G2O type which has the EIGEN_MAKE_ALIGNED_OPERATOR_NEW declaration.

While I could just drop -march=native, this is also problematic on ARM where I get the same problem even without this optimization flag. Presumably, Eigen vectorizes on ARM by default and thus I cannot avoid this.

Any help would be appreciated.

diophantine86, 2017-04-18:
have you solved this problem?

v01d, 2017-04-18:
Hi,
I "solved" it by compiling on x86_64 without march=native, which I guess
enables vectorization and thus requires your code to be aligned as pero
eigen guidelines and, if not, crashes. On ARM it seems that vectorization
is enabled with default flags and I had to directly recompile G2O with
EIGEN_DONT_VECTORIZE (and also specify it while compiling my code).

diophantine86, 2017-04-18:
I wanna run my slam code on raspberry, but it occurs a error said that "aligned assertion". I guess that I maybe modified g2o source code to fix the errors. thanks any way

RainerKuemmerle, 2021-02-06:
Be sure to compile g2o library and binaries using it with the same compiler flags. In particular, drop -march=native if not needed to achieve a better portability of the library.

closed 2021-02-06 by RainerKuemmerle