RainerKuemmerle/g2o #161: EDGE_SE3_TRACKXYZ parameters

OtacilioNeto, 2017-03-03:
Dears, could you please answer a question about EDGE_SE3_TRACKXYZ?

What is the third parameter of EDGE_SE3_TRACKXYZ in a .g2o file?

I understood that the:
first is pose id
second is point id
third ===> What is this parameter?
4-6 distance from pose to point
7-12 upper triangular information matrix   

Thank you very much

felixendres, 2017-03-05:
Check out  [the source](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/types/slam3d/edge_se3_pointxyz.cpp). In the read method of the edge you can see that it is the id of a parameter (usually at the top of your .g2o file). The parameter is a constant offset between the vertex position (e.g., the camera base) and the point from where the measurement is made (e.g., the optical center).

OtacilioNeto, 2017-03-05:
Ok. Thanks a lot!

closed 2017-03-05 by OtacilioNeto