RainerKuemmerle/g2o #165: Are there any multiple edge tutorial? a multiple edge use error?

rangerlcy, 2017-03-14:
Hi,
I have a new optimization problem. I think I need a multiple edge struct which link three vertex to optimize. Specifically, 1st and 2nd vertices are g2o::VertexSE3. 3rd vertex is a real number, so I think g2o::BaseVertex<1, double> is good. Meanwhile, I have some prior terms and I think I should use g2o::BaseUnaryEdge<1, double, MyRealNumberVertexType>.

In conclusion, I inherit 3 class. Respectively,  
`class MyRealNumberVertexType : public g2o::BaseVertex<1, double>`
To represent a real number variable.

`class MyPriorEdge : public g2o::BaseUnaryEdge<1, double, MyRealNumberVertexType >`
As a weight to link MyRealNumberVertexType

`class MyMultiEdge: public g2o::BaseMultiEdge<6, Eigen::Isometry3d>`
To link three vertices, two g2o::VertexSE3 and one MyRealNumberVertexType. By the way, this edge's measurement is a rigid transformation in 3 dimension world.

When I try optimize() after initializeOptimization(). I get a assert crash

> Assertion failed: (!(RowsAtCompileTime!=Dynamic) || (nbRows==RowsAtCompileTime)) && (!(ColsAtCompileTime!=Dynamic) || (nbCols==ColsAtCompileTime)) && (!(RowsAtCompileTime==Dynamic && MaxRowsAtCompileTime!=Dynamic) || (nbRows<=MaxRowsAtCompileTime)) && (!(ColsAtCompileTime==Dynamic && MaxColsAtCompileTime!=Dynamic) || (nbCols<=MaxColsAtCompileTime)) && nbRows>=0 && nbCols>=0 && "Invalid sizes when resizing a matrix or array.", file eigen\eigen3\eigen\src\core\plainobjectbase.h, line 241

After debugging,  I find the problem comes from 
```
//in g2o/core/sparse_block_matrix.hpp line 101
_block=new typename SparseBlockMatrix<MatrixType>::SparseMatrixBlock(rb,cb);
```

The assert is triggered because rb=1, cb=1, but MatrixType is Eigen::Matrix<double,6,6,0,6,6> when program is running. That is it try create a SparseMatrixBlock which col=1 and row =1, while MatrixType is 6x6 matrix.

My question is why the template MatrixType is not change to Eigen::Matrix<double,1,1,0,1,1>? Not only do I have g2o::VertexSE3 vertex, but I also have real number vertex<1, double>.

Where is my wrong?

Any help would be appreciated.

felixendres, 2017-03-14:
I read the code you posted, and I see no mistake.

It is not a tutorial, but in [this tarball](http://ais.informatik.uni-freiburg.de/projects/datasets/catadioptric-rgbd/rgbdslam-catadioptric.tar.gz) you'll find some MultiEdges I defined in the files g2o_edge_*, where g2o_edge_triple_se3.* is easy to understand (given that you already know how to create own edges).

rangerlcy, 2017-03-14:
I have read your _g2o_edge_triple_se3._, and noticed your multiple edge link three _VertexSE3_. That's different from _MyMultiEdge_. My multiple edge is a special edge which link two _VertexSE3_ and one real number(that is to say, it herit from _g2o::BaseVertex<1, double>_). 
Moreover, I hope the three jacobian matrices' dimension are 
```
_jacobianOplus[0].resize(6,6);
_jacobianOplus[1].resize(6,6);
_jacobianOplus[2].resize(6,1); 
```

So, I want to know whether the class BaseMultiEdge in BaseMultiEdge support link different type vertices or not?

By the way, why didn't you initialize the dimension of jacobian matrix? Will they be initialized a right dimension? As far as I know, jacobian matrix default dimension is 0x0, and the resize() method is
```
template <int D, typename E>
void BaseMultiEdge<D, E>::resize(size_t size)
{
  BaseEdge<D,E>::resize(size);
  int n = (int)_vertices.size();
  int maxIdx = (n * (n-1))/2;
  assert(maxIdx >= 0);
  _hessian.resize(maxIdx);
  _jacobianOplus.resize(size, JacobianType(0,0,0));
}
```

rangerlcy, 2017-03-14:
I have read your g2o_edge_triple_se3., and noticed your multiple edge link three VertexSE3. That's different from MyMultiEdge. My multiple edge is a special edge which link two VertexSE3 and one real number(that is to say, it herit from g2o::BaseVertex<1, double>).

So, I wonder whether the class BaseMultiEdge in BaseMultiEdge support link different type vertices or not?

Here is my multiple edge implement code
```
#include <g2o/types/slam3d/vertex_se3.h>
#include <g2o/types/slam3d/edge_se3.h>
#include <g2o/core/base_multi_edge.h>
#include <g2o/core/hyper_graph_action.h>
#include <g2o/types/slam3d/isometry3d_gradients.h>
#include "MyRealNumberVertexType.h"

class MyMultiEdge: public g2o::BaseMultiEdge<6, Eigen::Isometry3d>
{
  public:
    MyMultiEdge();

    virtual bool read(std::istream& is);
    virtual bool write(std::ostream& os) const;
    void computeError();
    void linearizeOplus();

    virtual void setMeasurement(const Eigen::Isometry3d& m){
      _measurement = m;
      _inverseMeasurement = m.inverse();
    }

  protected:
    Eigen::Isometry3d _inverseMeasurement;
};

using namespace std;
using namespace Eigen;

// ================================================
MyMultiEdge::MyMultiEdge() : g2o::BaseMultiEdge<6, Eigen::Isometry3d>()
{
	resize(3);
	//_jacobianOplus[0].resize(6,6);
	//_jacobianOplus[1].resize(6,6);
	//_jacobianOplus[2].resize(6,1);

}
// ================================================
bool MyMultiEdge::read(std::istream& is)
  {    
    g2o::Vector7d meas;
    for (int i=0; i<7; i++) 
      is >> meas[i];
    // normalize the quaternion to recover numerical precision lost by storing as human readable text
    Vector4d::MapType(meas.data()+3).normalize();
    setMeasurement(g2o::internal::fromVectorQT(meas));

    for (int i=0; i<6; i++)
      for (int j=i; j<6; j++) {
        is >> information()(i,j);
        if (i!=j)
          information()(j,i) = information()(i,j);
      }
    return true;

  }
// ================================================
bool MyMultiEdge::write(std::ostream& os) const
{
    g2o::Vector7d meas = g2o::internal::toVectorQT(measurement());
    for (int i=0; i<7; i++) os  << meas[i] << " ";
    for (int i = 0; i < 6; ++i)
      for (int j = i; j < 6; ++j)
        os << " " << information()(i, j);
    return os.good();
}

// ================================================
void MyMultiEdge::linearizeOplus()
{

    g2o::VertexSE3* from = static_cast<g2o::VertexSE3*>(_vertices[0]);
    g2o::VertexSE3* to = static_cast<g2o::VertexSE3*>(_vertices[1]);
    const MyRealNumberVertexType* v= static_cast<const MyRealNumberVertexType*>(_vertices[2]);

    Eigen::Isometry3d E;
    const Eigen::Isometry3d& Xi=from->estimate();
    const Eigen::Isometry3d& Xj=to->estimate();
    const Eigen::Isometry3d& Z=_measurement;
    g2o::internal::computeEdgeSE3Gradient(E, _jacobianOplus[0], _jacobianOplus[1], Z, Xi, Xj);

    _jacobianOplus[0]*=v->estimate();
    _jacobianOplus[1]*=v->estimate();

    // derivative real number vertex
    _jacobianOplus[2].setZero();
    _jacobianOplus[2] = g2o::internal::toVectorMQT(E) * v->gradient();
}

// ================================================
void MyMultiEdge::computeError()
{
    const g2o::VertexSE3* v1 = dynamic_cast<const g2o::VertexSE3*>(_vertices[0]);
    const g2o::VertexSE3* v2 = dynamic_cast<const g2o::VertexSE3*>(_vertices[1]);
    const MyRealNumberVertexType* v3 = static_cast<const MyRealNumberVertexType*>(_vertices[2]);

    Eigen::Isometry3d delta = _inverseMeasurement * (v1->estimate().inverse()*v2->estimate());
    _error = g2o::internal::toVectorMQT(delta) * v3->estimate();
}
```

felixendres, 2017-03-14:
I never had to resize the jacobians. Since the dimension of the measurement is known and the dimension of the vertices are known, g2o can infer the size of the Jacobian.
See the linearizeOplus method of base_multi_edge.hpp, where a block of the appropriate size is assigned for the blocks of the Jacobian that correspond to the individual vertices of the multi-edge.

rangerlcy, 2017-03-14:
Yes, you are right. I have modified my comment as above.  Please read.

felixendres, 2017-03-14:
Looks good to me. There are two things that could cause the error, but I can't check:
- Are you assigning the vertices in the right order? You use static_casts at some places, which have no runtime check.
- Is g2o:internal::tovectorMQT(E) * v->gradient() of the right dimensionality, i.e., 6x1? Or is it just 6, because v->gradient is not a 1x1 Matrix but just a double?

rangerlcy, 2017-03-15:
For 1, all vertices are right order.
For 2, _g2o:internal::tovectorMQT(E)_ return a _Vector6d. It is a 6x1 vector_, and _v->gradient_ return a double. So _dimension of g2o:internal::tovectorMQT(E) * v->gradient() is 6x1. It is right._

I need to say, my program crashed when it tried initialize block matrix because of dimension consistency. In other word, it hasn't run to _linearizeOplus() or computeError()_

So I wonder when g2o infer MatrixType as Eigen::Matrix<6,6,0,6,6> in g2o/core/sparse_block_matrix.hpp line 101
`_block=new typename SparseBlockMatrix<MatrixType>::SparseMatrixBlock(rb,cb);`

rangerlcy, 2017-03-15:
After my thorough diagnosis, I finally found my mistake.  The mistake comes from _linearSolver_ selection. formerly, I use _SlamLinearCSparseSolver_ whose block matrices are _Eigen::Matrix<6,6,0,6,6>_. So I modified _SlamLinearCSparseSolver_ to _g2o::LinearSolverCSparse<g2o::BlockSolverX::PoseMatrixType>_. Now my program can run normally.

closed 2017-03-15 by RainerKuemmerle

felixendres, 2018-12-01:
Hi @rangerlcy, I could just solve my one-day long bug-hunt with your last comment :-D. I had this weird problem with a tripple-edge, where the second vertex wasn't updated at all. Some cholmod errors happend, but no crash. Optimization even worked fine in g2o cli and viewer. After hours of experimenting, verifying that it is not fixed anywhere, checking that the Jacobians were fine, etc, I searched the repository issues and found our old conversation.

I guess that's karma for trying to help you last year :-).

Many thanks for posting the answer to your own question.

YangSiri, 2019-10-30:
> After hours of experimenting, verifying that it is not fixed anywhere, checking that the Jacobians were fine, etc, I searched the repository issues and found our old conversation.

Hi, @felixendres , so you were saying there **must** be a vertex fixed in the whole graph?  
If i am defining a new mulitedge which actually has **dynamic** number of **pose vertexs**,  how many functions should i redefine (like _computeError()_ etc) ? In addition, the _error_ type is not the same type as _measurement_, is this okay?

JunsukLee, 2022-06-13:
This issue has been very helpful to me.
Could you tell me how to register for each vertex?
Currently, in the case of basebinaryedge(), I use setVertex() to register.

Similarly, basemultiedge() tried to register using setvertex().

But I get a "Segmentation fault (core dumped)" error.

I wonder if it is a registration problem or if there is another usage.

If you are using it please help me!!