RainerKuemmerle/g2o #166: run the code of optimizer.addParameter ( camera ), program break down!

AlwaysLearningJary, 2017-03-24:
When I run the code "optimizer.addParameter ( camera );", my program is broken down.
Here is the log:
*** Error in `./pose_estimation_3d2d': malloc(): memory corruption: 0x000000000114d450 ***
Aborted (core dumped)
the code:
// parameter: camera intrinsics
    g2o::CameraParameters* camera = new g2o::CameraParameters (
        K.at<double> ( 0,0 ), Eigen::Vector2d ( K.at<double> ( 0,2 ), K.at<double> ( 1,2 ) ), 0
    );
    camera->setId ( 0 );
    cout<<"sfsfsf"<<endl; // This tip can output to the terminal.
    optimizer.addParameter ( camera ); // can't run this code.

I can't solve it.

joesparza, 2017-04-04:
I have the same problem, could you solve it?

maShimizu, 2018-02-25:
I have same problem, 
Have this issue already been closed?

the code:
g2o::ParameterSE3Offset* cameraOffset = new g2o::ParameterSE3Offset;
cameraOffset.setId(0);
optimizer.addParameter(cameraOffset)

maShimizu, 2018-09-27:
> I have the same question too.Can anyone fix this bug?

I can’t fix yet. 
And alternatively I implement optimization using ceres library released by google.

MichaelRuhnke, 2018-10-20:
Hi,
I did run into the same problem, it is an actual problem with Eigen memory alignment and having different SSE/AVX compile options for g2o and the the code that links against it.

g2o per default enables "-march native" which enables all instruction subsets supported by the local machine (according to https://gcc.gnu.org/onlinedocs/gcc-4.5.3/gcc/i386-and-x86_002d64-Options.html)
On machines with AVX or AVX2 like mine the code will be compiled with -mavx2 which disables all -msse1/2/3/4 compile options and replaces it with the corresponding avx vectorization code.

So either compile your code with "-march native" or disable this option in g2o:

`diff --git a/CMakeLists.txt b/CMakeLists.txt
index cc5cd3d..8d5d5f4 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -254,7 +254,7 @@ endif()
 
 # Compiler specific options for gcc
 if(CMAKE_COMPILER_IS_GNUCXX)
-  option (BUILD_WITH_MARCH_NATIVE "Build with \"-march native\"" ON)
+  option (BUILD_WITH_MARCH_NATIVE "Build with \"-march native\"" OFF)
   message(STATUS "Compiling with GCC")
 
   # Generic settings for optimisation
`

Will discuss with Rainer if it makes sense to have the option as OFF...

closed 2025-02-11 by github-actions[bot]