RainerKuemmerle/g2o #169: Tutorial output could not be loaded in g2o_viewer

olafbooij, 2017-03-27:
Hi!
The tutorial_slam2d code is I think a great place to start understanding how to use g2o. However, it seems that the .g2o files that it outputs cannot directly be visualized in the g2o_viewer application, because the tags that are used are not known by the g2o_viewer. This probably is because the tutorial example is implemented as a standalone example, not using the types defined in types/slam2d/* .
This is a bit confusing when starting to understand g2o.

I wonder if it is somehow possible to fix this issue. Perhaps a bit of a rude way is by registering the tutorial types as regular types as in the pull-request I just sent:
https://github.com/RainerKuemmerle/g2o/pull/168

However, perhaps this creates an unwanted dependency on types/slam2d ...

RainerKuemmerle, 2017-04-10:
With 2e3566968c1f0e5608155a2a26154b98d6d752dc I added the option for renaming types on the g2o_viewer command line. So you may use something like:
`$ bin/g2o_viewer -renameTypes TUTORIAL_VERTEX_SE2=VERTEX_SE2,TUTORIAL_VERTEX_POINT_XY=VERTEX_XY,TUTORIAL_PARAMS_SE2_OFFSET=PARAMS_SE2OFFSET,TUTORIAL_EDGE_SE2=EDGE_SE2,TUTORIAL_EDGE_SE2_POINT_XY=EDGE_SE2_XY tutorial_after.g2o`

Hope this helps.

closed 2017-04-10 by RainerKuemmerle

FaiScofield, 2019-12-04:
> @RainerKuemmerle Besides that, if I define a new type of vertex or edge, how could I use g2o_viewer to show them? I have defined the read and write function.

Did you have the answer of your problem? cause I was dealing with this too now.