RainerKuemmerle/g2o #170: Cross-compiling error for android

liming0791, 2017-03-29:
When I cross-compile for android:

```
 cmake -DANDROID_NATIVE_API_LEVEL=android-19 -DCMAKE_TOOLCHAIN_FILE=../script/android.toolchain.cmake -DANDROID_NDK=~/Projects/android-ndk-r13b/ -DCMAKE_BUILD_TYPE=Release -DANDROID_ABI="armeabi-v7a" -DEIGEN3_INCLUDE_DIR=/usr/include/eigen3 -DEIGEN3_VERSION_OK=ON .. && cmake --build .
```

I got such errors:

```
[ 79%] Building CXX object g2o/solvers/eigen/CMakeFiles/solver_eigen.dir/solver_eigen.cpp.o
In file included from /home/liming/Projects/android-ndk-r13b/sources/cxx-stl/gnu-libstdc++/4.9/include/cctype:42:0,
                 from /home/liming/Projects/android-ndk-r13b/sources/cxx-stl/gnu-libstdc++/4.9/include/bits/localefwd.h:42,
                 from /home/liming/Projects/android-ndk-r13b/sources/cxx-stl/gnu-libstdc++/4.9/include/ios:41,
                 from /home/liming/Projects/android-ndk-r13b/sources/cxx-stl/gnu-libstdc++/4.9/include/istream:38,
                 from /home/liming/Projects/android-ndk-r13b/sources/cxx-stl/gnu-libstdc++/4.9/include/sstream:38,
                 from /home/liming/Projects/android-ndk-r13b/sources/cxx-stl/gnu-libstdc++/4.9/include/complex:45,
                 from /usr/include/eigen3/Eigen/Core:28,
                 from /usr/include/eigen3/Eigen/SparseCore:4,
                 from /usr/include/eigen3/Eigen/Sparse:19,
                 from /home/liming/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:30,
                 from /home/liming/Projects/g2o/g2o/solvers/eigen/solver_eigen.cpp:17:
/usr/include/eigen3/Eigen/src/SparseLU/SparseLU.h:222:61: error: expected ',' or '...' before numeric constant
     bool _solve(const MatrixBase<Rhs> &B, MatrixBase<Dest> &_X) const
                                                             ^
In file included from /usr/include/eigen3/Eigen/SparseLU:47:0,
                 from /usr/include/eigen3/Eigen/Sparse:22,
                 from /home/liming/Projects/g2o/g2o/solvers/eigen/linear_solver_eigen.h:30,
                 from /home/liming/Projects/g2o/g2o/solvers/eigen/solver_eigen.cpp:17:
/usr/include/eigen3/Eigen/src/SparseLU/SparseLU.h: In member function 'bool Eigen::SparseLU<_MatrixType, _OrderingType>::_solve(const Eigen::MatrixBase<OtherDerived>&, Eigen::MatrixBase<OtherDerived>&) const':
/usr/include/eigen3/Eigen/src/SparseLU/SparseLU.h:224:18: error: request for member 'derived' in '64', which is of non-class type 'int'
       Dest& X(_X.derived());

make[2]: *** [g2o/solvers/eigen/CMakeFiles/solver_eigen.dir/solver_eigen.cpp.o] Error 1
make[1]: *** [g2o/solvers/eigen/CMakeFiles/solver_eigen.dir/all] Error 2
make: *** [all] Error 2
```

It seems that the error comes from Eigen3 ?

Banus, 2017-09-21:
Seems an `Eigen3` bug. Using the last `g2o` and `Eigen3 `code from the mirror here:

    https://github.com/RLovelett/eigen

The error disappears and the compilation ends correctly.

zsychange0713, 2018-09-25:
do you success in compiling g2o on android plattom? I compiling it with android.mk,but many errors like "undefined reference to `cholmod_factorize'".Does it  appears when you compiling?Thank you

swxind, 2019-07-03:
> do you success in compiling g2o on android plattom? I compiling it with android.mk,but many errors like "undefined reference to `cholmod_factorize'".Does it appears when you compiling?Thank you

I get the same problem. do you solve it?

vyi, 2020-12-07:
@zzy0713  @swxind  I was able to cross-compile for Android using the **ndk** provided CMake Toolchain file. 

This file is located in the following path **Sdk/ndk/VERSION_NUMBER/build/cmake/android.toolchain.cmake**.

closed 2025-02-11 by github-actions[bot]