RainerKuemmerle/g2o #171: GPU support

csp256, 2017-04-01:
What would it take to add basic GPU support for the sparse solver backend? 

I have seen the "OpenFO" project but a ~2x speedup for switching to a GPU for linear solves did not inspire a lot of confidence in me.

akssri, 2017-04-04:
Actually ~2x speedup is close to the state-of-the-art for sparse solvers (this is what CHOLMOD gets on average).

http://on-demand.gputechconf.com/gtc/2016/presentation/s6387-steven-rennich-gpu-acceleration-cholmod.pdf

csp256, 2017-06-28:
Sorry for the late response. 

To what extent can the problem-specific form of bundle adjustment improve upon the speed up of general-purpose sparse solvers? Could additional constraints (say, maximal number of observations per camera, or maximal number of cameras) improve upon this?

somedadaism, 2021-08-04:
Have you seen this?
Appears to be substantially faster.
https://github.com/fixstars/cuda-bundle-adjustment

closed 2025-02-11 by github-actions[bot]