RainerKuemmerle/g2o #172: How to add constraints between 3D points?

aseyfi, 2017-04-24:
Hello,

I am using g2o for 3D SLAM. In my application, the location of some of the landmarks (3D points) in the environment are known to me, for example I know the Euclidean distance between some of the landmarks, or I know if they are located on the same plane (on a wall for example). I would like to use this extra information to improve the results of the pose graph optimization.
Would somebody please tell me how these constraints should be added? Are they currently part of the g2o implementation? Do I need to created my own custom g2o edges?

Thank You,
Ahmad

AlejandroSilvestri, 2017-06-06:
@aseyfi , I am working on the same problem.

I believe you have to create your own custom edges, and perhaps create your own custom vertex.

JunsukLee, 2022-05-23:
Could you give me a solution to this problem??

Or could you provide some reference material?

closed 2025-02-11 by github-actions[bot]