RainerKuemmerle/g2o #181: What the abbreviation SE2 stands for?

neumann89, 2017-06-28:
Hi!
You mention that SE2 is the position x,y and orientation theta. But why is it called "SE"? I couldn't find any explanation for it. Could you please add to the documentation and code comments what the abbreviation SE2 stands for?

felixendres, 2017-06-28:
It's an abbreviation for the [Special Euclidean group](https://en.wikipedia.org/wiki/Euclidean_group#Direct_and_indirect_isometriesl) for 2-dimensional space.

SE2 contains the rigid body motions in 2D.

csp256, 2017-06-28:
I suggest Tom Drummond's notes for learning more about Lie algebras / groups in computer vision:

https://dl.dropboxusercontent.com/u/23948930/Papers/3DGeometry.pdf

The relevant part begins in chapter 2, but chapter 1 is good too.

neumann89, 2017-06-30:
Thank you very much!

closed 2017-06-30 by neumann89