RainerKuemmerle/g2o #182: g2o library doesn't export suitesparse related include headers

efernandez, 2017-06-28:
In a client (user) code of this library I need to depend on suitesparse explicitly, `find_package` and add its `_INCLUDE_DIRS` because g2o has headers that include suitesparse headers, but it doesn't export the suitesparse `_INCLUDE_DIRS`.

E.g.:

If I use a `LinearSolverCSparse`, I need to `#include <g2o/solvers/csparse/linear_solver_csparse.h>`.
This header includes `csparse_helper.h`:
https://github.com/RainerKuemmerle/g2o/blob/master/g2o/solvers/csparse/linear_solver_csparse.h#L30
Which includes `cs.h` from suitesparse:
https://github.com/RainerKuemmerle/g2o/blob/master/g2o/solvers/csparse/csparse_helper.h#L23

Since `LinearSolverCSparse` it's a template class, it's not trivial (if possible) to avoid this situation.
Either way, that's not wrong, the problem is that because of this, any client (user) code using this, needs to add the suitesparse headers path. The g2o library should export/add that to its `cflags` (I think), so `-I` has the headers path (as `-L` for libraries); if I knew how to do it, I'd create a PR.

@RainerKuemmerle What do you think?

FYI @v-mehta

vyi, 2020-12-07:
I wonder if this has anything to do with issue [248](https://github.com/RainerKuemmerle/g2o/issues/248)?

cyberbeat, 2022-09-23:
Yes, I had the same problem, and succeeded to fix the cause of this:

```
target_include_directories(csparse_extension PUBLIC
  $<BUILD_INTERFACE:${CSPARSE_INCLUDE_DIR}>
```

should be transformed to

```
target_include_directories(csparse_extension PUBLIC
${CSPARSE_INCLUDE_DIR}
```

So that the include not only happens at compile time.

PR #603 (2022-10-22): Set include directory for csparse_extension
Fixes #182
commit 5b454cdc (2022-10-22): Set include directory for csparse_extension (#603)

Fixes #182
closed 2022-10-22 by RainerKuemmerle