RainerKuemmerle/g2o #183: Why does my code in different pc generate absolutely different result?

Xbbei, 2017-07-04:
I type the code :
typedef g2o::BlockSolver_6_3 SlamBlockSolver;
typedef g2o::LinearSolverEigen< SlamBlockSolver::PoseMatrixType > SlamLinearSolver;

SlamLinearSolver* linearSolver = new SlamLinearSolver();
    linearSolver->setBlockOrdering( false );
    SlamBlockSolver* blockSolver = new SlamBlockSolver( linearSolver );
    g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg( blockSolver );

g2o::SparseOptimizer global_optimizer;
    global_optimizer.setAlgorithm( solver );
    global_optimizer.setVerbose( true );

and I set the vertex and edge.
when I optimize the vertex, it will generate:
iteration= 0	 chi2= -1360.953352	 time= 0.00219926	 cumTime= 0.00219926	 edges= 338	 schur= 0	 lambda= 0.000702	 levenbergIter= 2
iteration= 1	 chi2= -5243787.642195	 time= 0.000626821	 cumTime= 0.00282608	 edges= 338	 schur= 0	 lambda= 0.000468	 levenbergIter= 1
iteration= 2	 chi2= -1139883233772.110107	 time= 0.000619442	 cumTime= 0.00344552	 edges= 338	 schur= 0	 lambda= 0.000312	 levenbergIter= 1
iteration= 3	 chi2= -230676478668081987584.000000	 time= 0.000619451	 cumTime= 0.00406497	 edges= 338	 schur= 0	 lambda= 0.000104	 levenbergIter= 1
iteration= 4	 chi2= -2762827043123604488192.000000	 time= 0.00129577	 cumTime= 0.00536074	 edges= 338	 schur= 0	 lambda= 0.000555	 levenbergIter= 3
iteration= 5	 chi2= -112210538037353322149031444480.000000	 time= 0.000619263	 cumTime= 0.00598001	 edges= 338	 schur= 0	 lambda= 0.000185	 levenbergIter= 1
iteration= 6	 chi2= -141887745666085148483537548083200.000000	 time= 0.00196507	 cumTime= 0.00794507	 edges= 338	 schur= 0	 lambda= 0.126261	 levenbergIter= 5
iteration= 7	 chi2= -35546445197473244135341774428981690368.000000	 time= 0.000622789	 cumTime= 0.00856786	 edges= 338	 schur= 0	 lambda= 0.042087	 levenbergIter= 1
iteration= 8	 chi2= -1071679862142315552281009490296001561362432.000000	 time= 0.00062098	 cumTime= 0.00918884	 edges= 338	 schur= 0	 lambda= 0.028058	 levenbergIter= 1
iteration= 9	 chi2= -177584672305472208640276568714645664699674263552.000000 time= 0.0006243	 cumTime= 0.00981314	 edges= 338	 schur= 0	 lambda= 0.018705	 levenbergIter= 1
iteration= 10	 chi2= -165268394404339768897452522289721898676951648342900736.000000	 time= 0.000622746	 cumTime= 0.0104359	 edges= 338	 schur= 0	 lambda= 0.006235	 levenbergIter= 1
iteration= 11	 chi2= -2895933140023918056812143978912142505070446292335627075584.000000	 time= 0.000620297	 cumTime= 0.0110562	 edges= 338	 schur= 0	 lambda= 0.004157	 levenbergIter= 1
iteration= 12	 chi2= -4922149417770155474598420122673418363173840381458581028864.000000	 time= 0.00227704	 cumTime= 0.0133332	 edges= 338	 schur= 0	 lambda= 90.805705	 levenbergIter= 6
iteration= 13	 chi2= -5445395392279090261027955042837020495752375451383213588480.000000	 time= 0.000627103	 cumTime= 0.0139603	 edges= 338	 schur= 0	 lambda= 60.537137	 levenbergIter= 1
iteration= 14	 chi2= -71915755510549890563354046171744714924165287255502479163392.000000	 time= 0.000625353	 cumTime= 0.0145857	 edges= 338	 schur= 0	 lambda= 40.358091	 levenbergIter= 1
iteration= 15	 chi2= -93923107928784117732051564474840097220133587439139524968448.000000	 time= 0.000633922	 cumTime= 0.0152196	 edges= 338	 schur= 0	 lambda= 26.905394	 levenbergIter= 1
iteration= 16	 chi2= -309874730397970037822614057174958376393928862898083927687168.000000	 time= 0.00064005	 cumTime= 0.0158597	 edges= 338	 schur= 0	 lambda= 17.936929	 levenbergIter= 1
iteration= 17	 chi2= -881638885319915429278543145156262560365463333678101815099392.000000	 time= 0.000729959	 cumTime= 0.0165896	 edges= 338	 schur= 0	 lambda= 11.957953	 levenbergIter= 1
iteration= 18	 chi2= -15225118934136855989123706546809742128208259578815801316081664.000000	 time= 0.000692026	 cumTime= 0.0172816	 edges= 338	 schur= 0	 lambda= 7.971969	 levenbergIter= 1
iteration= 19	 chi2= -26585850622731453927225996746239761305082996935097149571465216.000000	 time= 0.000671697	 cumTime= 0.0179533	 edges= 338	 schur= 0	 lambda= 5.314646	 levenbergIter= 1
iteration= 20	 chi2= -164135539491094298971569132624965748496514172904249913923076096.000000	 time= 0.000663844	 cumTime= 0.0186172	 edges= 338	 schur= 0	 lambda= 3.543097	 levenbergIter= 1
iteration= 21	 chi2= -118065512186452534464839453137482720005509361910164825565036544.000000	 time= 0.000637089	 cumTime= 0.0192543	 edges= 338	 schur= 0	 lambda= 2.362065	 levenbergIter= 1
iteration= 22	 chi2= -421907013712478342946171224977556918368784612012161825536737280.000000	 time= 0.00163703	 cumTime= 0.0208913	 edges= 338	 schur= 0	 lambda= 100.781431	 levenbergIter= 4
iteration= 23	 chi2= -1160043297640683147081255257437290121156547456674128844982059008.000000	 time= 0.00063706	 cumTime= 0.0215284	 edges= 338	 schur= 0	 lambda= 67.187621	 levenbergIter= 1
iteration= 24	 chi2= -12789662780265806041454290204654813436896088375249986240999587840.000000	 time= 0.000636525	 cumTime= 0.0221649	 edges= 338	 schur= 0	 lambda= 44.791747	 levenbergIter= 1
iteration= 25	 chi2= -19408631699723473988378234671825592255628092842118634587593637888.000000	 time= 0.000637692	 cumTime= 0.0228026	 edges= 338	 schur= 0	 lambda= 29.861165	 levenbergIter= 1
iteration= 26	 chi2= -10413022926408379816951249520395680031353539464156079798116606279680.000000	 time= 0.000640523	 cumTime= 0.0234431	 edges= 338	 schur= 0	 lambda= 19.907443	 levenbergIter= 1
iteration= 27	 chi2= -1700945364831958244105807232035949602891185458928682062607663541256192.000000	 time= 0.000638818	 cumTime= 0.0240819	 edges= 338	 schur= 0	 lambda= 13.271629	 levenbergIter= 1
iteration= 28	 chi2= -22352159655475619412053294537800590171165469074691016139128760269012992.000000	 time= 0.000637396	 cumTime= 0.0247193	 edges= 338	 schur= 0	 lambda= 4.423876	 levenbergIter= 1
iteration= 29	 chi2= -57324508896912452971799461177132408799089255654240522430531684951654400.000000	 time= 0.000639793	 cumTime= 0.0253591	 edges= 338	 schur= 0	 lambda= 2.949251	 levenbergIter= 1
iteration= 30	 chi2= -77709520497009773908047042925655897493071366418475128331078524290990080.000000	 time= 0.000639178	 cumTime= 0.0259983	 edges= 338	 schur= 0	 lambda= 1.966167	 levenbergIter= 1
iteration= 31	 chi2= -186021723539878015403033626468749377632277690109632422618219990673784832.000000	 time= 0.000638949	 cumTime= 0.0266372	 edges= 338	 schur= 0	 lambda= 1.310778	 levenbergIter= 1
iteration= 32	 chi2= -186021723539878015403033626468749377632277690109632422618219990673784832.000000	 time= 0.00362699	 cumTime= 0.0302642	 edges= 338	 schur= 0	 lambda= 47225759857589184.000000	 levenbergIter= 10

The same code,  generate the result:
iteration= 0     chi2= 0.001805     time= 0.0024395     cumTime= 0.0024395     edges= 337     schur= 0     lambda= 0.000797     levenbergIter= 1
iteration= 1     chi2= 0.000850     time= 0.000915046     cumTime= 0.00335455     edges= 337     schur= 0     lambda= 0.000531     levenbergIter= 1
iteration= 2     chi2= 0.000821     time= 0.00097252     cumTime= 0.00432707     edges= 337     schur= 0     lambda= 0.000354     levenbergIter= 1
iteration= 3     chi2= 0.000819     time= 0.000885697     cumTime= 0.00521276     edges= 337     schur= 0     lambda= 0.000236     levenbergIter= 1
iteration= 4     chi2= 0.000819     time= 0.000881078     cumTime= 0.00609384     edges= 337     schur= 0     lambda= 0.000157     levenbergIter= 1
iteration= 5     chi2= 0.000819     time= 0.000880661     cumTime= 0.0069745     edges= 337     schur= 0     lambda= 0.000105     levenbergIter= 1
iteration= 6     chi2= 0.000819     time= 0.000878046     cumTime= 0.00785255     edges= 337     schur= 0     lambda= 0.000070     levenbergIter= 1
iteration= 7     chi2= 0.000819     time= 0.00088279     cumTime= 0.00873534     edges= 337     schur= 0     lambda= 0.000047     levenbergIter= 1
iteration= 8     chi2= 0.000819     time= 0.000878271     cumTime= 0.00961361     edges= 337     schur= 0     lambda= 0.000031     levenbergIter= 1
iteration= 9     chi2= 0.000819     time= 0.0008773     cumTime= 0.0104909     edges= 337     schur= 0     lambda= 0.000021     levenbergIter= 1
iteration= 10     chi2= 0.000819     time= 0.00435921     cumTime= 0.0148501     edges= 337     schur= 0     lambda= 3707.817224     levenbergIter= 8
iteration= 11     chi2= 0.000819     time= 0.00191199     cumTime= 0.0167621     edges= 337     schur= 0     lambda= 19775.025197     levenbergIter= 3
iteration= 12     chi2= 0.000819     time= 0.00241291     cumTime= 0.019175     edges= 337     schur= 0     lambda= 20249625.801562     levenbergIter= 4

Thank you very much!

sjulier, 2017-07-04:
Normally I find that problems like this can be due to unintialised variables (the effects of these are notoriously arbitrary on different platforms). For example, if you don't fully assign all the values to the information matrix, some of the terms can be random. This, in turn, can lead to non PSD covariance matrices and hence negative chi2 values. Have you tried anything like valgrind to see if there's an issue?

Can you post the entire code if you don't see a problem there?

Xbbei, 2017-07-04:
Thank you very much! but I can't understand your meaning. So I post my the entire code.
Initial the optimizer:
typedef g2o::BlockSolver_6_3 SlamBlockSolver;
typedef g2o::LinearSolverEigen< SlamBlockSolver::PoseMatrixType > SlamLinearSolver;

SlamLinearSolver* linearSolver = new SlamLinearSolver();
linearSolver->setBlockOrdering( false );
SlamBlockSolver* blockSolver = new SlamBlockSolver( linearSolver );
g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg( blockSolver );

g2o::SparseOptimizer global_optimizer;
global_optimizer.setAlgorithm( solver );
global_optimizer.setVerbose( true );

I set the first vertex fixed, 
g2o::VertexSE3 *v = new g2o::VertexSE3();
v->setId( global_time );
v->setEstimate( T );    T is the Identity.
v->setFixed( true );
global_optimizer.addVertex(v);

g2o::VertexSE3 *v = new g2o::VertexSE3();
v->setId( global_time );
v->setEstimate( T1 );       T1 is the index global_time initial vertex.
global_optimizer.addVertex(v);

g2o::EdgeSE3* edge = new g2o::EdgeSE3();
edge->setVertex( 0, global_optimizer.vertex( global_time - 1 ));
edge->setVertex( 1, global_optimizer.vertex( global_time ));
edge->setRobustKernel( new g2o::RobustKernelHuber() );
Eigen::Matrix<double, 6, 6> information = Eigen::Matrix< double, 6,6 >::Identity();
information(0,0) = information(1,1) = information(2,2) = 100;
information(3,3) = information(4,4) = information(5,5) = 100;
edge->setInformation( information );

edge->setMeasurement( T1 );    T1 is the global_time - 1 and global_time relative position
global_optimizer.addEdge(edge);

and I will check the loop.
Last,
global_optimizer.initializeOptimization();
global_optimizer.optimize( 1000 );

Thanks!

sjulier, 2017-07-04:
Uninitialised variables arise in code such as:

double x;
double y = x + 10;

if (y > 5) { ... do something ... }

Because no initial value has been assigned to x, the C / C++ standard says its value is undefined. Therefore, it could be anything including mapping to bits of memory where other variables had been stored in the past. Quite often the value is 0. However, because the value is undefined, it can depend entirely on the whim of the operating system, compiler, architecture, ....

Hmm... I don't see anything immediately wrong. Would it be possible to attach the complete source code? That way, we can copy paste and try it.

Also, can you confirm:

1. Which version of g2o are you using?
2. What platform / OS / compiler you are trying this on?

Cheers,

Simon

Xbbei, 2017-07-04:
1. Sorry I don't know which version of g2o I am using, I download it recently.
2. I use the Ubuntu 16.04 Operation System, and compiler: SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall -msse -msse2 -msse3 -O3 -std=c++11")

My complete source code:

#include <iostream>
#include <fstream>
#include <string>
#include <vector>

#include <g2o/types/slam3d/types_slam3d.h>
#include <g2o/core/sparse_optimizer.h>
#include <g2o/core/block_solver.h>
#include <g2o/core/factory.h>
#include <g2o/core/optimization_algorithm_factory.h>
#include <g2o/core/optimization_algorithm_gauss_newton.h>
#include <g2o/solvers/eigen/linear_solver_eigen.h>
#include <g2o/core/robust_kernel.h>
#include <g2o/core/robust_kernel_impl.h>
#include <g2o/core/optimization_algorithm_levenberg.h>

#include <Eigen/Core>
#include <Eigen/Geometry>

typedef g2o::BlockSolver_6_3 SlamBlockSolver;
typedef g2o::LinearSolverEigen< SlamBlockSolver::PoseMatrixType > SlamLinearSolver;
int main()
{
    FILE * trajectory = fopen("pose.txt", "r");
    std::string temp;
    int i;

    std::vector<Eigen::Isometry3d,Eigen::aligned_allocator<Eigen::Isometry3d>> camera_trajectory;
    while(fscanf(trajectory, "%s %d \n", temp.c_str(), &i) != EOF)
    {
        Eigen::Isometry3d posetemp;
        Eigen::Matrix<float, 3, 3, Eigen::RowMajor> R;
        Eigen::Vector3f t;

        Eigen::Vector4f te;
        fscanf(trajectory, "%f %f %f %f \n", &R(0,0),&R(0,1),&R(0,2),&t(0));
        fscanf(trajectory, "%f %f %f %f \n", &R(1,0),&R(1,1),&R(1,2),&t(1));
        fscanf(trajectory, "%f %f %f %f \n", &R(2,0),&R(2,1),&R(2,2),&t(2));
        fscanf(trajectory, "%f %f %f %f \n", &te(0),&te(1),&te(2),&te(3));

        Eigen::Matrix<double, 3, 3, Eigen::RowMajor> R1 = R.cast<double>();
        Eigen::Vector3d t1 = t.cast<double>();

        posetemp.setIdentity();
        posetemp.rotate(R1);
        posetemp.translation() = t1;

        camera_trajectory.push_back(posetemp);
    }
    fclose(trajectory);

    SlamLinearSolver* linearSolver = new SlamLinearSolver();
    linearSolver->setBlockOrdering( false );
    SlamBlockSolver* blockSolver = new SlamBlockSolver( linearSolver );
    g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg( blockSolver );

    g2o::SparseOptimizer globalOptimizer; 

    globalOptimizer.setAlgorithm( solver );
    globalOptimizer.setVerbose( true );

    g2o::VertexSE3* v = new g2o::VertexSE3();

    v->setId( 0 );
    v->setEstimate( camera_trajectory[0] );
[... 1798 lines cut ...]
-0.001960 0.996884 0.078851 0.894304 
0.506841 -0.066983 0.859433 0.170559 
0.000000 0.000000 0.000000 1.000000 
frame 346 
0.865238 0.051455 -0.498713 1.574152 
-0.011328 0.996472 0.083158 0.895686 
0.501233 -0.066302 0.862769 0.159370 
0.000000 0.000000 0.000000 1.000000 
frame 347 
0.868061 0.052929 -0.493627 1.548140 
-0.014357 0.996561 0.081607 0.897663 
0.496249 -0.063753 0.865836 0.144602 
0.000000 0.000000 0.000000 1.000000 
frame 348 
0.873771 0.042297 -0.484494 1.514104 
-0.009689 0.997527 0.069612 0.897299 
0.486240 -0.056131 0.872020 0.120187 
0.000000 0.000000 0.000000 1.000000 
frame 349 
0.880765 0.048731 -0.471041 1.481214 
-0.014848 0.997044 0.075384 0.897865 
0.473322 -0.059402 0.878885 0.090250 
0.000000 0.000000 0.000000 1.000000 

Thanks!

Xbbei, 2017-07-05:
Hello, I find I don't use pcl, My code can generate accurate result; But when I use the pcl in the project, My code generates strange result.

Xbbei, 2017-07-05:
Thank you very much! I use the pcl 1.7.2, and My code is normal.

closed 2017-07-05 by Xbbei

sjulier, 2017-07-05:
Thanks. This seems really odd though. Is it purely the act of linking different versions of PCL which were causing the issue?

RainerKuemmerle, 2017-07-05:
In my experience it relates to the compiler flags used for PCL.
I experienced strange effects when, for example, PCL uses heavy optimizing compiler flags and the other code does not.