RainerKuemmerle/g2o #184: g2o memory management

sebastiendalibard, 2017-07-04:
Hi,

g2o has a memory management pattern based on raw pointers, deleted by owner at object destruction. Although valid, this pattern hard to maintain without making mistakes. There actually are some memory leaks in the current code, for example [here](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/core/sparse_optimizer.cpp#L574) where _algorithm is not freed before reassigned.

The canonical solution in modern C++ would be to express object ownership with smart pointers. In particular, I would expect g2o to make an efficient usage of std::unique_ptr. The code would be lighter, without having to call delete, and less error prone.

Would you be willing to integrate patches going in that direction?

Thanks for your feedback,

Sébastien

jlblancoc, 2017-07-05:
👍 to this! Would also be happy of help pushing in that direction if @RainerKuemmerle agrees with the idea.

RainerKuemmerle, 2017-07-05:
Yes, I would appreciate contributions that incorporate smart pointer.

jlblancoc, 2017-07-05:
Great! Would you like to create a new branch for this feature? In this way, PRs would go against it and could be extensively tested before really merging them to `master`...

sjulier, 2017-07-08:
Don't know if it's of use, but I had started porting code to C++11 (c++11_stringent_tests branch). The changes I'd made so far were to change a lot of the iterators to use the : syntax and auto where possible to reduce the size of the code.

jumonb, 2017-07-21:
Hey y'all, so I've already got a version of g2o with disabled implicit ownership of memory to allow batched allocations, which makes creation time and destruction time much faster so you basically have non-owning pointers throughout the code for edges and vertices, but unique_ptrs for owning pointers. I believe this is the kind of change you're imagining, and once I fix my auth issues with my work github account I can create a pull request for it (probably next week)

jumonatr, 2017-07-21:
Nevermind got my issues worked out haha :)

zhangjun-xyz, 2020-03-05:
So, how's it going on? The reason I am here is I encountered a problem when using g2o. I used smart pointers to manage vertex and edge pointers, and without any doubt it failed me.

closed 2025-02-10 by github-actions[bot]