RainerKuemmerle/g2o #185: Unit of value returned by chi2

MalcolmMielle, 2017-07-07:
Hello,

I can't seem to find what is the unit of the value return by the chi2 function to get the mean error.
Does anyone know ?

Thanks a lot,

sjulier, 2017-07-07:
It is unitless. It's the squared Mahalanobis distance:

https://en.wikipedia.org/wiki/Mahalanobis_distance

When the distributions are Gaussian, it is chi^2 distributed and the number of degrees of freedom equals the dimension of the state.

MalcolmMielle, 2017-07-08:
Awesome thanks,

But then, how do we know if a change in the chi2 is significant if we don't have any unit ?

jlblancoc, 2017-07-08:
Read about "data association" in SLAM to get the idea of how chi2 "works". 
For example: 
http://ais.informatik.uni-freiburg.de/teaching/ws09/robotics2/pdfs/rob2-11-dataassociation.pdf

Cheers.

MalcolmMielle, 2017-07-08:
Thanks I will check it out !

_Sent from my Motorola XT1562 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

closed 2017-07-08 by MalcolmMielle

jlblancoc, 2017-07-08:
You're welcome!

We also wrote a kind of tutorial about this topic in [this paper](http://ingmec.ual.es/~jlblanco/papers/blanco2012amd.pdf). 

Cheers

sjulier, 2017-07-08:
I'm not sure what you want to use the chi^2 value for. As the discussion shows you can use it for data association (this uses explicit switching, alternatives would be to use a heavy tailed distribution - such as a Student t, see https://dl.dropboxusercontent.com/u/23948930/Papers/loxam-eccv-2008.pdf which can be encoded as a modified error function - or even something more mathematically sophisticated such as random sets). Conversely, chi2 should provide some sense of just whether the estimator is working "about right". If the converged value of chi2 is much bigger or much smaller than the sum of the dimensions of all the error vectors in the graph, it suggests the covariance is not a good match for the residuals which are being computed at the optimised solution.

MalcolmMielle, 2017-07-08:
I will need read more about all this since I'm not so familiar with it and it all sound very interesting! Especially the part about data association that I've never heard of before.

I wanted to use chi^2 to calculate and plot a convergence plot and look how many iterations are needed for the optimization to reach stability. I calculate chi^2 for each step and plot it and look at the evolution seeing when does the value gets stable. Does that sounds like a good way to just evaluate that.

As a side observation, I just wanted to evaluate the error in the graph but I didn't understood so much what chi^2 was. Hence my question earlier.

I'm sorry for all the simple questions I'm not sure what the dimensions of the error vectors are ? Is it the degree of freedom or is it something totally unrelated like the order of magnitude?

Thanks a lot to both of you for your help it's really helpful.

_Sent from my Motorola XT1562 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_