RainerKuemmerle/g2o #201: How did you derive Jacobian written in Edge_V_V_GICP::linearizeOplus()

takmin, 2017-08-29:
Hi, Thank you for this great open source library.
Now I'm trying to expand ICP algorithm to include scale optimization.
To do that, I'm implementing edge and vertex by modifying VertexSE3 and Edge_V_V_GICP.

My question is: how did you derive Jacobian written in Edge_V_V_GICP::linearizeOplus()?
In that code:
  // jacobian defined as:
  //    f(T0,T1) =  dR0.inv() * T0.inv() * (T1 * dR1 * p1 + dt1) - dt0
  //    df/dx0 = [-I, d[dR0.inv()]/dq0 * T01 * p1]
　//    df/dx1 = [R0, T01 * d[dR1]/dq1 * p1]

I read the paper of Generalized ICP, but could not find the equations above.
http://www.robots.ox.ac.uk/~avsegal/resources/papers/Generalized_ICP.pdf

I tried to derive it for myself, but couldn't.
If there are any papers or books which explain the above, please let me know.

takmin, 2017-09-29:
I've found the way.
It is derived from SO(3) Lie algebra and idea of exponential map 
http://ethaneade.com/lie.pdf

thanks

YangSiri, 2019-11-21:
Hi, @takmin 
Im glad you found your way. I just want to know how the _Edge_V_V_GICP_ works ?

LongruiDong, 2020-07-20:
> Hi, @takmin
> Im glad you found your way. I just want to know how the _Edge_V_V_GICP_ works ?

Hi, I am also questioning about  **Edge_V_V_GICP** : #430 
Have you solved now?

LongruiDong, 2020-07-20:
Hi, 
I am also questioning about  **Edge_V_V_GICP** : #430 
Have you solved now?

closed 2025-02-10 by github-actions[bot]