RainerKuemmerle/g2o #203: Segmentation fault when performing bundle adjustment

FinleyPan, 2017-09-09:
`void bundleAdjustment (
    const vector< cv::Point3f > points_3d,
    const vector< cv::Point2f > points_2d,
    const Mat& K,
    Mat& R, Mat& t )
{    
    typedef g2o::BlockSolver< g2o::BlockSolverTraits<6,3> > Block; 
    Block::LinearSolverType* linearSolver = new g2o::LinearSolverCSparse<Block::PoseMatrixType>(); 
    Block* solver_ptr = new Block ( linearSolver );    
    g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg ( solver_ptr );
    g2o::SparseOptimizer optimizer;
    optimizer.setAlgorithm ( solver );

    // vertex
   g2o::VertexSE3Expmap* pose = new g2o::VertexSE3Expmap(); // camera pose
    Eigen::Matrix3d R_mat;
    R_mat <<
          R.at<double> ( 0,0 ), R.at<double> ( 0,1 ), R.at<double> ( 0,2 ),
               R.at<double> ( 1,0 ), R.at<double> ( 1,1 ), R.at<double> ( 1,2 ),
               R.at<double> ( 2,0 ), R.at<double> ( 2,1 ), R.at<double> ( 2,2 );
    pose->setId ( 0 );
    pose->setEstimate ( g2o::SE3Quat (
                            R_mat,
                            Eigen::Vector3d ( t.at<double> ( 0,0 ), t.at<double> ( 1,0 ), t.at<double> ( 2,0 ) )
                        ) );
    optimizer.addVertex ( pose ); */   

    int index = 1;
    for (const  Point3f p:points_3d )   // landmarks,points_3d is world coordinate.
    {
        g2o::VertexSBAPointXYZ* point = new g2o::VertexSBAPointXYZ();
        point->setId ( index++ ); 
        point->setEstimate ( Eigen::Vector3d ( p.x, p.y, p.z ) );
        point->setMarginalized ( true ); 
       optimizer.addVertex ( point );
    }
    
    // parameter: camera intrinsics
    g2o::CameraParameters* camera = new g2o::CameraParameters (
        K.at<double> ( 0,0 ), Eigen::Vector2d ( K.at<double> ( 0,2 ), K.at<double> ( 1,2 ) ), 0
    );
    camera->setId ( 0 );
    optimizer.addParameter ( camera );

    // edges
    index = 1;
    for ( const Point2f p:points_2d )
    {
        g2o::EdgeProjectXYZ2UV* edge = new g2o::EdgeProjectXYZ2UV();
        edge->setId ( index );
        edge->setVertex ( 0, dynamic_cast<g2o::VertexSBAPointXYZ*> ( optimizer.vertex ( index ) ) );
        edge->setVertex ( 1, pose );
        edge->setMeasurement ( Eigen::Vector2d ( p.x, p.y ) );
        edge->setParameterId ( 0,0 );  //cameraparameter
        edge->setInformation ( Eigen::Matrix2d::Identity() );
        optimizer.addEdge ( edge );
        index++;
    }
 
    chrono::steady_clock::time_point t1 = chrono::steady_clock::now();
    optimizer.setVerbose ( true );
    optimizer.initializeOptimization();
    optimizer.optimize ( 100 );
    chrono::steady_clock::time_point t2 = chrono::steady_clock::now();
    chrono::duration<double> time_used = chrono::duration_cast<chrono::duration<double>> ( t2-t1 );
    cout<<"optimization costs time: "<<time_used.count() <<" seconds."<<endl;

    cout<<endl<<"after optimization:"<<endl;
    cout<<"T="<<endl<<Eigen::Isometry3d ( pose->estimate() ).matrix() <<endl;
    
}`

 Hello! I have defined a function to perform bundle adjustment. But when I ran the executable, terminal asserted **Segmentation Fault**. I found this problem too even if I commented all code except:
` g2o::VertexSE3Expmap* pose = new g2o::VertexSE3Expmap();`

closed 2017-10-07 by FinleyPan

beginnerFighting, 2017-11-07:
@FinleyPan 
Could you tell me how do you solve this problem?
I've met this situation too.

Easoncyx, 2018-05-08:
It is this line that has some problems. But I don't know how to solve it either. Somebody knows???
g2o::VertexSE3Expmap* pose = new g2o::VertexSE3Expmap(); // camera pose

YangSiri, 2019-10-26:
Hi, @FinleyPan 
i noticed you marked //cameraparameter at the end of setParameterId(0,0), could you explain more about this function ? Thanks a lot !

hwscut, 2019-10-29:
g2o::VertexSE3Expmap* pose = new g2o::VertexSE3Expmap(); 
double free or corruption (out)
have any solution?

YangSiri, 2019-11-09:
Hi, @hwscut  @Easoncyx 
Same problem, bros, have you tried older version of g2o?

hnsywangxin, 2019-11-28:
you can try change the mode of cmake(debug->release )

smallchimney, 2020-06-09:
> you can try change the mode of cmake(debug->release )

It works, but why?

smallchimney, 2020-06-09:
Here are the test codes:  
```
auto* v = new g2o::VertexSE3Expmap();
delete v
```
This will throw `double free or corruption (out)` and exit with a `SIGABRT` in Debug mode. If we change the compile mode from Debug to Release, the above code works.
```
g2o::SparseOptimizer optimizer;
auto* v = new g2o::VertexSE3Expmap();
v->setId(0);
optimizer.addVertex(v);
```
However, if we add just few lines, we get this error again, even in Release mode.
Here is my test environment:
* OS: Ubuntu 18.04 on X86;
* CMake: Ver 3.10.2 and Ver 3.12.4;
* g2o: [Latest master](https://github.com/RainerKuemmerle/g2o/commit/3c74d355165222bdcc5421ae370e2d78ebd33100)
* gcc: Ver 7.5 with c++11 and c++14

smallchimney, 2020-06-09:
I recompile the g2o library with `-DBUILD_WITH_MARCH_NATIVE=ON`, and after that all codes works fine in either Debug or Release mode.

hggaoq, 2020-06-10:
> 
> 
> I recompile the g2o library with `-DBUILD_WITH_MARCH_NATIVE=ON`, and after that all codes works fine in either Debug or Release mode.

Perfect solution!