RainerKuemmerle/g2o #204: how to do structure-only bundle adjustment with G2O

IQ17, 2017-09-09:
Hi, I want to do a simple bundle adjustment with only 2 camera vertices and about 100 landmark vertices
like the ORB-SLAM2 [BA here](https://github.com/raulmur/ORB_SLAM2/blob/5ad09bf8e9c1cb4f8f9a8789ebaffb2bb8d2fc98/src/Optimizer.cc#L49)

Specially, I would like to fix both camera vertices and only optimize 100 landmarks, how can I do this?
I tried to setFixed(true) and/or setMarginalized(true) both camera vertices, but the g2o crashes with debug info...

closed 2017-09-18 by IQ17

chi3x10, 2018-01-04:
Without optimizing the camera poses, you are just doing triangulation.

pascalenderli, 2018-08-14:
I face a similar problem and came across this issue here. I use a mixed dataset. Some nodes are fixed and some I want to optimize. I don't want to do triangulation of one subset first, because I am looking for a joint solution. Is it possible to fix some nodes in space?

salahkhan94, 2020-08-12:
Hi I'm attempting to do the same thing, triangulation using multiple views of known poses and ran into the same problem, did you manage to solve it ? 

> Without optimizing the camera poses, you are just doing triangulation. For triangulation, you don't even need g2o.
OpenCV has a triangulation function, but it's for a stereo frame kindly let me know of any toolsets with which I can do this for multiple views.

LongruiDong, 2021-03-24:
> Hi I'm attempting to do the same thing, triangulation using multiple views of known poses and ran into the same problem, did you manage to solve it ?

Hi  i  am doing the same thing with g2o, have you figure it out?

TomatoChewgum, 2021-11-04:
use this ： #include "g2o/solvers/structure_only/structure_only_solver.h"
I meet the same error ;  and then found this  method

goktugyildirim, 2022-05-30:
@TomatoChewgum hi, did you solve this problem using `include "g2o/solvers/structure_only/structure_only_solver.h"` ?
In the case of the structure only bundle adjustment if the view count is equal to two, It may be triangulation. I'm not sure