RainerKuemmerle/g2o #205: Double free or corruption when calling constructor with parameters of CameraParameters

FinleyPan, 2017-09-10:
The terminal asserted double free or corruption(out) when calling constructor with parameters of CameraParameters:
`g2o::CameraParameters* camera = new g2o::CameraParameters (K.at<double> ( 0,0 ), Eigen::Vector2d ( K.at<double> ( 0,2 ), K.at<double> ( 1,2 ) ), 0);  `
   `camera->setId ( 0 );`
  `  optimizer.addParameter ( camera );  `
  But it ran properly when I employed default constructor:
`g2o::CameraParameters* camera = new g2o::CameraParameters ();`

closed 2017-09-11 by FinleyPan

dingjiangang, 2018-05-26:
i met this problem again, so how to work it

dingjiangang, 2018-05-26:
@FinleyPan 
i met this problem again, so how to work it

jarvisyjw, 2024-12-19:
@dingjiangang Have you solved it? I also encountered this problem.