RainerKuemmerle/g2o #206: error: no matching function for call to ‘g2o

jiaxinLuTech, 2017-09-11:
When i make rpg_svo，then，there are two errors as fllow.I do not know what to do with.

tech-one@techone-virtual-machine:~/catkin_ws/src/rpg_svo/svo/build$ make 
Scanning dependencies of target svo
[  3%] Building CXX object CMakeFiles/svo.dir/src/bundle_adjustment.cpp.o
/home/tech-one/catkin_ws/src/rpg_svo/svo/src/bundle_adjustment.cpp: In function ‘void svo::ba::setupG2o(g2o::SparseOptimizer*)’:
/home/tech-one/catkin_ws/src/rpg_svo/svo/src/bundle_adjustment.cpp:357:76: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::BlockSolver(g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::LinearSolverType*&)’
   g2o::BlockSolver_6_3 * solver_ptr = new g2o::BlockSolver_6_3(linearSolver);
                                                                            ^
In file included from /usr/local/include/g2o/core/block_solver.h:199:0,
                 from /home/tech-one/catkin_ws/src/rpg_svo/svo/src/bundle_adjustment.cpp:20:
/usr/local/include/g2o/core/block_solver.hpp:40:1: note: candidate: g2o::BlockSolver<Traits>::BlockSolver(std::unique_ptr<typename Traits::LinearSolverType>) [with Traits = g2o::BlockSolverTraits<6, 3>; typename Traits::LinearSolverType = g2o::LinearSolver<Eigen::Matrix<double, 6, 6> >]
 BlockSolver<Traits>::BlockSolver(std::unique_ptr<LinearSolverType> linearSolver
 ^
/usr/local/include/g2o/core/block_solver.hpp:40:1: note:   no known conversion for argument 1 from ‘g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >::LinearSolverType* {aka g2o::LinearSolver<Eigen::Matrix<double, 6, 6> >*}’ to ‘std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, 6, 6> >, std::default_delete<g2o::LinearSolver<Eigen::Matrix<double, 6, 6> > > >’
/home/tech-one/catkin_ws/src/rpg_svo/svo/src/bundle_adjustment.cpp:358:100: error: no matching function for call to ‘g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(g2o::BlockSolver_6_3*&)’
 orithmLevenberg * solver = new g2o::OptimizationAlgorithmLevenberg(solver_ptr);
                                                                              ^
In file included from /home/tech-one/catkin_ws/src/rpg_svo/svo/src/bundle_adjustment.cpp:23:0:
/usr/local/include/g2o/core/optimization_algorithm_levenberg.h:47:16: note: candidate: g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::unique_ptr<g2o::Solver>)
       explicit OptimizationAlgorithmLevenberg(std::unique_ptr<Solver> solver);
                ^
/usr/local/include/g2o/core/optimization_algorithm_levenberg.h:47:16: note:   no known conversion for argument 1 from ‘g2o::BlockSolver_6_3* {aka g2o::BlockSolver<g2o::BlockSolverTraits<6, 3> >*}’ to ‘std::unique_ptr<g2o::Solver>’
CMakeFiles/svo.dir/build.make:398: recipe for target 'CMakeFiles/svo.dir/src/bundle_adjustment.cpp.o' failed
make[2]: *** [CMakeFiles/svo.dir/src/bundle_adjustment.cpp.o] Error 1
CMakeFiles/Makefile2:557: recipe for target 'CMakeFiles/svo.dir/all' failed
make[1]: *** [CMakeFiles/svo.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

RainerKuemmerle, 2017-09-11:
Please consider using an older version of g2o.
The latest master branch incorporates changes that are API incompatible.
For example, you may consider using
https://github.com/RainerKuemmerle/g2o/releases/tag/20170730_git

closed 2017-09-11 by RainerKuemmerle

jumonatr, 2017-09-11:
Optimization algorithms no longer take raw pointers as they own the solver (they delete it once they get deleted).

In order to make it clearer and avoid either double deletion or memory leaks the algorithm now takes a unique_ptr<Solver> which means you need to create your BlockSolver_6_3 using either the regular unique_ptr constructor, std::make_unique, or g2o::make_unique if you don't have the C++14 std enabled.

jiaxinLuTech, 2017-09-12:
I use the older version to solve the question,but there are other questions as fllow.

[ 63%] Linking CXX executable ../bin/test_feature_detection
../lib/libsvo.so：undefined reference to ‘g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)’
../lib/libsvo.so：undefined reference to ‘g2o::csparse_extension::cs_cholsolsymb(cs_di_sparse const*, double*, cs_di_symbolic const*, double*, int*)’
../lib/libsvo.so：undefined reference to ‘g2o::csparse_extension::writeCs2Octave(char const*, cs_di_sparse const*, bool)’
collect2: error: ld returned 1 exit status
CMakeFiles/test_feature_detection.dir/build.make:271: recipe for target '../bin/test_feature_detection' failed
make[2]: *** [../bin/test_feature_detection] Error 1
CMakeFiles/Makefile2:520: recipe for target 'CMakeFiles/test_feature_detection.dir/all' failed
make[1]: *** [CMakeFiles/test_feature_detection.dir/all] Error 2
[ 66%] Linking CXX executable ../bin/test_pose_optimizer
../lib/libsvo.so：undefined reference to ‘g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)’
../lib/libsvo.so：undefined reference to ‘g2o::csparse_extension::cs_cholsolsymb(cs_di_sparse const*, double*, cs_di_symbolic const*, double*, int*)’
../lib/libsvo.so：undefined reference to ‘g2o::csparse_extension::writeCs2Octave(char const*, cs_di_sparse const*, bool)’
collect2: error: ld returned 1 exit status
CMakeFiles/test_pose_optimizer.dir/build.make:271: recipe for target '../bin/test_pose_optimizer' failed
make[2]: *** [../bin/test_pose_optimizer] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/test_pose_optimizer.dir/all' failed
make[1]: *** [CMakeFiles/test_pose_optimizer.dir/all] Error 2
Makefile:83: recipe for target 'all' failed

I google this question and find a solution.

https://github.com/tum-vision/dvo_slam/issues/2

I make g2o again according to reply,but there are the same question.
I hope get your help,thank you!

daliskon, 2017-10-16:
Did you find a solution?

escrowdis, 2018-10-31:
I found someone has solved it. If you want to revise the file, try to use unique_ptr instead of new and feed it into solver, like this:

```c++
// new
// g2o::BlockSolver_6_3::LinearSolverType* linearSolver;
// linearSolver = new g2o::LinearSolverCholmod<g2o::BlockSolver_6_3::PoseMatrixType>();
// unique_ptr
std::unique_ptr<g2o::BlockSolver_6_3::LinearSolverType> linearSolver (new g2o::LinearSolverCholmod<g2o::BlockSolver_6_3::PoseMatrixType>());

// g2o::BlockSolver_6_3 * solver_ptr = new g2o::BlockSolver_6_3(linearSolver);
std::unique_ptr<g2o::BlockSolver_6_3> solver_ptr (new g2o::BlockSolver_6_3(std::move(linearSolver)));
g2o::OptimizationAlgorithmLevenberg * solver = new g2o::OptimizationAlgorithmLevenberg(std::move(solver_ptr));
```

Ref.: https://hk.saowen.com/a/f94e802cf38e63679d28e0bd8e614151cc131402d8553acdec8bb993c92e2430

mrgransky, 2018-11-30:
I fixed the problem with `std::unique_ptr` too. In the following I have a snap of my code showing the previous (commented) and new approach:
```
        // Setup optimizer
	g2o::SparseOptimizer optimizer;

	/*// old approach
	g2o::BlockSolverX::LinearSolverType * linearSolver;

	linearSolver = new g2o::LinearSolverEigen<g2o::BlockSolverX::PoseMatrixType>();

	g2o::BlockSolverX * solver_ptr = new g2o::BlockSolverX(linearSolver);

	//g2o::OptimizationAlgorithmLevenberg* solver = new g2o::OptimizationAlgorithmLevenberg(solver_ptr);
	*/

	// fix with std::unique_ptr, linear solver:
	std::unique_ptr<g2o::BlockSolverX::LinearSolverType> linearSolver 
				(new g2o::LinearSolverEigen<g2o::BlockSolverX::PoseMatrixType>());

	// fix with std::unique_ptr, solver_ptr:
	std::unique_ptr<g2o::BlockSolverX> solver_ptr (new g2o::BlockSolverX(std::move(linearSolver)));

//	g2o::OptimizationAlgorithmLevenberg * solver = new g2o::OptimizationAlgorithmLevenberg(std::move(solver_ptr));
```
Thanks to @escrowdis .

liamjwang, 2019-01-04:
As an alternative to the above solutions, if you happen to be looking for the source of g2o used in ROS you can download an older source from the ROS package host. 

For example, the g2o source used in ROS kinetic can be found at:
http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-libg2o/ros-kinetic-libg2o_2016.4.24.orig.tar.gz

Quanghai32, 2020-05-15:
> Please consider using an older version of g2o.
> The latest master branch incorporates changes that are API incompatible.
> For example, you may consider using
> https://github.com/RainerKuemmerle/g2o/releases/tag/20170730_git

How can I install this version? Thank you!

sidiangongyuan, 2023-12-11:
> I fixed the problem with `std::unique_ptr` too. In the following I have a snap of my code showing the previous (commented) and new approach:
>
> ```
>         // Setup optimizer
> 	g2o::SparseOptimizer [...]

That works, thank you.